arendst/Tasmota #11120: Scripting MQTT subscribe not updating variable with 9.3.1

infoflo, 2021-02-24:
### PROBLEM DESCRIPTION
After upgrade to 9.3.1 Scripting is not updating the variable anymore. Manually printing the variables shows empty string.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Basic, Sonoff Touch, Gosund SP1
- [x] Tasmota binary firmware version number used: 9.3.1
  - [ ] Pre-compiled
  - [x] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
>D
F=""
P=""
N=""

>S
if mqttc>0 {
  +>Subscribe F,home/F
  +>Subscribe P,home/P
  +>Subscribe N,home/N
}

>F
if upd[F]>0
or upd[P]>0
or upd[N]>0 {
...
}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
no output (even if it should have some)

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Update to 9.3.1. Use MQTT subscribe with scripting. Same script worked before with 9.2.0 (downgrade made it work again)

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Variables updated and upd[x] triggered.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

gemu2015, 2021-02-25:
please compile with 
#define DEBUG_MQTT_EVENT
and check for console messages and report here

nagyrobi, 2021-02-26:
> report here
Hi @gemu2015 can you please have a look at this too: https://github.com/arendst/Tasmota/discussions/11133

infoflo, 2021-02-27:
17:29:20.533 Script: MQTT Topic home/DG/Fire, Event false
17:29:20.535 Script: Match MQTT message Topic home/DG/Fire with subscription topic home/DG/Fire and key 
17:29:20.538 Script: Match MQTT message Topic home/DG/Fire with subscription topic home/DG/Presence and key 
17:29:20.540 Script: Match MQTT message Topic home/DG/Fire with subscription topic home/DG/Night and key 
17:29:20.633 Script: MQTT Topic home/DG/Presence, Event true
17:29:20.636 Script: Match MQTT message Topic home/DG/Presence with subscription topic home/DG/Fire and key 
17:29:20.639 Script: Match MQTT message Topic home/DG/Presence with subscription topic home/DG/Presence and key 
17:29:20.642 Script: Match MQTT message Topic home/DG/Presence with subscription topic home/DG/Night and key 
17:29:20.835 Script: MQTT Topic home/DG/Night, Event false
17:29:20.838 Script: Match MQTT message Topic home/DG/Night with subscription topic home/DG/Fire and key 
17:29:20.841 Script: Match MQTT message Topic home/DG/Night with subscription topic home/DG/Presence and key 
17:29:20.843 Script: Match MQTT message Topic home/DG/Night with subscription topic home/DG/Night and key 
17:29:48.444 CMD: Script >print %N%
17:29:48.446 SRC: WebConsole from 192.168.178.105
17:29:48.449 CMD: Group 0, Index 1, Command "SCRIPT", Data ">print %N%"
17:29:48.454 
17:29:48.459 MQT: tasmota/DG/KU/Ceiling/stat/RESULT = {"Script":">print %N%"}

gemu2015, 2021-02-27:
ok i think i found the bug. you are using no keys, this options was missing since a recent code change.
please try to compile with this file

[xdrv_10_scripter.ino.zip](https://github.com/arendst/Tasmota/files/6054954/xdrv_10_scripter.ino.zip)

closed 2021-02-28 by arendst

infoflo, 2021-03-01:
Tested your attached scripter code and it works again. Thx a lot!

PR #11150 (2021-02-28): Fixes
## Description:

**Related issue (if applicable):** fixes #11120
fixes subscribe with no keys
fixes sml math scaling and immediate numbers

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works on Tasmota core ESP8266 V.2.7.4.9
  - [x] The code change is tested and works on Tasmota core ESP32 V.1.0.5-rc6
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/C [... cut]