arendst/Tasmota #11124: Tasmota-display crashes with USE_UFILESYS enabled

nonix, 2021-02-25:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): WeMos D1 R2 mini with TFT ILI9341
- [x ] Tasmota binary firmware version number used: 9.3.1.1(display)
  - [ ] Pre-compiled --No
  - [ x] Self-compiled
- [ x] Flashing tools used: Tasmotizer 1.2.1
- [ x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
12:30:41.905 RSL: stat/tasmota_1800EB/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
12:30:42.111 RSL: stat/tasmota_1800EB/RESULT = {"Module":{"18":"Generic"}}
12:30:42.363 RSL: stat/tasmota_1800EB/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
2:31:40.312 RSL: stat/tasmota_1800EB/STATUS = {"Status":{"Module":18,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_1800EB","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
12:31:40.317 RSL: stat/tasmota_1800EB/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Exception","Uptime":"0T00:14:32","StartupUTC":"2021-02-25T11:17:08","Sleep":50,"CfgHolder":4617,"BootCount":13,"BCResetTime":"2021-02-25T10:32:43","SaveCount":33,"SaveAddress":"FB000"}}
12:31:40.321 RSL: stat/tasmota_1800EB/STATUS2 = {"StatusFWR":{"Version":"9.3.1.1(display)","BuildDateTime":"2021-02-25T12:13:49","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"354/699"}}
12:31:40.325 RSL: stat/tasmota_1800EB/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["0325125441",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000080","00006000","00000000"]}}
12:31:40.334 RSL: stat/tasmota_1800EB/STATUS4 = {"StatusMEM":{"ProgramSize":611,"Free":392,"Heap":28,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0FAA858E","04049FA1","000000C3","00000000","00004080","00000020","403F8000","00000000"],"Drivers":"1,2,4,5,8,9,10,13,16,50","Sensors":"1,2,5,6"}}
12:31:40.340 RSL: stat/tasmota_1800EB/STATUS5 = {"StatusNET":{"Hostname":"Display","IPAddress":"192.168.13.108","Gateway":"192.168.13.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.13.1","Mac":"DC:4F:22:18:00:EB","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
12:31:40.344 RSL: stat/tasmota_1800EB/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_1800EB","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
12:31:40.351 RSL: stat/tasmota_1800EB/STATUS7 = {"StatusTIM":{"UTC":"2021-02-25T11:31:40","Local":"2021-02-25T12:31:40","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"07:39","Sunset":"18:26"}}
12:31:40.357 RSL: stat/tasmota_1800EB/STATUS10 = {"StatusSNS":{"Time":"2021-02-25T12:31:40"}}
12:31:40.362 RSL: stat/tasmota_1800EB/STATUS11 = {"StatusSTS":{"Time":"2021-02-25T12:31:40","Uptime":"0T00:14:32","UptimeSec":872,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"0325125441","BSSId":"34:31:C4:CA:7D:E4","Channel":6,"RSSI":80,"Signal":-60,"LinkCount":1,"Downtime":"0T00:00:03"}}}
12:31:40.369 RSL: stat/tasmota_1800EB/STATUS12 = {"StatusSTK":{"Exception":28,"Reason":"Exception","EPC":["40243a16","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["4020a26b","40101288","40224ba4","40101288","40239650","40101288","4021742e","40218e35","40219087","4024d038","4021840c","402198f3","4021a2c8","40224cc4","40224d79","40215c7d","402266ff","40226a79","4024650c","40101e89"]}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
12:33:22.435 QPC: Reset
12:33:23.414 RSL: tele/tasmota_1800EB/STATE = {"Time":"2021-02-25T12:33:23","Uptime":"0T00:00:09","UptimeSec":9,"Heap":29,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"0325125441","BSSId":"34:31:C4:CA:7D:E4","Channel":6,"RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:03"}}
12:34:14.690 MOD: Generic Module, IO0 0, IO1 0, IO2 0, IO3 0, IO4 0, IO5 0, IO12 0, IO13 0, IO14 0, IO15 5920, IO16 5888, IO17 0
12:34:16.535 APP: Restarting

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 3584, room 16
tail 0
chksum 0xb0
csum 0xb0
v3969889e
~ld

00:00:00.050 CFG: Loaded from flash at F9, Count 35
00:00:00.055 QPC: Count 1
00:00:00.058 SPI: Hardware using GPIO14(CLK), GPIO13(MOSI) and GPIO12(MISO)
00:00:00.912 DSP: ILI9341

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (28):
epc1=0x40243a16 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffbf0 end: 3fffffc0 offset: 0190
3ffffd80:  3ffffdc0 3ffffd90 00000008 3fff1568
3ffffd90:  3fff1a68 ffffffff 00000002 4020a26b
3ffffda0:  3a505344 494c4920 31343339 50a3d700
3ffffdb0:  3fff2534 0000004e 3ffffd90 3fff0316
3ffffdc0:  00000003 00000032 00000002 00000000
3ffffdd0:  007a1200 00000001 0000000f 40101288
3ffffde0:  000000ef 0000002c 3ffffe70 40224ba4
3ffffdf0:  00000000 00000000 00000010 40101288
3ffffe00:  00000140 000000f0 3fff3b5c 40239650
3ffffe10:  00000000 00000001 0000000f 40101288
3ffffe20:  3fff1a68 3ffffe40 3ffffe30 00000008 <
3ffffe30:  3fff1a68 ffffffff 00000001 00000001
3ffffe40:  00000000 00000000 3ffeeecc 4021742e
3ffffe50:  3fff1a68 ffffffff 00000001 40218e35
3ffffe60:  0000000f ffffffff 00000001 40219087
3ffffe70:  4024d038 00000000 000003e8 4021840c
3ffffe80:  00000000 00000000 00000000 00000000
3ffffe90:  00000000 ffe0ffe0 ffe0ffe0 402198f3
3ffffea0:  fffffffc 00000000 00000000 4021a2c8
3ffffeb0:  3fff1368 00000000 00000034 00000f00
3ffffec0:  00000001 3fff1568 3ffeeecc 40224cc4
3ffffed0:  00000001 0000001c 00030000 40224d79
3ffffee0:  0000000f 0000000c 00000003 40215c7d
3ffffef0:  0000000f 00000004 00000008 402266ff
3fffff00:  ffe0ffe0 ffe0ffe0 ffe0ffe0 00000000
3fffff10:  ffe00000 0000ffe0 ffe0ffe0 ffe0ffe0
3fffff20:  ffe0ffe0 3fffff30 0000000c ffffffff
3fffff30:  3ffef97a 3ffefe76 ffffffff 00000030
3fffff40:  3fff31d8 00000000 00000008 3ffefbb0
3fffff50:  3ffefbc0 3ffefbe1 3ffef958 40226a79
3fffff60:  feefeffe feefeffe feefeffe feefeffe
3fffff70:  feefeffe feefeffe feefeffe feefeffe
3fffff80:  3fff1368 3fff1768 feefeffe feefeffe
3fffff90:  feefeffe feefeffe feefeffe 3fff2858
3fffffa0:  3fffdad0 00000000 3fff2818 4024650c
3fffffb0:  feefeffe feefeffe 3ffe864c 40101e89
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

```

### TO REPRODUCE
HW: WeMosD1 R2 mini with TFT ILI9341
Generic module configuration:
ILI9341_CS => GPIO16
ILI9341_DC => GPIO15

Compile tasmota-display with user_override setting: USE_UFILESYS

### EXPECTED BEHAVIOUR
Not to crash

### SCREENSHOTS
N/A -- See the console log above

### ADDITIONAL CONTEXT
The display works as expected when USE_UFILESYS is NOT defined.

**(Please, remember to close the issue when the problem has been addressed)**

gemu2015, 2021-02-25:
did you use a 4m2m linker file ?
for USE_UFILESYS you need file system space in flash 

in platform_override 
board_build.ldscript        =  eagle.flash.4m2m.ld

closed 2021-02-25 by arendst

nonix, 2021-02-25:
Hi guys, thank you so much, all perfect and works well as expected.

Cheers,
 N.

PR #11125 (2021-02-25): Fixes
## Description:

**Related issue (if applicable):** fixes #11124 ???
fixes a potential crash when USE_UFILESYS compiled but no file system on flash found
sml dump recoded

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works on Tasmota core ESP8266 V.2.7.4.9
  - [x] The code change is tested and works on Tasmota core ESP32 V.1.0.5-rc6
  - [x] I accept the [CLA](https://github.com/ar [... cut]