arendst/Tasmota #11129: status reply returns invalid json

adkrprog, 2021-02-25:
### PROBLEM DESCRIPTION
The response to http://IP/cm?cmnd=Status+0 generates invalid JSON. 
This breaks tasmoadmin with modules configured with two shutters

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ X Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): sonoff pro R2
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [X] Self-compiled latest
- [ ] Flashing tools used: _____
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
19:15:52.948 MQT: stat/atrium/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
19:15:53.172 MQT: stat/atrium/RESULT = {"Module":{"23":"Sonoff 4CH Pro"}}
19:15:53.429 MQT: stat/atrium/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"226":"Relay3"},"GPIO5":{"225":"Relay2"},"GPIO9":{"33":"Button2"},"GPIO10":{"34":"Button3"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"35":"Button4"},"GPIO15":{"227":"Relay4"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [X] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
"Status":{"Module":23,"DeviceName":"atrium","FriendlyName":["atrium l up","atrium l down","atrium r up","atrium r down"],"Topic":"atrium","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
19:17:18.334 MQT: stat/atrium/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T20:38:00","StartupUTC":"2021-02-24T21:39:18","Sleep":50,"CfgHolder":4617,"BootCount":29,"BCResetTime":"2020-06-17T18:57:51","SaveCount":229,"SaveAddress":"F9000"}}
19:17:18.352 MQT: stat/atrium/STATUS2 = {"StatusFWR":{"Version":"9.3.1(tasmota)","BuildDateTime":"2021-02-24T22:26:57","Boot":7,"Core":"2_7_4_3","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"415/699"}}
19:17:18.369 MQT: stat/atrium/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["myWlan",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","40000080","00206000","00000000"]}}
19:17:18.390 MQT: stat/atrium/STATUS4 = {"StatusMEM":{"ProgramSize":599,"Free":404,"Heap":21,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8F8AC7C3","04368001","0000000F","010013C0","C000F981","00004004","00001000","00000000"],"Drivers":"1,2,3,4,5,6,8,9,10,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4"}}
19:17:18.407 MQT: stat/atrium/STATUS5 = {"StatusNET":{"Hostname":"atrium","IPAddress":"192.168.13.170","Gateway":"192.168.13.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.13.1","Mac":"2C:F4:32:C4:DD:45","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
19:17:18.423 MQT: stat/atrium/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.10.10","MqttPort":8883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_C4DD45","MqttUser":"tasmota","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
19:17:18.442 MQT: stat/atrium/STATUS7 = {"StatusTIM":{"UTC":"2021-02-25T18:17:18","Local":"2021-02-25T19:17:18","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"07:39","Sunset":"18:26"}}
19:17:18.459 MQT: stat/atrium/STATUS10 = {"StatusSNS":{"Time":"2021-02-25T19:17:18","Shutter1":{"Position":0,"Direction":0,"Target":0},"Shutter2":{"Position":100,"Direction":0,"Target":100}}}
19:17:18.478 MQT: stat/atrium/STATUS11 = {"StatusSTS":{"Time":"2021-02-25T19:17:18","Uptime":"0T20:38:00","UptimeSec":74280,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"myWlan","BSSId":"CC:32:E5:5F:FC:FB","Channel":11,"RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:03"}}}
19:17:18.495 MQT: stat/atrium/STATUS13 = {"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":100,"Close":100,"50perc":50,"Delay":0,"Opt":"0000","Calib":[300,500,700,900,1000],"Mode":"0"}},{"SHT1":{"Relay1":3,"Relay2":4,"Open":100,"Close":100,"50perc":50,"Delay":0,"Opt":"0000","Calib":[300,500,700,900,1000],"Mode":"0"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
- create two shutters
- access http://IP/cm?cmnd=Status+0 (what tasmoadmin does)
- The output is invalid JSON
{"Status":{"Module":23,"DeviceName":"atrium","FriendlyName":["atrium l up","atrium l down","atrium r up","atrium r down"],"Topic":"atrium","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0},"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T08:34:00","StartupUTC":"2021-02-24T21:39:18","Sleep":50,"CfgHolder":4617,"BootCount":29,"BCResetTime":"2020-06-17T18:57:51","SaveCount":229,"SaveAddress":"F9000"},"StatusFWR":{"Version":"9.3.1(tasmota)","BuildDateTime":"2021-02-24T22:26:57","Boot":7,"Core":"2_7_4_3","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"415/699"},"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["myWlan",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","40000080","00206000","00000000"]},"StatusMEM":{"ProgramSize":599,"Free":404,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8F8AC7C3","04368001","0000000F","010013C0","C000F981","00004004","00001000","00000000"],"Drivers":"1,2,3,4,5,6,8,9,10,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4"},"StatusNET":{"Hostname":"atrium","IPAddress":"192.168.13.170","Gateway":"192.168.13.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.13.1","Mac":"2C:F4:32:C4:DD:45","Webserver":2,"WifiConfig":4,"WifiPower":17.0},"StatusMQT":{"MqttHost":"192.168.10.10","MqttPort":8883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_C4DD45","MqttUser":"tasmota","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30},"StatusTIM":{"UTC":"2021-02-25T06:13:18","Local":"2021-02-25T07:13:18","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"07:39","Sunset":"18:26"},"StatusSNS":{"Time":"2021-02-25T07:13:18","Shutter1":{"Position":0,"Direction":0,"Target":0},"Shutter2":{"Position":100,"Direction":0,"Target":100}},"StatusSTS":{"Time":"2021-02-25T07:13:18","Uptime":"0T08:34:00","UptimeSec":30840,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"myWlan","BSSId":"CC:32:E5:5F:FC:FB","Channel":11,"RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:03"}},"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":100,"Close":100,"50perc":50,"Delay":0,"Opt":"0000","Calib":[300,500,700,900,1000],"Mode":"0"}},{"SHT1":{"Relay1":3,"Relay2":4,"Open":100,"Close":100,"50perc":50,"Delay":0,"Opt":"0000","Calib":[300,500,700,900,1000],"Mode":"0"}}}

### EXPECTED BEHAVIOUR
Here is the manually corrected JSON: 
[... cut ...] SHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":100,"Close":100,"50perc":50,"Delay":0,"Opt":"0000","Calib":[300,500,700,900,1000],"Mode":"0"}},"SHT1":{"Relay1":3,"Relay2":4,"Open":100,"Close":100,"50perc":50,"Delay":0,"Opt":"0000","Calib":[300,500,700,900,1000],"Mode":"0"}}

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2021-02-25:
Hi, thanks for reporting. If you do `status 13` on the console it also produces the invalid JSON:

```lua
STATUS13 = {"StatusSHT":{"SHT0":{"Relay1":1,"......,"Mode":"0"}},{"SHT1":{"Relay1":3,........,"Mode":"0"}}}
```

And should be:

```lua
STATUS13 = {"StatusSHT":{"SHT0":{"Relay1":1,"......,"Mode":"0"},"SHT1":{"Relay1":3,........,"Mode":"0"}}}
```

adkrprog, 2021-02-26:
Hi, I can confirm the bug is fixed
Many thanks!

closed 2021-02-26 by adkrprog