arendst/Tasmota #11131: Small but very noticeable delay on Sonoff mini toggle.

GijsTimmers, 2021-02-25:
### PROBLEM DESCRIPTION
I have a Sonoff Mini. S1 and S2 are connected to an external switch. When pressing the switch, a MQTT message is sent, toggling another light bulb. This takes way longer than when I manually send a MQTT message.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Mini
- [x] Tasmota binary firmware version number used: 9.3.1
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: OTA from earliler Tasmota
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
00:09:03.388 CMD: Backlog Template; Module; GPIO 255
00:09:03.452 MQT: stat/ganglampschakelaar-boven/RESULT = {"NAME":"Sonoff Mini","GPIO":[0,0,0,0,160,0,0,0,0,320,0,0,1,0],"FLAG":0,"BASE":1}
00:09:03.477 MQT: stat/ganglampschakelaar-boven/RESULT = {"Module":{"0":"Sonoff Mini"}}
00:09:03.500 MQT: stat/ganglampschakelaar-boven/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"160":"Switch1"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
00:12:38.290 MQT: stat/ganglampschakelaar-boven/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":60,"Free":451,"Rules":"ON SWITCH1#STATE DO Publish cmnd/ganglamp/POWER toggle ENDON"}
00:12:38.311 MQT: stat/ganglampschakelaar-boven/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
00:12:38.331 MQT: stat/ganglampschakelaar-boven/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
```
- [x] Provide the output of this command: `Status 0`:
```lua
00:13:04.451 MQT: stat/ganglampschakelaar-boven/STATUS = {"Status":{"Module":0,"DeviceName":"ganglampschakelaar-boven","FriendlyName":["ganglampschakelaar-boven"],"Topic":"ganglampschakelaar-boven","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
00:13:04.488 MQT: stat/ganglampschakelaar-boven/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:04:25","StartupUTC":"2021-02-25T23:08:39","Sleep":20,"CfgHolder":4617,"BootCount":17,"BCResetTime":"2021-02-01T21:10:11","SaveCount":94,"SaveAddress":"FA000"}}
00:13:04.527 MQT: stat/ganglampschakelaar-boven/STATUS2 = {"StatusFWR":{"Version":"9.3.1(tasmota)","BuildDateTime":"2021-02-23T12:02:38","Boot":7,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"474/699"}}
00:13:04.545 MQT: stat/ganglampschakelaar-boven/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":2,"SysLog":2,"LogHost":"","LogPort":514,"SSId":["Dok 18",""],"TelePeriod":0,"Resolution":"558180C0","SetOption":["0000A009","280500000100060000005A0A000000000000","00000080","00006000","00000000"]}}
00:13:04.576 MQT: stat/ganglampschakelaar-boven/STATUS4 = {"StatusMEM":{"ProgramSize":593,"Free":408,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
00:13:04.611 MQT: stat/ganglampschakelaar-boven/STATUS5 = {"StatusNET":{"Hostname":"glsbo","IPAddress":"192.168.1.191","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"60:01:94:E6:6C:C0","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
00:13:04.634 MQT: stat/ganglampschakelaar-boven/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.4","MqttPort":1883,"MqttClientMask":"ganglampschakelaar-boven","MqttClient":"ganglampschakelaar-boven","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
00:13:04.659 MQT: stat/ganglampschakelaar-boven/STATUS7 = {"StatusTIM":{"UTC":"2021-02-25T23:13:04","Local":"2021-02-26T00:13:04","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"07:37","Sunset":"18:28"}}
00:13:04.680 MQT: stat/ganglampschakelaar-boven/STATUS10 = {"StatusSNS":{"Time":"2021-02-26T00:13:04","Switch1":"OFF"}}
00:13:04.690 MQT: stat/ganglampschakelaar-boven/STATUS11 = {"StatusSTS":{"Time":"2021-02-26T00:13:04","Uptime":"0T00:04:25","UptimeSec":265,"Heap":25,"SleepMode":"Dynamic","Sleep":20,"LoadAvg":49,"MqttCount":1,"Wifi":{"AP":1,"SSId":"XXXX","BSSId":"XXXX","Channel":11,"RSSI":100,"Signal":-46,"LinkCount":1,"Downtime":"0T00:00:13"}}}
```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
00:14:39.258 RUL: SWITCH1#STATE performs "Publish cmnd/ganglamp/POWER toggle"
00:14:39.261 SRC: Rule
00:14:39.264 CMD: Group 0, Index 1, Command "PUBLISH", Data "cmnd/ganglamp/POWER toggle"
00:14:39.273 MQT: cmnd/ganglamp/POWER = toggle
```

### TO REPRODUCE
Have one light listen to `cmnd/ganglamp/toggle`, have a Sonoff Mini set up with an external switch on S1 and S2, set Rule1 as `ON SWITCH1#STATE DO Publish cmnd/ganglamp/POWER toggle ENDON`. Pressing the external button will control 

### EXPECTED BEHAVIOUR
Have the light toggle its power with very little delay.

### SCREENSHOTS
I made a video so you can see the delay: https://imgur.com/a/SlcYwez . It's small but noticeable.

### ADDITIONAL CONTEXT
-I asked this question on the Tasmota discord but nobody over there could help me.
- There was at least 1 other user in the Tasmota discord who experienced the same problem.
- `SetOption13` is set to 1.
- Disabling the rule and just using a SwitchTopic created the same behaviour.

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2021-02-26:
Have you tried to change sleep setting? You can try `sleep 0` This disables Tasmota sleep function. 
More responsive but needs more energy

GijsTimmers, 2021-02-26:
Hi, yes, I tried that, but that did not help.

ascillato, 2021-02-26:
I don't think this could be improved from the software side since, as you see in your log, from the event of the switch pushed until the mqtt message is sent, you have 15 milliseconds. So, your delay seems to be in your wifi network. May be it is too crowded? Too many devices? Old router? Too many networks on the same or close channels? High traffic usage?
I use something similar to yours and the delay is not noticeable at all. It gives you the impression as the switch is actually connected to the lamp. The only delay I saw on that was when I was testing wifi channels and bandwidths in my router and had a bad configuration. When I tried a super cheap router, I always have a small delay, but when using a TP-Link router I didn't. The best timing I have achieved was when changing my router and APs to mikrotik.

GijsTimmers, 2021-02-26:
Thanks for taking the time looking into my issue. I agree that there's nothing wrong with the time between `RUL` and `MQT`. That can't be the issue.

On the other hand, I do believe that the issue is in the device and/or in Tasmota and not in the wifi network. I have a few reasons to believe this:

- Issuing MQTT commands from my phone results in almost instantaneous light toggling (making me believe that the MQTT server, wifi itself and receiving Tasmota are fine)
- Using a wifi hotspot on my phone, connecting only the two Tasmotas to it, led to the same 'sometimes delayed' behaviour (making me believe that the issue is wifi independent)
- I also tried substituting the MQTT command to a direct `WebSend` to the other tasmota's IP, omitting the MQTT server, but that led to the same behaviour (making me believe that the MQTT server is fine).

Please let me know if there is anything I can do to help resolve this issue. I understand that this might be a difficult issue to solve remotely, but I'd really like to have it fixed.

nagyrobi, 2021-02-26:
I use this to reduce delay to almost 0:
`Backlog SetOption1 1; SetOption13 1; SetOption65 1`

GijsTimmers, 2021-02-26:
@nagyrobi thanks, I used your command but sadly this did not change its behaviour.

nagyrobi, 2021-02-26:
How many Sonoff Minis around the house and they don't experience this. Maybe it's faulty?

jaymemaurice, 2021-02-26:
So you speculate the issue is the time between data being added to the socket on MQTT client, it leaving the hardware, and arriving on the network/remote device? 

Can you make the esp device is sync'd to the same NTP server as your computer/router/mqtt server and take a packet capture of the mqtt traffic corresponding to the log captures with timestamps? this should give some useful profiling data which should give helpful clues...

nagyrobi, 2021-02-27:
Can you post a video?

BCsabaEngine, 2021-02-27:
I see the power off is instantly, but power on has a small constant delay. It can be occured by LED lamp initial latency? I have got an expensive kitchen light with 400ms delay, normally.

GijsTimmers, 2021-02-27:
I will reply to the other comments soon. I made a video to demonstrate the physical switch behaviour compared to MQTT signals from Openhab.

https://user-images.githubusercontent.com/972314/109390019-6d4a7a80-790f-11eb-8e54-58111e08fd93.MOV

GijsTimmers, 2021-02-27:
https://user-images.githubusercontent.com/972314/109399325-799bfb00-7942-11eb-8c7a-b917988926d6.mp4

GijsTimmers, 2021-02-27:
> So you speculate the issue is the time between data being added to the socket on MQTT client, it leaving the hardware, and arriving on the network/remote device?
>
> Can you make the esp device is [...]

ganglampschakelaar-boven:
```lua
21:27:35.504 CMD: NtpServer
21:27:35.512 MQT: stat/ganglampschakelaar-boven/RESULT = {"NtpServer1":"pool.ntp.org","NtpServer2":"nl.pool.ntp.org","NtpServer3":"0.nl.pool.ntp.org"}
21:30:38.400 RUL: SWITCH1#STATE performs "Publish cmnd/ganglamp/POWER toggle"
21:30:38.407 MQT: cmnd/ganglamp/POWER = toggle
21:32:29.368 RUL: SWITCH1#STATE performs "Publish cmnd/ganglamp/POWER toggle"
21:32:29.378 MQT: cmnd/ganglamp/POWER = toggle
```
ganglamp:
```lua
21:27:29.491 CMD: NtpServer
21:27:29.503 MQT: stat/ganglamp/RESULT = {"NtpServer1":"pool.ntp.org","NtpServer2":"nl.pool.ntp.org","NtpServer3":"0.nl.pool.ntp.org"}
21:30:38.498 MQT: stat/ganglamp/RESULT = {"POWER":"OFF"}
21:30:38.504 MQT: stat/ganglamp/POWER = OFF
21:32:29.505 MQT: stat/ganglamp/RESULT = {"POWER":"ON"}
21:32:29.511 MQT: stat/ganglamp/POWER = ON
```
There's a ~100ms delay

nagyrobi, 2021-02-28:
It looks like the MQTT broker server gets message late from Sonoff Mini...
What happens when you toggle from the web interface?

In my case `SetOption1 1` and `SetOption13 1` really helped, because these options disable detection of double/multi/long-click of the physical button. Don't know why this doesn't work.

GijsTimmers, 2021-02-28:
>  What happens when you toggle from the web interface?

There's no physical button on my web interface to do so, but issuing

    Publish cmnd/ganglamp/POWER toggle

from `ganglampschakelaar-boven` IS delayed. But we should probably keep in mind that this command is issued from a wifi-connected laptop which may create a delay by itself.

nagyrobi, 2021-03-01:
Try with this template:
https://templates.blakadder.com/sonoff_mini.html

nagyrobi, 2021-03-01:
Also you could try with the [Lite](http://ota.tasmota.com/tasmota/release/tasmota-lite.bin) version image.

GijsTimmers, 2021-03-01:
@nagyrobi Thank you for your continued support.

>  https://templates.blakadder.com/sonoff_mini.html

Thanks, I was actually running that template already.

GijsTimmers, 2021-03-03:
Flashing the Lite version image did not help. I flashed [esphome ](https://esphome.io/) on this device, configured it and it suffers from the same problem. I conclude that it's not Tasmota related. Thanks for all help though!

closed 2021-03-03 by GijsTimmers