arendst/Tasmota #11132: NTP from DHCP is not working on 9.3.1 (but is working on 8.5.0)

l3nticular, 2021-02-26:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
One Tasmota unit does not attempt to use the NTP server listed in the DHCP option.  Manually setting NTP server with "NTPServer1 10.70.70.1" works fine.

Today I set up a fresh D1 mini with Tasmota on my IoT VLAN.  The other 4 Tasmota devices on this VLAN get the NTP from DHCP just fine (these are all running 8.5.0).  I put 9.3.1 on the new device and it's not able to get the NTP server from the DHCP option.

DHCP+DNS server is dnsmasq running in a Docker container on a Pi (this is a standalone dnsmasq instance for the VLAN, not PiHole).

**NOTE**: After writing but before submitting this Issue, I completely erased the failing unit and flashed 8.5.0 with Tasmotizer.  I was able to see this function correctly work on this unit.  I then completely erased and flashed 9.3.1 and immediately the unit failed again (no configuration loaded, just WiFi SSID+Password manually entered).

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
   
There are old closed issues that seemed to have been due to malformed DHCP options.  Here it's working on one version of Tasmota and not the other.

- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): _____  D1 Mini 4MB
- [x] Tasmota binary firmware version number used: _____  Version 9.3.1(tasmota)-2_7_4_9(2021-02-23T12:02:38)
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: _____ tasmotizer.py (serial) 
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
08:55:11.738 MQT: stat/WaterRecirc/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
08:55:11.972 MQT: stat/WaterRecirc/RESULT = {"Module":{"1":"Sonoff Basic"}}
08:55:12.225 MQT: stat/WaterRecirc/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
08:57:40.636 MQT: stat/WaterRecirc/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
08:57:40.867 MQT: stat/WaterRecirc/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
08:57:41.119 MQT: stat/WaterRecirc/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
08:56:08.325 MQT: stat/WaterRecirc/STATUS = {"Status":{"Module":1,"DeviceName":"WaterRecircCtrl","FriendlyName":["Water Recirculation Control"],"Topic":"WaterRecirc","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
08:56:08.331 MQT: stat/WaterRecirc/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:22:25","StartupUTC":"2021-02-26T07:33:43","Sleep":50,"CfgHolder":4617,"BootCount":18,"BCResetTime":"2021-02-26T08:18:18","SaveCount":45,"SaveAddress":"F7000"}}
08:56:08.337 MQT: stat/WaterRecirc/STATUS2 = {"StatusFWR":{"Version":"9.3.1(tasmota)","BuildDateTime":"2021-02-23T12:02:38","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"398/699"}}
08:56:08.343 MQT: stat/WaterRecirc/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Smarties",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
08:56:08.352 MQT: stat/WaterRecirc/STATUS4 = {"StatusMEM":{"ProgramSize":593,"Free":408,"Heap":23,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
08:56:08.360 MQT: stat/WaterRecirc/STATUS5 = {"StatusNET":{"Hostname":"Water_Recirc_Ctrl","IPAddress":"10.70.70.15","Gateway":"10.70.70.1","Subnetmask":"255.255.255.0","DNSServer":"10.70.70.1","Mac":"5C:CF:7F:44:B7:18","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
08:56:08.366 MQT: stat/WaterRecirc/STATUS6 = {"StatusMQT":{"MqttHost":"berryblast.lan","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_44B718","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
08:56:08.373 MQT: stat/WaterRecirc/STATUS7 = {"StatusTIM":{"UTC":"2021-02-26T07:56:08","Local":"2021-02-26T08:56:08","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"07:37","Sunset":"18:28"}}
08:56:08.378 MQT: stat/WaterRecirc/STATUS10 = {"StatusSNS":{"Time":"2021-02-26T08:56:08"}}
08:56:08.384 MQT: stat/WaterRecirc/STATUS11 = {"StatusSTS":{"Time":"2021-02-26T08:56:08","Uptime":"0T00:22:25","UptimeSec":1345,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Smarties","BSSId":"D8:07:B6:20:EB:AA","Channel":1,"RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
00:00:00.050 CFG: Loaded from flash at F6, Count 38
00:00:00.055 QPC: Count 1
00:00:00.060 SRC: Restart
00:00:00.061 Project tasmota WaterRecircCtrl Version 9.3.1(tasmota)-2_7_4_9(2021-02-23T12:02:38)
00:00:00.212 WIF: Checking connection...
00:00:00.213 WIF: Attempting connection...
00:00:00.549 WIF: Connecting to AP1 Smarties Channel 1 BSSId D8:07:B6:20:EB:AA in mode 11n as Water_Recirc_Ctrl...
00:00:01.753 WIF: Checking connection...
00:00:01.754 WIF: Connected
00:00:02.005 HTP: Web server active on Water_Recirc_Ctrl with IP address 10.70.70.15
00:00:03.426 NTP: No reply
[... 63 lines cut ...]
dnsmasq    | dnsmasq: config error is REFUSED
dnsmasq    | dnsmasq: query[A] 0.nl.pool.ntp.org from 10.70.70.15
dnsmasq    | dnsmasq: config error is REFUSED
dnsmasq    | dnsmasq: query[A] pool.ntp.org from 10.70.70.15
dnsmasq    | dnsmasq: config error is REFUSED

```

From failing unit + dnsmasq log:
[failed.txt](https://github.com/arendst/Tasmota/files/6048204/failed.txt)

From working unit (v8.5.0) + dnsmasq log:
[working.txt](https://github.com/arendst/Tasmota/files/6048205/working.txt)

### TO REPRODUCE
Set up DHCP server with IP address of desired NTP server (which happens to be the same as the DHCP server).  Connect an 8.5.0 unit and a 9.3.1 unit to the network.  8.5.0 unit works, 9.3.1 unit doesn't.

### EXPECTED BEHAVIOUR
Unit should use the provided NTP server and not the FW defaults.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
See attached files

Jason2866, 2021-02-26:
Since of many problems (generated from routers which do not correctly provide the NTP server via DHCP)
Tasmota do NOT use this way anymore. NTP servers are set in Tasmota (only).
See https://github.com/arendst/Tasmota/blob/development/CHANGELOG.md#1022-20160814

ascillato2, 2021-02-26:
Closing this issue as it has been answered.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-02-26 by ascillato2

l3nticular, 2021-02-26:
Thanks!  Glad to know that it was intentional behavior.

nferch, 2021-11-21:
This seems to conflict with https://tasmota.github.io/docs/FAQ/#timers-trigger-at-the-wrong-time but matches the behavior I've been observing. Should the documentation be updated?

It'd also be nice to have a configuration setting to use the NTP server from the DHCP response, to avoid to hardcode the server in the configuration.

Jason2866, 2021-11-21:
@nferch  Thx! Outdated info removed.

totti4ever, 2022-02-11:
> Since of many problems (generated from routers which do not correctly provide the NTP server via DHCP)
> Tasmota do NOT use this way anymore. NTP servers are set in Tasmota (only).

Wow, this should be a sticky/FAQ somewhere? Was trying stuff since weeks to get this running :-D

Jason2866, 2022-02-11:
@totti4ever  We have Tasmota discussions for such stuff. Normally you get a answer in a few hours. If we stick any relevant info, we would have a forrest of sticked informations. The info is there. Changelog