arendst/Tasmota #11137: Dimmer value change incorrectly ignored

ericyanush, 2021-02-26:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Changing the dimmer value of a Tuya (TreatLife DS03) dimmer via MQTT is ignored on first change after switch is turned on.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): TreatLife DS03
- [ ] Tasmota binary firmware version number used: 9.3.1
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: OTA from earliler Tasmota
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

16:58:32.268 CMD: Backlog Template; Module; GPIO 255
16:58:32.271 SRC: WebConsole from 10.0.1.58
16:58:32.274 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
16:58:32.341 SRC: Backlog
16:58:32.344 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
16:58:32.353 MQT: stat/master_bedroom_fan/TEMPLATE = {"NAME":"DS03 Fan/Light","GPIO":[0,2272,0,2304,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":54}
16:58:32.548 SRC: Backlog
16:58:32.552 CMD: Group 0, Index 1, Command "MODULE", Data ""
16:58:32.561 MQT: stat/master_bedroom_fan/MODULE = {"Module":{"0":"DS03 Fan/Light"}}
16:58:32.758 SRC: Backlog
16:58:32.761 CMD: Group 0, Index 1, Command "GPIO", Data "255"
16:58:32.775 MQT: stat/master_bedroom_fan/GPIO = {"GPIO0":{"0":"None"},"GPIO1":{"2272":"Tuya Tx"},"GPIO2":{"0":"None"},"GPIO3":{"2304":"Tuya Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

17:00:24.875 CMD: Backlog Rule1; Rule2; Rule3
17:00:24.878 SRC: WebConsole from 10.0.1.58
17:00:24.881 CMD: Group 0, Index 1, Command "BACKLOG", Data "Rule1; Rule2; Rule3"
17:00:24.901 SRC: Backlog
17:00:24.904 CMD: Group 0, Index 1, Command "RULE", Data ""
17:00:24.912 MQT: stat/master_bedroom_fan/RULE = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
17:00:25.111 SRC: Backlog
17:00:25.115 CMD: Group 0, Index 2, Command "RULE", Data ""
17:00:25.123 MQT: stat/master_bedroom_fan/RULE = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
17:00:25.320 SRC: Backlog
17:00:25.323 CMD: Group 0, Index 3, Command "RULE", Data ""
17:00:25.332 MQT: stat/master_bedroom_fan/RULE = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

17:01:51.506 CMD: STATUS 0
17:01:51.509 SRC: WebConsole from 10.0.1.58
17:01:51.512 CMD: Group 0, Index 1, Command "STATUS", Data "0"
17:01:51.519 MQT: stat/master_bedroom_fan/STATUS = {"Status":{"Module":0,"DeviceName":"masterbedroom_fan","FriendlyName":["masterbedroom_fan","Tasmota2"],"Topic":"master_bedroom_fan","ButtonTopic":"0","Power":3,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":0,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
17:01:51.529 MQT: stat/master_bedroom_fan/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:30:30","StartupUTC":"2021-02-26T15:31:21","Sleep":50,"CfgHolder":4617,"BootCount":10,"BCResetTime":"2021-02-02T04:18:37","SaveCount":555,"SaveAddress":"F4000"}}
17:01:51.538 MQT: stat/master_bedroom_fan/STATUS2 = {"StatusFWR":{"Version":"9.3.1(tasmota)","BuildDateTime":"2021-02-23T12:02:38","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"427/699"}}
17:01:51.546 MQT: stat/master_bedroom_fan/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Searching...",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00108018","2805C8000100060000005A0A000000000000","00000090","0000E000","00000000"]}}
17:01:51.559 MQT: stat/master_bedroom_fan/STATUS4 = {"StatusMEM":{"ProgramSize":593,"Free":408,"Heap":24,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
17:01:51.569 MQT: stat/master_bedroom_fan/STATUS5 = {"StatusNET":{"Hostname":"master_bedroom_fan-2980","IPAddress":"10.0.1.89","Gateway":"10.0.1.1","Subnetmask":"255.255.255.0","DNSServer":"10.0.1.1","Mac":"48:3F:DA:83:CB:A4","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
17:01:51.576 MQT: stat/master_bedroom_fan/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.1.100","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_83CBA4","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
17:01:51.585 MQT: stat/master_bedroom_fan/STATUS7 = {"StatusTIM":{"UTC":"2021-02-26T16:01:51","Local":"2021-02-26T17:01:51","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"07:37","Sunset":"18:28"}}
17:01:51.594 MQT: stat/master_bedroom_fan/STATUS10 = {"StatusSNS":{"Time":"2021-02-26T17:01:51"}}
[... 93 lines cut ...]
17:03:15.480 {"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}
17:03:15.483 TYA: Heartbeat
17:03:15.576 CFG: Saved to flash at F7, Count 560, Bytes 4096

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

1. Change dimmer value with buttons on switch
2. Change dimmer value with MQTT message 

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

The dimmer value will be updated from the command sent via MQTT

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
The timing of the MQTT message in relation to the last button press event does not seem to matter. The same behaviour is observed if the MQTT message within a few seconds of the last switch event or a after several minutes.  

**(Please, remember to close the issue when the problem has been addressed)**

jaymemaurice, 2021-02-26:
"Send dim skipped value 500 for dpid 10"

Can you please try my build in https://github.com/arendst/Tasmota/pull/11121 or > 9.3.1.2

ericyanush, 2021-02-26:
@jamesturton I flashed one of my switches to the firmware you uploaded in discussion of #11121, and it seems to have fixed the issue. 🎉 

As I see #11121 has been merged into the dev branch, I assume we will see this fix in an upcoming release, so I think we are good to close this issue. Thanks for the super quick response!

jaymemaurice, 2021-02-26:
@ericyanush no problem :) glad to help

closed 2021-02-26 by ascillato2