arendst/Tasmota #11143: TuyaMCU 71 is showing the wrong value

Sthopeless, 2021-02-27:
### PROBLEM DESCRIPTION
My [Tuya Kettle](https://templates.blakadder.com/proficook_PC-WKS_1167.html) as a DPID for the Temperature on which I use `TuyaMCU 71` https://tasmota.github.io/docs/TuyaMCU/#sensors

On Tasmota v9.2.0 and before it would display the right temperature:
```
15:16:54 MQT: tele/Kettle/RESULT = {"TuyaReceived":{"Data":"55AA03070008690200040000004BCB","Cmnd":7,"CmndData":"690200040000004B","DpType2Id105":75,"105":{"DpId":105,"DpIdType":2,"DpIdData":"0000004B"}}}

15:16:54 MQT: tele/Kettle/SENSOR = {"TuyaSNS":{"Temperature":75.0}}
```

But on Tasmota v9.3.0 started to be:
```
15:01:25.341 MQT: tele/Kettle/RESULT = {"TuyaReceived":{"Data":"55AA03070008690200040000004BCB","Cmnd":7,"CmndData":"690200040000004B","DpType2Id105":75,"105":{"DpId":105,"DpIdType":2,"DpIdData":"0000004B"}}}

15:01:25.347 MQT: tele/Kettle/SENSOR = {"TuyaSNS":{"Temperature":7.5}}
```

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

jaymemaurice, 2021-02-27:
@Sthopeless 
This was a result of a breaking change to fix other temperature sensors. https://github.com/arendst/Tasmota/pull/10922

Use the console command TempRes 0 and the kettle will now report 75*C
If you change to TempRes 1, it will report 7.5. TempRes 2, 0.75 etc.
I assume, before your kettle would go from 75.0 to 76.0 instead of 75 to 76...

jaymemaurice, 2021-02-27:
Also, I will work on docs to document this better.

Sthopeless, 2021-02-27:
Thank you @jay-wilkinson `TempRes 0` did indeed work! 

Tbh never checked if it did 75.1, 75.2 or if jumped directly to 76 not very important for this device.

closed 2021-02-27 by Sthopeless

yvesdm3000, 2021-03-01:
My Beca BHT-002 doesn't need a divide by 10 but a divide by 2... So changing the value using TempRes like here doesn't solve this problem...