arendst/Tasmota #11144: Battery info of Temp sensor lost after setting friendly name Zigbee2Tasmota SONOFF Bridge

sincze, 2021-02-27:
### PROBLEM DESCRIPTION
Missing values in MQTT report of a Xiaomi Temp/Humid/Bar sensor once friendly name has been set.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Sonoff Zigbee Gateway flashed with Tasmota / Xiaomi Temp / Humi / Bar sensor
- [X] Tasmota binary firmware version number used: 9.3.0
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: n.a.
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
19:25:35.372 HTP: Main Menu
19:26:03.927 HTP: Console
19:26:07.732 CMD: Backlog Template; Module; GPIO 255
19:26:07.733 SRC: WebConsole from 192.168.2.10
19:26:07.735 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
19:26:07.798 SRC: Backlog
19:26:07.800 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
19:26:07.804 MQT: stat/tasmota_zigbee_bridge/RESULT = {"NAME":"Sonoff ZBBridge","GPIO":[320,3552,0,3584,5312,0,0,0,0,576,0,0,32,0],"FLAG":0,"BASE":18}
19:26:08.007 SRC: Backlog
19:26:08.009 CMD: Group 0, Index 1, Command "MODULE", Data ""
19:26:08.013 MQT: stat/tasmota_zigbee_bridge/RESULT = {"Module":{"0":"Sonoff ZBBridge"}}
19:26:08.257 SRC: Backlog
19:26:08.259 CMD: Group 0, Index 1, Command "GPIO", Data "255"
19:26:08.266 MQT: stat/tasmota_zigbee_bridge/RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"3552":"Zigbee Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3584":"Zigbee Rx"},"GPIO4":{"5312":"Zigbee Rst"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"32":"Button1"},"GPIO17":{"0":"None"}}
19:26:14.734 MQT: tele/tasmota_zigbee_bridge/STATE = {"Time":"2021-02-27T19:26:14","Uptime":"0T04:10:09","UptimeSec":15009,"Vcc":3.536,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"Wifi":{"AP":1,"SSId":"dd-wrt-iot","BSSId":"78:8A:20:54:33:D9","Channel":1,"RSSI":100,"Signal":-39,"LinkCount":2,"Downtime":"0T00:00:09"}}

```  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
19:28:11.917 CMD: Backlog Rule1; Rule2; Rule3
19:28:11.919 SRC: WebConsole from 192.*.*.*
19:28:11.921 CMD: Group 0, Index 1, Command "BACKLOG", Data "Rule1; Rule2; Rule3"
19:28:11.992 SRC: Backlog
19:28:11.994 CMD: Group 0, Index 1, Command "RULE", Data ""
19:28:11.999 MQT: stat/tasmota_zigbee_bridge/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
19:28:12.202 SRC: Backlog
19:28:12.204 CMD: Group 0, Index 2, Command "RULE", Data ""
19:28:12.209 MQT: stat/tasmota_zigbee_bridge/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
19:28:12.451 SRC: Backlog
19:28:12.453 CMD: Group 0, Index 3, Command "RULE", Data ""
19:28:12.457 MQT: stat/tasmota_zigbee_bridge/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here: 19:27:28.023 CMD: STATUS 0
19:27:28.025 SRC: WebConsole from 192.168.2.10
19:27:28.027 CMD: Group 0, Index 1, Command "STATUS", Data "0"
19:27:28.031 MQT: stat/tasmota_zigbee_bridge/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_zigbee_bridge","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
19:27:28.036 MQT: stat/tasmota_zigbee_bridge/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-zbbridge.bin","RestartReason":"Power On","Uptime":"0T04:11:23","StartupUTC":"2021-02-27T14:16:05","Sleep":50,"CfgHolder":4617,"BootCount":20,"BCResetTime":"2020-08-22T14:26:14","SaveCount":59,"SaveAddress":"F8000"}}
19:27:28.041 MQT: stat/tasmota_zigbee_bridge/STATUS2 = {"StatusFWR":{"Version":"9.3.0(zbbridge)","BuildDateTime":"2021-02-19T11:10:58","Boot":7,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":160,"Hardware":"ESP8266EX","CR":"451/699"}}
19:27:28.046 MQT: stat/tasmota_zigbee_bridge/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":3,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["dd-wrt-iot",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000080","00046002","00000000"]}}
19:27:28.053 MQT: stat/tasmota_zigbee_bridge/STATUS4 = {"StatusMEM":{"ProgramSize":573,"Free":1216,"Heap":26,"ProgramFlashSize":2048,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F1007C4","04400001","00000002","00000000","00000000","00020000","00000000","00000000"],"Drivers":"1,2,7,9,10,20,23,41","Sensors":""}}
19:27:28.058 MQT: stat/tasmota_zigbee_bridge/STATUS5 = {"StatusNET":{"Hostname":"tasmota_zigbee_bridge","IPAddress":"192.*.*.*","Gateway":"192.*.*.*","Subnetmask":"255.255.255.0","DNSServer":"192.*.*.*","Mac":"**:**:**:**:**:**","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
19:27:28.062 MQT: stat/tasmota_zigbee_bridge/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.***.lan","MqttPort":****,"MqttClientMask":"DVES_****","MqttClient":"****","MqttUser":"****","MqttCount":2,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
19:27:28.068 MQT: stat/tasmota_zigbee_bridge/STATUS7 = {"StatusTIM":{"UTC":"2021-02-27T18:27:28","Local":"2021-02-27T19:27:28","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"07:35","Sunset":"18:29"}}
19:27:28.074 MQT: stat/tasmota_zigbee_bridge/STATUS10 = {"StatusSNS":{"Time":"2021-02-27T19:27:28"}}
19:27:28.081 MQT: stat/tasmota_zigbee_bridge/STATUS11 = {"StatusSTS":{"Time":"2021-02-27T19:27:28","Uptime":"0T04:11:23","UptimeSec":15083,"Vcc":3.523,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"Wifi":{"AP":1,"SSId":"dd-wrt-iot","BSSId":"78:8A:20:54:33:D9","Channel":1,"RSSI":100,"Signal":-39,"LinkCount":2,"Downtime":"0T00:00:09"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Wait for the MQTT message to be published.
Friendly named device TH7_Tuin is missing 2 fields in the MQTT output BatteryPercentage , BatteryVoltage, compared to the non friendly name device. 0x4924

18:44:05.440 MQT: tele/tasmota_zigbee_bridge/SENSOR = {"TH7_Tuin":{"Device":"0x1D46","Name":"TH7_Tuin","Temperature":5.29,"Humidity":76.23,"Pressure":1045,"PressureScale":-1,"PressureScaledValue":10459,"SeaPressure":1046,"Endpoint":1,"LinkQuality":68}}
18:44:10.440 MQT: tele/tasmota_zigbee_bridge/SENSOR = {"TH7_Tuin":{"Device":"0x1D46","Name":"TH7_Tuin","Temperature":4.76,"Humidity":77.42,"Pressure":1045,"PressureScale":-1,"PressureScaledValue":10459,"SeaPressure":1046,"Endpoint":1,"LinkQuality":68}}
18:46:42.313 MQT: tele/tasmota_zigbee_bridge/SENSOR = {"0x4924":{"Device":"0x4924","**BatteryVoltage**":3.04,"**BatteryPercentage**":100,"Temperature":19.06,"Humidity":46.19,"Pressure":1040,"SeaPressure":1041,"Endpoint":1,"LinkQuality":45}}

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
---> 
{"TH7_Tuin":{"Device":"0x1D46","Name":"TH7_Tuin","BatteryVoltage":3.04,"BatteryPercentage":100,"Temperature":5.29,"Humidity":76.23,"Pressure":1045,"PressureScale":-1,"PressureScaledValue":10459,"SeaPressure":1046,"Endpoint":1,"LinkQuality":68}}

So including BatteryVoltage BatteryPercentage

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

nagyrobi, 2021-03-02:
Same here with a [TS0201](https://zigbee.blakadder.com/Tuya_TS0201.html) :

```
21:55:47.706 MQT: tele/zigbee1/888C/SENSOR = {"0x888C":{"Device":"0x888C","Name":"Hőmérő","Temperature":21.5,"Endpoint":1,"LinkQuality":21}} (retained)
21:55:48.234 MQT: tele/zigbee1/888C/SENSOR = {"0x888C":{"Device":"0x888C","Name":"Hőmérő","Humidity":43.6,"Endpoint":1,"LinkQuality":18}} (retained)
21:55:48.777 MQT: tele/zigbee1/888C/SENSOR = {"0x888C":{"Device":"0x888C","Name":"Hőmérő","BatteryVoltage":3,"BatteryPercentage":100,"Endpoint":1,"LinkQuality":24}} (retained)
```

mister-gefahrensucher, 2021-03-20:
Same her on my Bridge.
I,m using Tasmots 9.3.1. After i added a friendly name for my thermostat then 

before 
sonoffzigbee/tele/SENSOR = {"ZbReceived":{"0xF54E":{"Device":"0xF54E","BatteryVoltage":2.6,"BatteryPercentage":13,"Endpoint":3,"LinkQuality":5}}}

also i lost
sonoffzigbee/tele/SENSOR = {"ZbReceived":{"0xF54E":{"Device":"0xF54E","LocalTemperature":18.84,"PIHeatingDemand":100,"Endpoint":3,"LinkQuality":0}}}

after i named my device "eCozy" the information are gone.
sonoffzigbee/tele/SENSOR = {"ZbReceived":{"eCozy":{"Device":"0xF54E","Name":"eCozy","Endpoint":3,"LinkQuality":5}}}

nagyrobi, 2021-03-20:
Same here with Ikea symfonisk knob

sincze, 2021-04-05:
Unsure if it has something to do with the name. 
I had to start over with my bridge (now on 9.3.1) as it was not working anymore (blue light on and green led blinking. did not get IP)

After reflashing:
I notice 2 types of messages for a xiaomi temp/humid/baro sensor.
`22:03:06.553 MQT: tele/tasmota_zigbee_bridge/SENSOR = {"Device":"0x1D46","BatteryVoltage":2.99,"BatteryPercentage":92,"Temperature":2.12,"Humidity":87.46,"Pressure":1021,"SeaPressure":1021,"Endpoint":1,"LinkQuality":63}
22:10:21.568 MQT: tele/tasmota_zigbee_bridge/SENSOR = {"Device":"0x1D46","Temperature":2.23,"Humidity":88.48,"Pressure":1020,"PressureScale":-1,"PressureScaledValue":10209,"SeaPressure":1020,"Endpoint":1,"LinkQuality":63}
`

So one message has battery and the other one has not.

rebel7580, 2021-05-02:
@s-hadinger 
I have an Aqara temperture/humidity/pressure sensor (ModelId: lumi.weather). It doesn't always report battery level/percentage. This causes gaps in Home Assistant's battery sensor. Battery values are active after a periodic report only until a delta report arrives, at which time the battery percentage disappears (since it wasn't in the reported json.) This may be a problem with how HA's battery sensor is set up, not being able to bridge the gaps, but I'm not that versed in HA yet to figure that out.

In any case, it would be nice if at least battery percentage was reported in delta reports.

Periodic (timed) responses include T/H/P and battery level/percentage:

```
13:38:33.989 MQT: tele/tasmota_zbBridge/52B1/ZbReceived = {"Device":"0x52B1","Name":"Outside_Temp","BatteryVoltage":3.13,"BatteryPercentage":100,"Temperature":20.16,"Humidity":41.35,"Pressure":1008,"SeaPressure":1008,"Endpoint":1,"LinkQuality":60}
```

However, responses due to temperature changes only report T/H/P values but don't report battery level/percentage:

```
13:42:46.831 MQT: tele/tasmota_zbBridge/52B1/ZbReceived = {"Device":"0x52B1","Name":"Outside_Temp","Temperature":21.5,"Humidity":48.01,"Pressure":1007,"PressureScale":-1,"PressureScaledValue":10075,"SeaPressure":1007,"Endpoint":1,"LinkQuality":63}
```

ZbInfo responses report T?H/P and battery percentage but not battery level:

```
13:43:58.112 MQT: tele/tasmota_zbBridge/52B1/ZbInfo = {"Device":"0x52B1","Name":"Outside_Temp","IEEEAddr":"0x00158D00067BD906","ModelId":"lumi.weather","Manufacturer":"LUMI","Endpoints":[1],"Config":["T01"],"Temperature":21.5,"Pressure":1007,"Humidity":48.01,"Reachable":true,"BatteryPercentage":100,"LastSeen":72,"LastSeenEpoch":1619977366,"LinkQuality":63}
```

s-hadinger, 2021-05-03:
I have no control on how often the Aqara reports battery level. Tasmota reports to MQTT what it received from Zigbee.

Any high-level management or calculation of data should be done at gateway level. They have immensely more memory and compute capacity to do such transformations.

closed 2021-05-08 by github-actions[bot]