arendst/Tasmota #11152: ESP32 - Miflora - Wrong config in MQTT for Home Assistant #11151

MrFloppy, 2021-02-28:
### PROBLEM DESCRIPTION
I just flashed Tasmota Bluetooth to an ESP32 to bridge the data from BLE sensors via MQTT into my homeassistant. Unfortunately the Home Assistant Tasmota Plugin does not recognise the sensors properly, so I activated Home Assistant discovery ("SetOption19 1").
My LYWSD03MMC / ATC were properly recognised and configured. 

Unfortunately my MiFlora Sticks get the following config, but still reports correct values afterwards:

```lua
09:14:25.810 MQT: homeassistant/sensor/Flora676d23/Temperature/config = {"availability":[],"device":{"identifiers":["BLEFlora676d23"],"name":"Flora676d23","manufacturer":"tas","model":"Flora","via_device":"M5Stick-Bridge-4520"},"dev_cla":"temperature","expire_after":600,"json_attr_t":"tele/tasmota_ble/Flora676d23","name":"Flora676d23_Temperature","state_topic":"tele/tasmota_ble/Flora676d23","uniq_id":"Flora676d23_Temperature","unit_of_meas":"°C","val_tpl":"{{ value_json.Temperature }}"}
09:14:25.849 MQT: homeassistant/sensor/Flora676d23/Humidity/config = {"availability":[],"device":{"identifiers":["BLEFlora676d23"],"name":"Flora676d23","manufacturer":"tas","model":"Flora","via_device":"M5Stick-Bridge-4520"},"dev_cla":"humidity","expire_after":600,"json_attr_t":"tele/tasmota_ble/Flora676d23","name":"Flora676d23_Humidity","state_topic":"tele/tasmota_ble/Flora676d23","uniq_id":"Flora676d23_Humidity","unit_of_meas":"%","val_tpl":"{{ value_json.Humidity }}"}
09:14:25.890 MQT: homeassistant/sensor/Flora676d23/DewPoint/config = {"availability":[],"device":{"identifiers":["BLEFlora676d23"],"name":"Flora676d23","manufacturer":"tas","model":"Flora","via_device":"M5Stick-Bridge-4520"},"dev_cla":"temperature","expire_after":600,"json_attr_t":"tele/tasmota_ble/Flora676d23","name":"Flora676d23_DewPoint","state_topic":"tele/tasmota_ble/Flora676d23","uniq_id":"Flora676d23_DewPoint","unit_of_meas":"°C","val_tpl":"{{ value_json.DewPoint }}"}
09:14:25.933 MQT: homeassistant/sensor/Flora676d23/Battery/config = {"availability":[],"device":{"identifiers":["BLEFlora676d23"],"name":"Flora676d23","manufacturer":"tas","model":"Flora","via_device":"M5Stick-Bridge-4520"},"dev_cla":"battery","expire_after":600,"json_attr_t":"tele/tasmota_ble/Flora676d23","name":"Flora676d23_Battery","state_topic":"tele/tasmota_ble/Flora676d23","uniq_id":"Flora676d23_Battery","unit_of_meas":"%","val_tpl":"{{ value_json.Battery }}"}
09:14:25.972 MQT: homeassistant/sensor/Flora676d23/RSSI/config = {"availability":[],"device":{"identifiers":["BLEFlora676d23"],"name":"Flora676d23","manufacturer":"tas","model":"Flora","via_device":"M5Stick-Bridge-4520"},"dev_cla":"signal_strength","expire_after":600,"json_attr_t":"tele/tasmota_ble/Flora676d23","name":"Flora676d23_RSSI","state_topic":"tele/tasmota_ble/Flora676d23","uniq_id":"Flora676d23_RSSI","unit_of_meas":"dB","val_tpl":"{{ value_json.RSSI }}"}
```

Is this my fault?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): M5Stick-C
- [X] Tasmota binary firmware version number used: 9.2.1
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: Espressif
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
09:21:39.211 MQT: stat/M5Stick-Bridge/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
09:21:39.430 MQT: stat/M5Stick-Bridge/RESULT = {"Module":{"1":"ESP32-DevKit"}}
09:21:39.691 MQT: stat/M5Stick-Bridge/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"768":"SPI CS"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"736":"SPI CLK"},"GPIO14":{"0":"None"},"GPIO15":{"704":"SPI MOSI"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"640":"I2C SDA"},"GPIO22":{"608":"I2C SCL"},"GPIO23":{"800":"SPI DC"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"}}
09:21:39.744 MQT: stat/M5Stick-Bridge/RESULT = {"GPIO28":{"0":"None"},"GPIO29":{"0":"None"},"GPIO30":{"0":"None"},"GPIO31":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
None

```
- [X] Provide the output of this command: `Status 0`:
```lua
09:22:12.194 MQT: stat/M5Stick-Bridge/STATUS = {"Status":{"Module":1,"DeviceName":"M5Stick-Bridge","FriendlyName":["M5Stick-Bridge"],"Topic":"M5Stick-Bridge","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
09:22:12.229 MQT: stat/M5Stick-Bridge/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:57:52","StartupUTC":"2021-02-28T07:24:20","Sleep":50,"CfgHolder":4617,"BootCount":24,"BCResetTime":"2021-02-27T08:42:36","SaveCount":56}}
09:22:12.261 MQT: stat/M5Stick-Bridge/STATUS2 = {"StatusFWR":{"Version":"9.3.1(bluetooth)","BuildDateTime":"2021-02-23T12:22:22","Core":"1_0_5_RC6","SDK":"v3.3.4-402-gcd59d107b","CpuFrequency":80,"Hardware":"ESP32","CR":"455/699"}}
09:22:12.281 MQT: stat/M5Stick-Bridge/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["LAN-Guste",""],"TelePeriod":600,"Resolution":"558180C0","SetOption":["000A8009","2805C8000100060000005A0A000000000000","00000280","00006000","00000002"]}}
09:22:12.315 MQT: stat/M5Stick-Bridge/STATUS4 = {"StatusMEM":{"ProgramSize":1461,"Free":1984,"Heap":123,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":80,"FlashMode":2,"Features":["00000809","8FDAC787","00140001","000000CF","010013C0","40000981","00004000","00200000","00000000"],"Drivers":"1,2,3,4,5,7,8,9,10,12,20,21,24,26,27,29,50,79","Sensors":"1,2,3,5,6,52,62"}}
09:22:12.352 MQT: stat/M5Stick-Bridge/STATUS5 = {"StatusNET":{"Hostname":"M5Stick-Bridge-4520","IPAddress":"192.168.1.55","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.58","Mac":"50:02:91:8D:11:A8","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
09:22:12.373 MQT: stat/M5Stick-Bridge/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.37","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_8D11A8","MqttUser":"zigbee2mqtt","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
09:22:12.394 MQT: stat/M5Stick-Bridge/STATUS7 = {"StatusTIM":{"UTC":"2021-02-28T08:22:12","Local":"2021-02-28T09:22:12","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"07:33","Sunset":"18:31"}}
09:22:12.413 MQT: stat/M5Stick-Bridge/STATUS10 = {"StatusSNS":{"Time":"2021-02-28T09:22:12"}}
09:22:12.425 MQT: stat/M5Stick-Bridge/STATUS11 = {"StatusSTS":{"Time":"2021-02-28T09:22:12","Uptime":"0T00:57:52","UptimeSec":3472,"Heap":121,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"LAN-Guste","BSSId":"2C:91:AB:3A:F5:5C","Channel":1,"RSSI":50,"Signal":-75,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```

### TO REPRODUCE
See above.

### EXPECTED BEHAVIOUR
See above.

### SCREENSHOTS
N/A

### ADDITIONAL CONTEXT
None yet

**(Please, remember to close the issue when the problem has been addressed)**

Tediore, 2021-03-28:
I'm experiencing the same issue. `SO19 1` produces most of the expected entities (battery, temp, humidity [moisture], and RSSI), but humidity is always 0% even though Tasmota shows the expected value. In addition, an entity for illuminance is not created.

If I understand correctly, HA autodiscovery (`SO19 1`) for Tasmota is no longer being developed now that there is the Tasmota HA integration, so perhaps these issues will be addressed when Tasmota autodiscovery is (hopefully) updated to support the Mi/ATC bluetooth thermometer and MiFlora entities?

MrFloppy, 2021-04-01:
@Tediore: Good point. Maybe I should address it to the Tasmota Integration Developers. I just switched it off in this case, because the integration is not doing well, when you solder your own ESP8266 and sensors. It just does not do the job as well as the HA discovery in Tasmota did until this problem occured.

conrad784, 2021-04-24:
I am also using some ATC + Miflora devices in my environment and had this problem.

I just upgraded to 9.4.0 and now old-style MQTT-autodiscovery worked with HA. There was no definite entry in the changelog, but they apparently introduced new ways of Tasmota-Autodiscovery which might have changed all other auto-discovery implementations as well?

closed 2021-05-24 by github-actions[bot]