arendst/Tasmota #11153: Tasmota-bluetooh: missing command "ping"

demel42, 2021-02-28:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP32-DevKit
- [ ] Tasmota binary firmware version number used: 9.3.1(bluetooth) - 2021-02-23T12:22:22
  - [ ] Pre-compiled
  - [x] Self-compiled
- [X] Flashing tools used: esptool.py
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
10:48:37.405 CMD: Backlog Template; Module; GPIO 255
10:48:37.407 SRC: WebConsole from 192.168.178.34
10:48:37.409 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
10:48:37.479 SRC: Backlog
10:48:37.481 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
10:48:37.497 MQT: stat/tasmotaBT_Wall/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
10:48:37.730 SRC: Backlog
10:48:37.732 CMD: Group 0, Index 1, Command "MODULE", Data ""
10:48:37.742 MQT: stat/tasmotaBT_Wall/RESULT = {"Module":{"1":"ESP32-DevKit"}}
10:48:37.980 SRC: Backlog
10:48:37.982 CMD: Group 0, Index 1, Command "GPIO", Data "255"
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
10:49:38.750 SRC: WebConsole from 192.168.178.34
10:49:38.752 CMD: Group 0, Index 1, Command "STATUS", Data "0"
10:49:38.761 MQT: stat/tasmotaBT_Wall/STATUS = {"Status":{"Module":1,"DeviceName":"TasmotaBT-Wall","FriendlyName":["TasmotaBT-Wall"],"Topic":"tasmotaBT_Wall","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
10:49:38.796 MQT: stat/tasmotaBT_Wall/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32-bluetooth.bin","RestartReason":"Software reset CPU","Uptime":"0T18:16:41","StartupUTC":"2021-02-27T15:32:57","Sleep":50,"CfgHolder":4617,"BootCount":27,"BCResetTime":"2021-02-24T17:26:27","SaveCount":53}}
10:49:38.829 MQT: stat/tasmotaBT_Wall/STATUS2 = {"StatusFWR":{"Version":"9.3.1(bluetooth)","BuildDateTime":"2021-02-23T12:22:22","Core":"1_0_5_RC6","SDK":"v3.3.4-402-gcd59d107b","CpuFrequency":80,"Hardware":"ESP32","CR":"487/699"}}
10:49:38.849 MQT: stat/tasmotaBT_Wall/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":3,"LogHost":"192.168.178.91","LogPort":514,"SSId":["cd-wlan",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000081","00006000","00000002"]}}
10:49:38.892 MQT: stat/tasmotaBT_Wall/STATUS4 = {"StatusMEM":{"ProgramSize":1461,"Free":1984,"Heap":117,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","00140001","000000CF","010013C0","40000981","00004000","00200000","00000000"],"Drivers":"1,2,3,4,5,7,8,9,10,12,20,21,24,26,27,29,50,79","Sensors":"1,2,3,5,6,52,62"}}
10:49:38.925 MQT: stat/tasmotaBT_Wall/STATUS5 = {"StatusNET":{"Hostname":"tasmotaBT-Wall","IPAddress":"192.168.178.201","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.54","Mac":"3C:61:05:13:FF:44","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
10:49:38.949 MQT: stat/tasmotaBT_Wall/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.68","MqttPort":1024,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_13FF44","MqttUser":"mqtt4ips","MqttCount":2,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
10:49:38.971 MQT: stat/tasmotaBT_Wall/STATUS7 = {"StatusTIM":{"UTC":"2021-02-28T09:49:38","Local":"2021-02-28T10:49:38","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"07:33","Sunset":"18:31"}}
10:49:38.991 MQT: stat/tasmotaBT_Wall/STATUS10 = {"StatusSNS":{"Time":"2021-02-28T10:49:38"}}
10:49:39.007 MQT: stat/tasmotaBT_Wall/STATUS11 = {"StatusSTS":{"Time":"2021-02-28T10:49:38","Uptime":"0T18:16:41","UptimeSec":65801,"Heap":118,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"Wifi":{"AP":1,"SSId":"cd-wlan","BSSId":"00:11:32:A5:4E:78","Channel":11,"RSSI":52,"Signal":-74,"LinkCount":2,"Downtime":"0T00:00:09"}}}
10:49:42.541 WIF: Checking connection...
```

- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
10:50:42.946 CMD: ping 192.168.178.1
10:50:42.948 SRC: WebConsole from 192.168.178.34
10:50:42.950 CMD: Group 0, Index 1, Command "PING", Data "192.168.178.1"
10:50:42.963 MQT: stat/tasmotaBT_Wall/RESULT = {"Command":"Unknown"}
```

### TO REPRODUCE
use PING with any IP address 

### EXPECTED BEHAVIOUR
the result, described in the doc (https://tasmota.github.io/docs/Commands/)

### SCREENSHOTS
i wand to reboot the ESP when not connected to a WLAN for some time and would make a rule like this (untested, because the described error occurred on the web-console)

```lua
Rule2
  ON system#boot do Var1 0 ENDON

  ON Time#Minute|5 DO Ping4 192.168.178.1 ENDON
  ON Ping#192.168.178.1#Success==0 DO Add1 1 ENDON
  ON Ping#192.168.178.1#Success>0 DO Var1 0 ENDON
  ON Var1#State>3 DO Reset 1 ENDON
Rule2 1
```

My Problem: under some circumstances (i.e. after a power-break) the Wifi-MESH is not up when the ESP is ready. the ESP do (so my observations) don't reconnect automatically

my be, that the precompiled binary has no `#define USE_PING`

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2021-02-28:
Hi, 

In the file https://github.com/arendst/Tasmota/blob/development/BUILDS.md is shown which features are included on each build. Ping is not included on any build. You have to compile by yourself. You can check the docs for that.

If you need further assistance, please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md)

Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-02-28 by ascillato2

demel42, 2021-02-28:
hi,

thanks for your help and your time