arendst/Tasmota #11158: SONOFF F2X(8) - DS18B20 sensor not working after FW 9.3.1

PetrJakes, 2021-02-28:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): SONOFF F20EU Type F
- [x] Tasmota binary firmware version number used: 9.3.1
  - [x] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
17:39:40.413 CMD: Backlog Template; Module; GPIO 255
17:39:40.433 MQT: stat/pristresek/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
17:39:40.662 MQT: stat/pristresek/RESULT = {"Module":{"8":"Sonoff S2X"}}
17:39:40.917 MQT: stat/pristresek/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"1312":"DS18x20"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```

- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
17:42:22.390 CMD: Status 0
17:42:22.398 MQT: stat/pristresek/STATUS = {"Status":{"Module":8,"DeviceName":"Přístřešek","FriendlyName":["Tasmota"],"Topic":"pristresek","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
17:42:22.404 MQT: stat/pristresek/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T02:03:39","StartupUTC":"2021-02-28T14:38:43","Sleep":50,"CfgHolder":4617,"BootCount":26,"BCResetTime":"2020-12-30T23:38:18","SaveCount":109,"SaveAddress":"FB000"}}
17:42:22.409 MQT: stat/pristresek/STATUS2 = {"StatusFWR":{"Version":"9.3.1(tasmota)","BuildDateTime":"2021-02-23T12:02:38","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"403/699"}}
17:42:22.413 MQT: stat/pristresek/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["FUCK_MUSLIMS",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
17:42:22.422 MQT: stat/pristresek/STATUS4 = {"StatusMEM":{"ProgramSize":593,"Free":408,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
17:42:22.429 MQT: stat/pristresek/STATUS5 = {"StatusNET":{"Hostname":"pristresek-5282","IPAddress":"192.168.0.65","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"213.46.172.38","Mac":"84:F3:EB:B0:B4:A2","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
17:42:22.435 MQT: stat/pristresek/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.210","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_B0B4A2","MqttUser":"foerster","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
17:42:22.443 MQT: stat/pristresek/STATUS7 = {"StatusTIM":{"UTC":"2021-02-28T16:42:22","Local":"2021-02-28T17:42:22","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"07:33","Sunset":"18:31"}}
17:42:22.447 MQT: stat/pristresek/STATUS10 = {"StatusSNS":{"Time":"2021-02-28T17:42:22"}}
17:42:22.453 MQT: stat/pristresek/STATUS11 = {"StatusSTS":{"Time":"2021-02-28T17:42:22","Uptime":"0T02:03:39","UptimeSec":7419,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"FUCK_MUSLIMS","BSSId":"90:F6:52:85:D0:22","Channel":11,"RSSI":56,"Signal":-72,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
DS18B20 connected to the GPIO3 with Tasmota 9.2.0 working.
After upgrade to 9.3.1 stopped to work

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
DS18B20 Temperature text and value on the Tasmota main screen.
--

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2021-02-28:
Some users have experienced better results with tasmota-sensors.bin.gz - reasons for this not clear.

ascillato2, 2021-02-28:
Closing this issue as it is **duplicated**. Sorry.

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-02-28 by ascillato2

PetrJakes, 2021-02-28:
didn't find the duplicate even trying to find it hard before opening the
new issue.
When closed because of the "duplicate", it can be nice to give the link to
the previously accepted issue ..

ascillato, 2021-02-28:
https://github.com/arendst/Tasmota/issues/11117