arendst/Tasmota #11178: Sonoff v1.1 ZbBridge Module restarts every 16s

Haroldas-sys, 2021-03-01:
### PROBLEM DESCRIPTION
Sonoff v1.1 ZbBridge Module restarts every 16s, if program module type are set to Sonoff ZbBridghe (75)

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Sonoff v1.1 ZbBridge
- [ ] Tasmota binary firmware version number used: 9.3.1.1
  - [ ] Pre-compiled http://ota.tasmota.com/tasmota/release/tasmota-zbbridge.bin
  - [ ] Self-compiled
- [ ] Flashing tools used: Tamotizer 1.2 from original to tasmota 9.3.1 and later throw tasmota web upgrade tools to 9.3.1.1 
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
18:34:17.694 RSL: RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"3552":"Zigbee Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3584":"Zigbee Rx"},"GPIO4":{"5312":"Zigbee Rst"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"608":"I2C SCL"},"GPIO15":{"0":"None"},"GPIO16":{"32":"Button1"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
00:00:04.003 HTP: Web server active on tasmota_D7F36B-4971 with IP address 192.168.1.118
00:00:04.005 RSL: INFO1 = {"Module":"Sonoff ZbBridge","Version":"9.3.1.1(zbbridge)","FallbackTopic":"cmnd/DVES_D7F36B_fb/","GroupTopic":"cmnd/tasmotas/"}
00:00:04.006 RSL: INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota_D7F36B-4971","IPAddress":"192.168.1.118"}
00:00:04.010 RSL: INFO3 = {"RestartReason":{"Exception":9,"Reason":"Exception","EPC":["4023ab4b","00000000","00000000"],"EXCVADDR":"00004bca","DEPC":"00000000","CallChain":["4023a9cc","4020e85c","4020e855","4000050c","4000050c","4026b00f","4026b05e","401004a4","401029fc","4026a11f","4025376f","401062d3","40286c69","4024e8e2","402470cd","40286c6e","40244af4","40247356","40247774","402478ed","4024ddd1","4024e878","40104e52","4024e90a","4024eff7","4024b1c4","40100f3f","401063e9","4024b21b","4010666d","4023df84"]}}
00:00:04.480 RTC: UTC 2021-03-01T17:35:17, DST 2021-03-28T02:00:00, STD 2021-10-31T03:00:00
18:35:18.456 HTP: Main Menu
18:35:19.511 QPC: Reset
18:35:21.416 APP: Boot Count 260
```

### TO REPRODUCE
Set Sonoff ZbBridge Module to Sonoff ZbBridge (75)

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2021-03-01:
Is it the precompiled firmware or self compiled?

Haroldas-sys, 2021-03-01:
Really I don’t know at this time how to compile, so I downloaded from http://ota.tasmota.com/tasmota/release/tasmota-zbbridge.bin
Any suggestion to try ?

s-hadinger, 2021-03-01:
I will look in the map file if I can locate the crash

s-hadinger, 2021-03-02:
Could you please flash this firmware and report again the crash details?
https://file.hadinger.fr/807ird

MattWestb, 2021-03-02:
Is MQTT defined and if yes is it configurated ?
If not tasmota is rebooting.

s-hadinger, 2021-03-02:
I don't think the crash is related to Mqtt. But let's wait for the feedback with the other firmware

Haroldas-sys, 2021-03-02:
> Could you please flash this firmware and report again the crash details?
> https://file.hadinger.fr/807ird

With this firmware as you send (9.3.1) result the same, even restarts more frequently, so i can't go to console.
Sonoff now without mqtt configured (i can't write settings due <16 secons), but i tried earlier witt mqtt - reboots the same.

s-hadinger, 2021-03-02:
Any chance you can have syslog server?

If you have the console at least once, please grab the crash exception list

Haroldas-sys, 2021-03-02:
Yes, i have log server and will write log soon 

Haroldas

s-hadinger, 2021-03-02:
Thanks this is useful.

Can you please try the following firmware. It will hopefully not crash anymore and will show the faulty log line. One of the logs should show `<--INVALID-->`

https://file.hadinger.fr/awj5z6

Haroldas-sys, 2021-03-02:
I did't crash anymore. Thanks !

s-hadinger, 2021-03-02:
Sorry, I still need you. Can you please set log to level 3 or 4 and see where you see `<--INVALID-->`

The bug is still there, hidden in the dark

Haroldas-sys, 2021-03-02:
I run sonoff for 10 minutes, but no -invalid- text in the log
I'm now trying pair sonoff devices to Sonoff ZBB, but at the the Permit Join got
"4:25:30.946 SRC: WebGui from 192.168.1.101
14:25:30.948 CMD: Group 0, Index 1, Command "ZBPERMITJOIN", Data ""
14:25:30.952 MQT: stat/tasmota_D7F36B/RESULT = {"ZbPermitJoin":"Zigbee not started"}"

s-hadinger, 2021-03-02:
This log should happen at the moment of the former crash, right after `ZIG: Resetting EZSP device`. It may only display at `Weblog 4` but this may get quickly out of the buffer.

You may need to check in the syslog server at loglevel 4.

Haroldas-sys, 2021-03-02:
I restarted device by menu "restart"

<13>1 2021-03-02T15:38:19.357365+02:00 tasmota_D7F36B-4971 ESP-WIF - - - ESP-WIF: Checking connection...
<13>1 2021-03-02T15:38:39.338874+02:00 tasmota_D7F36B-4971 ESP-WIF - - - ESP-WIF: Checking connection...
<13>1 2021-03-02T15:38:40.334259+02:00 tasmota_D7F36B-4971 ESP-HTP - - - ESP-HTP: Main Menu
<13>1 2021-03-02T15:38:42.348434+02:00 tasmota_D7F36B-4971 ESP-HTP - - - ESP-HTP: Configuration
<13>1 2021-03-02T15:38:44.346285+02:00 tasmota_D7F36B-4971 ESP-HTP - - - ESP-HTP: Configure Logging
<13>1 2021-03-02T15:38:54.349313+02:00 tasmota_D7F36B-4971 ESP-HTP - - - ESP-HTP: Configure Logging
<13>1 2021-03-02T15:38:54.349313+02:00 tasmota_D7F36B-4971 ESP-CFG - - - ESP-CFG: CR 403/699, Busy 0
<13>1 2021-03-02T15:38:54.355615+02:00 tasmota_D7F36B-4971 ESP-LOG - - - ESP-LOG: SerialLog 0, WebLog 4, MqttLog 4, SysLog 4, LogHost 192.168.1.250, LogPort 514, TelePeriod 300
<13>1 2021-03-02T15:38:54.356208+02:00 tasmota_D7F36B-4971 ESP-HTP - - - ESP-HTP: Configuration
<13>1 2021-03-02T15:38:55.350890+02:00 tasmota_D7F36B-4971 ESP-CFG - - - ESP-CFG: Saved to flash at FD, Count 115, Bytes 4096
<13>1 2021-03-02T15:38:55.353699+02:00 tasmota_D7F36B-4971 ESP-HTP - - - ESP-HTP: Main Menu
<13>1 2021-03-02T15:38:59.320124+02:00 tasmota_D7F36B-4971 ESP-WIF - - - ESP-WIF: Checking connection...
<13>1 2021-03-02T15:39:19.328971+02:00 tasmota_D7F36B-4971 ESP-WIF - - - ESP-WIF: Checking connection...
<13>1 2021-03-02T15:39:39.339340+02:00 tasmota_D7F36B-4971 ESP-WIF - - - ESP-WIF: Checking connection...
<13>1 2021-03-02T15:39:45.344551+02:00 tasmota_D7F36B-4971 ESP-MQT - - - ESP-MQT: tele/tasmota_D7F36B/STATE = {"Time":"2021-03-02T14:39:43","Uptime":"0T00:15:11","UptimeSec":911,"Vcc":3.510,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Speed5G","BSSId":"30:23:03:97:5D:AB","Channel":8,"RSSI":94,"Signal":-53,"LinkCount":1,"Downtime":"0T00:00:05"}}
<13>1 2021-03-02T15:39:59.353425+02:00 tasmota_D7F36B-4971 ESP-WIF - - - ESP-WIF: Checking connection...
<13>1 2021-03-02T15:40:13.332832+02:00 tasmota_D7F36B-4971 ESP-HTP - - - ESP-HTP: Restart
<13>1 2021-03-02T15:40:13.334049+02:00 tasmota_D7F36B-4971 ESP-SRC - - - ESP-SRC: WebGui from 0.0.0.0
<13>1 2021-03-02T15:40:18.839566+02:00 tasmota_D7F36B-4971 ESP-WIF - - - ESP-WIF: Checking connection...
<13>1 2021-03-02T15:40:18.840269+02:00 tasmota_D7F36B-4971 ESP-WIF - - - ESP-WIF: Attempting connection...
<13>1 2021-03-02T15:40:19.834602+02:00 tasmota_D7F36B-4971 ESP-WIF - - - ESP-WIF: Checking connection...
<13>1 2021-03-02T15:40:19.835935+02:00 tasmota_D7F36B-4971 ESP-WIF - - - ESP-WIF: Connected
<13>1 2021-03-02T15:40:19.837760+02:00 tasmota_D7F36B-4971 ESP-HTP - - - ESP-HTP: Web server active on tasmota_D7F36B-4971 with IP address 192.168.1.121
<13>1 2021-03-02T15:40:20.873189+02:00 tasmota_D7F36B-4971 ESP-RTC - - - ESP-RTC: UTC 2021-03-02T13:40:18, DST 2021-03-28T02:00:00, STD 2021-10-31T03:00:00
<13>1 2021-03-02T15:40:20.874384+02:00 tasmota_D7F36B-4971 ESP-MQT - - - ESP-MQT: Attempting connection...
<13>1 2021-03-02T15:40:20.876837+02:00 tasmota_D7F36B-4971 ESP-MQT - - - ESP-MQT: Connected
<13>1 2021-03-02T15:40:20.879198+02:00 tasmota_D7F36B-4971 ESP-MQT - - - ESP-MQT: tele/tasmota_D7F36B/LWT = Online (retained)
<13>1 2021-03-02T15:40:20.885281+02:00 tasmota_D7F36B-4971 ESP-MQT - - - ESP-MQT: cmnd/tasmota_D7F36B/POWER = 
<13>1 2021-03-02T15:40:20.886974+02:00 tasmota_D7F36B-4971 ESP-MQT - - - ESP-MQT: Subscribe to cmnd/tasmota_D7F36B/#
<13>1 2021-03-02T15:40:20.887556+02:00 tasmota_D7F36B-4971 ESP-MQT - - - ESP-MQT: Subscribe to cmnd/tasmotas/#
<13>1 2021-03-02T15:40:20.888477+02:00 tasmota_D7F36B-4971 ESP-MQT - - - ESP-MQT: Subscribe to cmnd/DVES_D7F36B_fb/#
<13>1 2021-03-02T15:40:20.889156+02:00 tasmota_D7F36B-4971 ESP-MQT - - - ESP-MQT: tele/tasmota_D7F36B/INFO1 = {"Module":"Sonoff ZbBridge","Version":"9.3.1.1(zbbridge)","FallbackTopic":"cmnd/DVES_D7F36B_fb/","GroupTopic":"cmnd/tasmotas/"}
<13>1 2021-03-02T15:40:20.890238+02:00 tasmota_D7F36B-4971 ESP-MQT - - - ESP-MQT: tele/tasmota_D7F36B/INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota_D7F36B-4971","IPAddress":"192.168.1.121"}
<13>1 2021-03-02T15:40:20.890871+02:00 tasmota_D7F36B-4971 ESP-MQT - - - ESP-MQT: tele/tasmota_D7F36B/INFO3 = {"RestartReason":"Software/System restart"}
<13>1 2021-03-02T15:40:21.934550+02:00 tasmota_D7F36B-4971 ESP-QPC - - - ESP-QPC: Reset
<13>1 2021-03-02T15:40:23.841795+02:00 tasmota_D7F36B-4971 ESP-HTP - - - ESP-HTP: Main Menu
<13>1 2021-03-02T15:40:23.842863+02:00 tasmota_D7F36B-4971 ESP-APP - - - ESP-APP: Boot Count 93
<13>1 2021-03-02T15:40:24.852350+02:00 tasmota_D7F36B-4971 ESP-CFG - - - ESP-CFG: Saved to flash at FC, Count 116, Bytes 4096
<13>1 2021-03-02T15:40:25.871466+02:00 tasmota_D7F36B-4971 ESP-MQT - - - ESP-MQT: tele/tasmota_D7F36B/STATE = {"Time":"2021-03-02T14:40:23","Uptime":"0T00:00:11","UptimeSec":11,"Vcc":3.510,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Speed5G","BSSId":"30:23:03:97:5D:AB","Channel":8,"RSSI":90,"Signal":-55,"LinkCount":1,"Downtime":"0T00:00:05"}}
<13>1 2021-03-02T15:40:30.838493+02:00 tasmota_D7F36B-4971 ESP-ZIG - - - ESP-ZIG: Resetting EZSP device
<13>1 2021-03-02T15:40:30.840536+02:00 tasmota_D7F36B-4971 ESP-ZbEZSPReceived - - - ESP-ZbEZSPReceived: time-out, discarding ō $@|ŲĢ?dwarv
<13>1 2021-03-02T15:40:35.850876+02:00 tasmota_D7F36B-4971 ESP-ZIG - - - ESP-ZIG: timeout, goto label 99
[... cut ...] }
<13>1 2021-03-02T15:40:35.853138+02:00 tasmota_D7F36B-4971 ESP-ZIG - - - ESP-ZIG: Abort
<13>1 2021-03-02T15:40:35.854640+02:00 tasmota_D7F36B-4971 ESP-ZIG - - - ESP-ZIG: Stopping (99)
<13>1 2021-03-02T15:40:39.868112+02:00 tasmota_D7F36B-4971 ESP-WIF - - - ESP-WIF: Checking connection...
<13>1 2021-03-02T15:40:59.858143+02:00 tasmota_D7F36B-4971 ESP-WIF - - - ESP-WIF: Checking connection...
<13>1 2021-03-02T15:41:19.860169+02:00 tasmota_D7F36B-4971 ESP-WIF - - - ESP-WIF: Checking connection...

s-hadinger, 2021-03-02:
Thanks I found the bug :)

closed 2021-03-02 by s-hadinger

Haroldas-sys, 2021-03-02:
But how to start Zigbee on this Sonoff ZBB ? I allways get the same, trying to start pair, o view Zigbee Map ?

15:32:55.723 CMD: ZbPermitJoin 1
15:32:55.724 SRC: WebConsole from 192.168.1.101
15:32:55.726 CMD: Group 0, Index 1, Command "ZBPERMITJOIN", Data "1"
15:32:55.730 MQT: stat/tasmota_D7F36B/RESULT = {"ZbPermitJoin":"Zigbee not started"}
15:32:57.770 WIF: Checking connection...
15:33:17.796 WIF: Checking connection...

s-hadinger, 2021-03-02:
That's another issue. Did you successfully flashed EZSP on the MCU?

Haroldas-sys, 2021-03-02:
After flash on mcu, zigbee starts and works cool ! Thank you.

PR #11192 (2021-03-02): Zigbee fix crash when bad frame is received
## Description:

Fixes: #11178 

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works on Tasmota core ESP8266 V.2.7.4.9
  - [x] The code change is tested and works with core ESP32 V.1.0.5
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreement-cla).

_NOTE: The code change must [... cut]