arendst/Tasmota #11185: TOGGLE command are inverted in Sonoff Touch 2/3CH, Latest Firmware

FabrizioRomanoGenovese, 2021-03-02:
### PROBLEM DESCRIPTION
In Modules Sonoff T1 2CH and Sonoff T1 3CH, the TOGGLE commands are inverted. What I mean is this: Consider the 2CH module. If I press or hold the leftmost button, I correctly get `/stat/POWER1 = ON` and `/cmnd/POWER1 = HOLD`, respectively. But if I double press the same button, then I get `/cmnd/POWER2 = TOGGLE`. So, as for toggle, the two buttons are mixed up. Same thing holds for the T1 3CH module. Could it maybe be some typo in the latest firmware model?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff T1 2CH
- [x] Tasmota binary firmware version number used: 9.3.1
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: Tasmotizer
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
{"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
{"Module":{"29":"Sonoff T1 2CH"}}
{"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"225":"Relay2"},"GPIO9":{"33":"Button2"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua

{"Status":{"Module":29,"DeviceName":"SwitchesBedroomLiving","FriendlyName":["SwitchesBedroomLiving","Sonoff2"],"Topic":"living","ButtonTopic":"livingcmnd","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
12:30:17.886 APP: Button1 multi-press 1
12:30:18.085 APP: Button1 multi-press 2
12:30:18.596 MQT: switches/bedroom/livingcmnd/cmnd/POWER2 = TOGGLE
```

### TO REPRODUCE
Just install the latest firmware on Sonoff T1 2/3CH.

### EXPECTED BEHAVIOUR
The toggle behavior shouldn't swap the 2 buttons.

### SCREENSHOTS

### ADDITIONAL CONTEXT

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2021-03-02:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-03-02 by ascillato2

FabrizioRomanoGenovese, 2021-03-02:
I was pretty sure Github was the right place for signaling bugs...