arendst/Tasmota #11189: Adding support of M5stack core 1.0 ILI9341 display (two registers to change)

ouinouin, 2021-03-02:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

on the M5stack , the display ILI9341 is slightly different than the standard one, , and the display is showed mirrored (and partially greyed)
I took a look in esphome sourcecode since they have this variant  supported, , the issue relies in a left to right  register and also the framecontrol : 
in the file : 
https://github.com/arendst/Tasmota/blob/development/lib/lib_display/ILI9341-gemu-1.0/ILI9341_2.cpp
according to : 
https://github.com/esphome/esphome/blob/dev/esphome/components/ili9341/ili9341_init.h

for the standard one implemented in tasmota  ` ILI9341_2_FRMCTR1 , 2, 0x00, 0x18,`
for M5stack should be ILI9341_2_FRMCTR1 ,` 2, 0x00, 0x13,`

for the standard one implemented in tasmota :  `ILI9341_2_MADCTL  , 1, 0x48`
for M5stack should be `ILI9341_2_MADCTL  , 1, 0x08`
0X40 seems to have something to do with a right to left thing.

@gemu2015 I miss coding skill (near to 0) to know how to implement this correctly but im interrested in how you would do that (declare the M5stack variant as another screen , add a conffig stuff somewhere ?.
to what i see the other tft is also nearly the same, is there a trick with progmem to save space since they are nearly the same declaration of registers ? 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X ] Device used (e.g., Sonoff Basic): M5stack core 1.0
- [X] Tasmota binary firmware version number used: 9.3.1.1
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: esptool.py
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

gemu2015, 2021-03-02:
this should already work:
try
#define USE_DISPLAY_ILI9342
#define USE_M5STACK_CORE_BASIC
and define SPI_MOSI, SPI_MISO,SPI_SCK and SD_Card_CS according to table
(leave DC,CS,RST and BL undefined as these are defined by the driver)

[image]

ouinouin, 2021-03-02:
Hi, I ll try that, unfortunately i miss bandwidth to quickly downolad the sourcecode, I m trying to make a template for the M5stack core 1.0 , but to what i understant its a compile time option, so any chance to compile a version for the m5stack in the releases ?

gemu2015, 2021-03-02:
i currently modify the source to configure ILI9341/2 by setup so that for CORE1 this would be included in the binary. This however will not work for CORE2 because this board uses dedicated pins of a peripheral chip for the display. since i do not have a core1 i can't test directly. i may provide a binary here for testing.

ouinouin, 2021-03-02:
@gemu2015 I actually have CORE1 + spare time (actually in quarantine...) so feel free to push a binary i ll test it quickly (as far as my angolese bandwidth allows it.

gemu2015, 2021-03-02:
ok, try this version.

define ALL pins from the table above for ILI9341 including SD Card

then you have a console cmd  displayilimode N
N = 1 selects display ILI9341
N = 2 selects display ILI9342

so try with 2 for core1

[tasmota32.bin.zip](https://github.com/arendst/Tasmota/files/6069348/tasmota32.bin.zip)

ouinouin, 2021-03-02:
did the test : displayilimode 1 displays the same issue. 
displayilimode 2 give me a black screen. 

the registers set for the 9341 on the core V1 are not the same than ILI9342 did you replace the config of 9342 by the one from 9341for the M5stack core 1? 
[image]

gemu2015, 2021-03-02:
ok, this driver works with ILI9341 with ilimode 1  (i have an ILI9341, so testet) should have mirror screen on your device

works with ilimode 2 on core 2 which has an ILI9342. (i have a core2 so also testet)

the docs of the core1 say its an ili9342 too
 
had there been an older core1 version with ili9341 ?

ouinouin, 2021-03-02:
hi, yes as per the datasheet i get in the box :  its a ILI9341 , but its a tweaked wersion (look at the first post i got the correct tweaks from esphome.

the PCB  shows 03 /2018
damn i dont know how the chinese do to have so much tweaked versions of hardware...

[image]
[image]
[image]

gemu2015, 2021-03-02:
ok then there are 2 different version. if you are sure about these 2 registers i will provide a version that allows to switch between these 2 settings.

ouinouin, 2021-03-02:
[image]
some progress, still some grayish stuff on the left side.
on your build I cannot switch to displaymode 1 for example, it is normal ? (on the tasmota-display build i had no issue with that

gemu2015, 2021-03-02:
you said displayilimode 2  remained black. please try again and set display dimmer to 100
because it should  also show something. just want to know

ouinouin, 2021-03-02:
nothing even dimmed at 100percent with displayilimode 2

checking history of this file from m5 stack i m trying to see if there is other register set differences
https://github.com/m5stack/M5Stack_TFT_ILI9341/blob/f7147ec9c9f9a5be24f3a1e0b4235f99301fe3cc/components/tft/tftspi.c
maybe the ILI9341_2_FRMCTR1  setting is not correct against the freq of the SPI or something like that, this is the refresh rate of the screen.

gemu2015, 2021-03-02:
in displayilimode 2 can you see that the backlight is turned on ?  it definitely must be on.
and also test displayilimode 3 just to know what happens

and is the sd card recognized ? (please insert one with fat32 format)

ouinouin, 2021-03-02:
yes the on displayilimode 2 i have the control over the backlight , i can dim the light  but nothing is displayed

displayilimode 3 is nice :  here in the picture the displayilimode 3 : white font upon black background,  is this what we re expect to see by default ? 
[image]

ouinouin, 2021-03-02:
i could draw a white rectangle with {"DisplayText":"[x50y50r200:100]"}
"displaymode 1" is not working.

gemu2015, 2021-03-02:
ok that's what i thought! it IS an ILI9342 !!!! (thats what is stated in newer docs)

the difference between mode 2 and 3 is only that SPI is initialized in mode 3 and not in mode 2

displaymodes 1-5 are disabled in this build. i never use them

gemu2015, 2021-03-02:
if you had an sd card enabled and in place also ilimode 2 would work.

ok, i will do a pr as it is solved now

ouinouin, 2021-03-02:
ok, so finally the solution for this device is to have this ilimode option activated and the to choose the correct one if you get or not an sdcard ? 
(i m looking in all my devices but I dont have a microsdcard available now, so I wont be able to do a test with SDcard  before a while)

gemu2015, 2021-03-02:
no problem, you may always use mode 3, will have no side effect

PR #11199 (2021-03-03): ili9341/2 software configurable
## Description:

**Related issue (if applicable):** fixes #11189

now ili9341/2 ist treated as ili9341 CS and DC
displayilimode the sets the display to mode for chip type 1 or 2 (3 = mode 2 + reinit spi )
displayinvert 0,1 inverts ili display colors

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works on Tasmota core ESP8266 V.2.7.4.9
  - [x] The code chan [... cut]
closed 2021-03-03 by arendst