arendst/Tasmota #11215: MCP230xx output mode not syncing relays

theOzzieRat, 2021-03-04:
### PROBLEM DESCRIPTION
Using the MCP230xx driver in output mode does not sync the power settings when the pin is configured as output, or when the device is booted. This only occurs if the output pin is configured to output a 1 as its default. The MCP230xx chip gets the right value but the POWER is wrong.

```12:38:42.172 CMD: setoption0 0
12:38:42.175 SRC: WebConsole from 10.0.10.1
12:38:42.178 CMD: Group 0, Index 0, Command "SETOPTION", Data "0"
12:38:42.184 MQT: stat/pi_power/RESULT = {"SetOption0":"OFF"}
12:38:57.941 CMD: sensor29 reset1
12:38:57.944 SRC: WebConsole from 10.0.10.1
12:38:57.948 CMD: Group 0, Index 29, Command "SENSOR", Data "reset1"
12:38:57.957 MQT: stat/pi_power/RESULT = {"Sensor29_D99":{"MODE":1,"PULL_UP":"OFF","INT_MODE":"DISABLED","STATE":""}}
12:39:15.119 CMD: sensor29 0,5,1
12:39:15.122 SRC: WebConsole from 10.0.10.1
12:39:15.125 CMD: Group 0, Index 29, Command "SENSOR", Data "0,5,1"
12:39:15.136 MQT: stat/pi_power/RESULT = {"Sensor29_D0":{"MODE":5,"PULL_UP":"ON","INT_MODE":"DISABLED","STATE":"ON"}}
12:39:15.336 CFG: Saved to flash at F4, Count 246, Bytes 4096
12:39:34.274 CMD: sensor29 0,?
12:39:34.277 SRC: WebConsole from 10.0.10.1
12:39:34.280 CMD: Group 0, Index 29, Command "SENSOR", Data "0,?"
12:39:34.287 MQT: stat/pi_power/RESULT = {"Sensor29_D0":{"MODE":5,"PULL_UP":"ON","INT_MODE":"DISABLED","STATE":"ON"}}
12:39:41.917 CMD: power1
12:39:41.920 SRC: WebConsole from 10.0.10.1
12:39:41.923 CMD: Group 0, Index 1, Command "POWER", Data ""
12:39:41.928 MQT: stat/pi_power/RESULT = {"POWER":"OFF"}
12:39:41.932 MQT: stat/pi_power/POWER = OFF
```

And after a reboot

```16:55:35.568 CMD: sensor29 0,?
16:55:35.577 MQT: stat/pi_power/RESULT = {"Sensor29_D0":{"MODE":5,"PULL_UP":"ON","INT_MODE":"DISABLED","STATE":"ON"}}
16:55:40.630 CMD: power1
16:55:40.637 MQT: stat/pi_power/RESULT = {"POWER":"OFF"}
16:55:40.640 MQT: stat/pi_power/POWER = OFF
```

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): Wemos D1 Mini with MCP230xx attached
- [x] Tasmota binary firmware version number used: 9.3.1.1
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: OTA using HA plugin
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
12:47:47.106 SRC: Backlog
12:47:47.109 CMD: Group 0, Index 1, Command "MODULE", Data ""
12:47:47.115 MQT: stat/pi_power/RESULT = {"Module":{"18":"Generic"}}
12:47:47.357 SRC: Backlog
12:47:47.360 CMD: Group 0, Index 1, Command "GPIO", Data "255"
12:47:47.369 MQT: stat/pi_power/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"640":"I2C SDA"},"GPIO5":{"608":"I2C SCL"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"1312":"DS18x20"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
Rules not used
- [x] Provide the output of this command: `Status 0`:
```lua
  12:49:26.150 MQT: stat/pi_power/STATUS = {"Status":{"Module":18,"DeviceName":"Pi Power","FriendlyName":["Pi Power"],"Topic":"pi_power","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":0,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
12:51:11.923 CMD: Group 0, Index 0, Command "SETOPTION", Data "0"
12:51:11.928 MQT: stat/pi_power/RESULT = {"SetOption0":"OFF"}
12:51:19.429 CMD: sensor29 reset1
12:51:19.432 SRC: WebConsole from 10.0.10.1
12:51:19.435 CMD: Group 0, Index 29, Command "SENSOR", Data "reset1"
12:51:19.443 MQT: stat/pi_power/RESULT = {"Sensor29_D99":{"MODE":1,"PULL_UP":"OFF","INT_MODE":"DISABLED","STATE":""}}
12:51:20.337 CFG: Saved to flash at FA, Count 248, Bytes 4096
12:51:23.503 WIF: Checking connection...
12:51:28.070 CMD: sensor29 0,5,1
12:51:28.073 SRC: WebConsole from 10.0.10.1
12:51:28.076 CMD: Group 0, Index 29, Command "SENSOR", Data "0,5,1"
12:51:28.085 MQT: stat/pi_power/RESULT = {"Sensor29_D0":{"MODE":5,"PULL_UP":"ON","INT_MODE":"DISABLED","STATE":"ON"}}
12:51:28.306 CFG: Saved to flash at F9, Count 249, Bytes 4096
12:51:33.830 CMD: sensor29 0,?
12:51:33.832 SRC: WebConsole from 10.0.10.1
12:51:33.835 CMD: Group 0, Index 29, Command "SENSOR", Data "0,?"
12:51:33.842 MQT: stat/pi_power/RESULT = {"Sensor29_D0":{"MODE":5,"PULL_UP":"ON","INT_MODE":"DISABLED","STATE":"ON"}}
12:51:37.932 CMD: power1
12:51:37.935 SRC: WebConsole from 10.0.10.1
12:51:37.938 CMD: Group 0, Index 1, Command "POWER", Data ""
12:51:37.943 MQT: stat/pi_power/RESULT = {"POWER":"OFF"}
12:51:37.947 MQT: stat/pi_power/POWER = OFF
```

Note that after a restart the POWER value is also incorrect.

```12:57:42.294 CFG: Saved to flash at F8, Count 250, Bytes 4096
12:57:44.512 CMD: sensor29 0,?
12:57:44.514 SRC: WebConsole from 10.0.10.1
12:57:44.517 CMD: Group 0, Index 29, Command "SENSOR", Data "0,?"
12:57:44.524 MQT: stat/pi_power/RESULT = {"Sensor29_D0":{"MODE":5,"PULL_UP":"ON","INT_MODE":"DISABLED","STATE":"ON"}}
12:57:45.416 MQT: tasmota/discovery/483FDA9E9ECE/config = {"ip":"10.0.10.18","dn":"Pi Power","fn":["Pi Power",null,null,null,null,null,null,null],"hn":"Sonoff-18","mac":"483FDA9E9ECE","md":"Generic","ty":0,"if":0,"ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"9.3.1.1","t":"pi_power","ft":"%prefix%/%topic%/","tp":["cmnd","stat","tele"],"rl":[1,0,0,0,0,0,0,0],"swc":[-1,-1,-1,-1,-1,-1,-1,-1],"swn":[null,null,null,null,null,null,null,null],"btn":[0,0,0,0,0,0,0,0],"so":{"4":0,"11":0,"13":0,"17":0,"20":0,"30":0,"68":0,"73":0,"82":0,"114":0,"117":0},"lk":1,"lt_st":0,"sho":[0,0,0,0],"ver":1} (retained)
12:57:45.426 MQT: tasmota/discovery/483FDA9E9ECE/sensors = {"sn":{"Time":"2021-03-04T12:57:45","DS18B20-1":{"Id":"0000012F1B5A","Temperature":23.6},"DS18B20-2":{"Id":"00000D19D6E9","Temperature":23.4},"MCP230XX":{"D0":1,"D1":0,"D2":0,"D3":0,"D4":0,"D5":0,"D6":0,"D7":0,"D8":0,"D9":0,"D10":0,"D11":0,"D12":0,"D13":0,"D14":0,"D15":0,"MCP230_OUT":{"OUT_D0":"ON","END":1}},"TempUnit":"C"},"ver":1} (retained)
12:57:46.918 CMD: power1
12:57:46.921 SRC: WebConsole from 10.0.10.1
12:57:46.924 CMD: Group 0, Index 1, Command "POWER", Data ""
12:57:46.930 MQT: stat/pi_power/RESULT = {"POWER":"OFF"}
12:57:46.936 MQT: stat/pi_power/POWER = OFF
```

### TO REPRODUCE
Compile Tasmota with
#define USE_MCP230xx
    #define USE_MCP230xx_ADDR 0x20
    #define USE_MCP230xx_OUTPUT

Execute steps as in the debug log above.

### EXPECTED BEHAVIOUR
POWER1 should return ON

### SCREENSHOTS
n/a

### ADDITIONAL CONTEXT
I have discussed briefly with barbudor on discord. He has looked at the issue but said it was not super easy to fix quickly and asked me to raise it as an issue.

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2021-03-04:
Hi,

As designed. As explained in old issues:
https://github.com/arendst/Tasmota/issues/8965
https://github.com/arendst/Tasmota/issues/3350
The power command is not related to the MCP230xx states. 

Closing this issue as it is **duplicated**. Sorry.

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

theOzzieRat, 2021-03-04:
I believe PR #9025 changed this behaviour? If after configuring a pin on the MCP230xx as an output sensor29 0,x with change the corresponding POWER state. Also POWER1 ON/OFF changes the MCP230xx output.

PR #11362 (2021-03-17): Fix MCP2300x initialisation.
## Description:

**Related issue (if applicable):** fixes #11215 

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works on Tasmota core ESP8266 V.2.7.4.9
  - [ ] The code change is tested and works with core ESP32 V.1.0.5
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreement-cla).

_NOTE: T [... cut]
closed 2021-03-17 by arendst