arendst/Tasmota #11216: no update with esp 8285

rebel1985, 2021-03-04:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

i have do an upgrade from 8.1 to 8.5.1 to 9.1 and i dont can do it to 9.3.1 and i have no solution why that dont do it. can you help me to update it?
but i have more from the plugs and the another are all updated to 9.3.1 without problems.
i do upgrade with the link or the local file on my pc and dosend work

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ X] Device used (e.g., Sonoff Basic):  Sonoff Basic DE
- [ X] Tasmota binary firmware version number used: actually  9.1.0
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: web and local upload
- [X ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
{"NAME":"Teckin","GPIO":[0,32,0,321,2720,2656,0,0,2624,320,224,0,0,0],"FLAG":0,"BASE":52}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ X] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
07:15:53 CMD: Status 0
07:15:53 MQT: stat/Steckdose_Computer/STATUS = {"Status":{"Module":0,"DeviceName":"Computer","FriendlyName":["Computer"],"Topic":"Steckdose_Computer","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
07:15:53 MQT: stat/Steckdose_Computer/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-minimal.bin","RestartReason":"Software/System restart","Uptime":"0T00:11:16","StartupUTC":"2021-03-04T06:04:37","Sleep":50,"CfgHolder":4617,"BootCount":64,"BCResetTime":"2020-09-07T16:11:03","SaveCount":1558,"SaveAddress":"FB000"}}
07:15:53 MQT: stat/Steckdose_Computer/STATUS2 = {"StatusFWR":{"Version":"9.1.0(tasmota)","BuildDateTime":"2020.11.07 12:04:01","Boot":31,"Core":"2_7_4_5","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"426/699"}}
07:15:53 MQT: stat/Steckdose_Computer/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Terminator",""],"TelePeriod":30,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
07:15:53 MQT: stat/Steckdose_Computer/STATUS4 = {"StatusMEM":{"ProgramSize":601,"Free":400,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000407","8FDAE797","04368001","000000CF","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
07:15:53 MQT: stat/Steckdose_Computer/STATUS5 = {"StatusNET":{"Hostname":"Computer","IPAddress":"192.168.178.XX","Gateway":"192.168.178.XX","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.XXX","Mac":"XXXXXXXXX","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
07:15:53 MQT: stat/Steckdose_Computer/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.XX","MqttPort":1883,"MqttClientMask":"Steckdose Computer","MqttClient":"Steckdose Computer","MqttUser":"XXXXX","MqttCount":3,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
07:15:53 MQT: stat/Steckdose_Computer/STATUS7 = {"StatusTIM":{"UTC":"2021-03-04T06:15:53","Local":"2021-03-04T07:15:53","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"07:25","Sunset":"18:37"}}
07:15:53 MQT: stat/Steckdose_Computer/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
07:15:53 MQT: stat/Steckdose_Computer/STATUS10 = {"StatusSNS":{"Time":"2021-03-04T07:15:53","ENERGY":{"TotalStartTime":"2020-04-06T08:48:50","Total":316.543,"Yesterday":1.988,"Today":0.166,"Power":207,"ApparentPower":209,"ReactivePower":30,"Factor":0.99,"Voltage":231,"Current":0.904}}}
07:15:53 MQT: stat/Steckdose_Computer/STATUS11 = {"StatusSTS":{"Time":"2021-03-04T07:15:53","Uptime":"0T00:11:16","UptimeSec":676,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":3,"POWER":"ON","Wifi":{"AP":1,"SSId":"XXXXXX"

```
- [X ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

07:19:50 MQT: stat/Steckdose_Computer/RESULT = {"WebLog":2}
07:19:56 CMD: weblog 4
07:19:56 MQT: stat/Steckdose_Computer/RESULT = {"WebLog":4}
07:19:56 CFG: in Flash gespeichert am FB, zählen 1559, Bytes 4096
07:20:11 WIF: Prüfe Verbindung...
07:20:15 MQT: tele/Steckdose_Computer/STATE = {"Time":"2021-03-04T07:20:15","Uptime":"0T00:15:38","UptimeSec":938,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":3,"POWER":"ON","Wifi":{"AP":1,"SSId":"Terminator","BSSId":"2C:3A:FD:97:7B:B6","Channel":8,"RSSI":50,"Signal":-75,"LinkCount":1,"Downtime":"0T00:00:03"}}
07:20:15 MQT: tele/Steckdose_Computer/SENSOR = {"Time":"2021-03-04T07:20:15","ENERGY":{"TotalStartTime":"2020-04-06T08:48:50","Total":316.558,"Yesterday":1.988,"Today":0.180,"Period":2,"Power":208,"ApparentPower":212,"ReactivePower":40,"Factor":0.98,"Voltage":233,"Current":0.908}}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

07:10:52 HTP: Web-Server angehalten
07:11:07 HTP: Web-Server aktiv bei Computer mit IP-Adresse 192.168.178.xx
07:11:07 MQT: Verbindungsversuch...
07:11:07 MQT: verbunden
07:11:07 MQT: tele/Steckdose_Computer/LWT = Online (beibehalten)
07:11:07 MQT: cmnd/Steckdose_Computer/POWER = 
07:11:08 MQT: stat/Steckdose_Computer/UPGRADE = {"Upgrade":"Failed Nicht kompatibel"}

**(Please, remember to close the issue when the problem has been addressed)**

schrej-zz, 2021-03-04:
Hi @rebel1985,
did you use the gzipped versions of tasmota (min) bin files? Only those seem to work now due to size constraints.
Jierg

rebel1985, 2021-03-04:
> 
> 
> Hi @rebel1985,
> did you use the gzipped versions of tasmota (min) bin files? Only those seem to work now due to size constraints.
> Jierg

are you german? lets talk in it 
i dont understand you sorry?

ascillato2, 2021-03-04:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-03-04 by ascillato2

schrej-zz, 2021-03-04:
@rebel1985 
On the Release Website use the files on the left yellow side.
E.g. for tasmota-minimal.bin.gz
http://ota.tasmota.com/tasmota/release/tasmota-minimal.bin.gz

rebel1985, 2021-03-05:
> 
> 
> @rebel1985
> On the Release Website use the files on the left yellow side.
> E.g. for tasmota-minimal.bin.gz
> http://ota.tasmota.com/tasmota/release/tasmota-minimal.bin.gz

thanks for your comment but this doesnt work , the message is : incompatibel for this device