arendst/Tasmota #11226: Connect device with application eWeLink

ngokhachung1998, 2021-03-05:
I want to connect ESP8266 with application eWeLink. I have tried but  failed. Can you have me

closed 2021-03-05 by close-issue-app[bot]

ascillato, 2021-03-05:
Tasmota won't connect to ewelink. Not supported

ngokhachung1998, 2021-03-12:
> Tasmota won't connect to ewelink. Not supported

Do you know any software similar to ewelink? Ewelink made it a bit easy for me to configure the device

ascillato, 2021-03-12:
Ewelink is a software from iTead for using the original firmware of sonoff devices. It is cloud based and not related and not compatible with Tasmota. 

If you haven't flashed Tasmota on your device, you can use ewelink.

If you have flashed Tasmota in your device, ewelink is not going to work anymore due to Tasmota won't use iTead Cloud Servers.

If you need a software to manage your Tasmotas, please check the Tasmota docs for TDM (Tasmota Device Manager), TasmoAdmin and others.

ascillato, 2021-03-12:
If you need further assistance, please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md)

Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)