arendst/Tasmota #11237: Counter is notworking correctly on falling edge

iot-sle, 2021-03-06:
### PROBLEM DESCRIPTION
I'm using a IR sensor and a hall sensor (switch mode) as a counter. Counterdebounce 1000. Counterdebouncelow 0 counterdebouncehigh 0.

Doing a short trigger  (< 1s ) on both sensors is working fine. As soon as the trigger event is taking longer as the debounce periond of 1 s. The count is 2. One ( after only rising edge is done and the end of debounce period is reached) and a second when the falling edge is detected after the debounce period. I expect only the second count, because the debounce period should only start after the falling edge. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): _____
- [x ] Tasmota binary firmware version number used: __9.3.1.1__
  - [ ] Pre-compiled
  - [ x] Self-compiled
- [ x] Flashing tools used: _OTA____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
16:28:26.889 CMD: Backlog Template; Module; GPIO 255
16:28:26.962 MQT: stat/MiniBoard02/RESULT = {"NAME":"ESP-01S","GPIO":[1,1,1,1,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":18}
16:28:27.173 MQT: stat/MiniBoard02/RESULT = {"Module":{"0":"ESP-01S"}}
16:28:27.427 MQT: stat/MiniBoard02/RESULT = {"GPIO0":{"1312":"DS18x20"},"GPIO1":{"385":"Counter_n2"},"GPIO2":{"0":"None"},"GPIO3":{"384":"Counter_n1"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
16:29:36.756 MQT: stat/MiniBoard02/STATUS = {"Status":{"Module":0,"DeviceName":"TAS-MiniBoard2","FriendlyName":["MiniBoard2"],"Topic":"MiniBoard02","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
16:29:36.761 MQT: stat/MiniBoard02/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:11:14","StartupUTC":"2021-03-06T15:18:22","Sleep":50,"CfgHolder":4617,"BootCount":96,"BCResetTime":"2020-11-09T18:13:29","SaveCount":226,"SaveAddress":"FA000"}}
16:29:36.768 MQT: stat/MiniBoard02/STATUS2 = {"StatusFWR":{"Version":"9.3.1.1(tasmota)","BuildDateTime":"2021.03.06 16:16:09","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"434/699"}}
16:29:36.773 MQT: stat/MiniBoard02/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["xxxxx",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
16:29:36.784 MQT: stat/MiniBoard02/STATUS4 = {"StatusMEM":{"ProgramSize":594,"Free":408,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14605E","FlashFrequency":40,"FlashMode":3,"Features":["00000407","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
16:29:36.791 MQT: stat/MiniBoard02/STATUS5 = {"StatusNET":{"Hostname":"MiniBoard02-3705","IPAddress":"192.168.252.87","Gateway":"192.168.252.2","Subnetmask":"255.255.255.0","DNSServer":"192.168.252.2","Mac":"A4:...","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
16:29:36.798 MQT: stat/MiniBoard02/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.252.5","MqttPort":1883,"MqttClientMask":"TAS-MiniBoard2","MqttClient":"TAS-MiniBoard2","MqttUser":"mqtt","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
16:29:36.805 MQT: stat/MiniBoard02/STATUS7 = {"StatusTIM":{"UTC":"2021-03-06T15:29:36","Local":"2021-03-06T16:29:36","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"07:21","Sunset":"18:40"}}
16:29:36.811 MQT: stat/MiniBoard02/STATUS10 = {"StatusSNS":{"Time":"2021-03-06T16:29:36","COUNTER":{"C1":2,"C2":30},"DS18S20":{"Id":"000801707F3E","Temperature":24.3},"TempUnit":"C"}}
16:29:36.817 MQT: stat/MiniBoard02/STATUS11 = {"StatusSTS":{"Time":"2021-03-06T16:29:36","Uptime":"0T00:11:14","UptimeSec":674,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"xxxxxx","BSSId":"DC:...:33","Channel":1,"RSSI":94,"Signal":-53,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
see description

### EXPECTED BEHAVIOUR
only count falling edge . Don't count end of debounce as an event.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2021-03-06:
What exactly do you mean by "switch event"? Doing a test with `Counterdebounce 1000; Counterdebouncelow 0; counterdebouncehigh 0`, using a momentary button (pulling down to ground) to trigger the counter, I'm seeing an increment of 1, both with short (under 1 sec) and long (over 1 sec) pushes. How is your pin input state different from that?

iot-sle, 2021-03-07:
I'm using a IR distance sensor and a Hall sensor to count. Both of them have a trigger level on board to determine the level when the output level is set to high. The IR sensor is NC and will become low when something is passing. The hall sensor is NO and will go to high if the magnet is passing. The web interface is showing the behavior described above. But I will do more test during next week. This morning the ESP was showing a "serial buffer overrun".  If we don't find a solution the second ways is to switch to an ESP32 and use the analog inputs conected to the analog outputs of the sensor boards. 

It may take some time. I need to think about solving a problem on the Wiegand module.