arendst/Tasmota #11247: Sonoff 4CHPROR3 not booting after module selection

Leox89, 2021-03-06:
### PROBLEM DESCRIPTION
After flashing (with tasmotizer) and correcting rebooting, configuration of wi-fi is working fine, after selecting module type 
23 | Sonoff 4Ch Pro via web-gui (same happening configuring via tasmotizer)

device is not reeboting anymore, keeps flashing unresponsive and needs to be re-flashed.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): SONOFF 4CHPROR3_____
- [ ] Tasmota binary firmware version number used: 9.3.0
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [ x] Flashing tools used: TASMOTIZER
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
UNABLE, device unresponsive

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
UNABLE, device unresponsive

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
UNABLE, device unresponsive

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
UNABLE, device unresponsive

```

### TO REPRODUCE

as decribed in issue

### EXPECTED BEHAVIOUR

normal boot with configuration different from "sonoff basic"

### SCREENSHOTS
N/A

### ADDITIONAL CONTEXT
N/A

**(Please, remember to close the issue when the problem has been addressed)**

schrej-zz, 2021-03-07:
Hi @Leox89,
I suggest to start with the generic model, import the Sonoff 4CH template from [Blackadder](https://templates.blakadder.com/sonoff_4CH.html) and play/test with it to nail down the issue.
Joerg

schrej-zz, 2021-03-07:
I just found a video about flashing 4CHP R3 with tasmota 8.31 and standard 4CH template.
So there seem to be no differences between older and newer boards with respect to tasmota.

Leox89, 2021-03-07:
> Blackadder

Hi @schrej-zz, with template from Blackadder behaviour is the same, not also on blackadder page is not mentioned R3

Leox89, 2021-03-07:
> I just found a video about flashing 4CHP R3 with tasmota 8.31 and standard 4CH template from [notenoughtech](https://youtu.be/1K1vTNzsTxI)
> So there seem to be no differences between older and newer boards with respect to tasmota.

Flashing is ok, template seems to be the problem, would like to specify mine is as printed on the board SONOFF_4CH_PRO_R3 V1.0 2020-04-13

schrej-zz, 2021-03-07:
@Leox89,
notenouchtech mentioned in his video, that 4CH R3 works without any issues with the standard template.
At least with tasmota 8.3.1 (current version as of date of video).
You can try with 8.3.1 and if you still have issues, I dought, that it's related to tasmota.
Joerg

Leox89, 2021-03-07:
@schrej-zz,
of course is not possibile to find 8.3.1 anymore. With 8.5.1 is the same. Can't understand where could be the problem.

Leox89, 2021-03-07:
@schrej-zz,
Of course is not possibile to find 8.3.1 anymore.
With 8.5.1 is working fine.
So, should I stay with this version? Could it be defined "an issue" that with newer fw version is not working?
Thanks

schrej-zz, 2021-03-07:
Good that you found 8.5.1 and that it'sworking, but of course  you can also download 8.3.1.
https://github.com/arendst/Tasmota/tree/v8.3.1
And yes, maybe an admin can change the topic.
Joerg

Noschvie, 2021-03-17:
My hardware "SONOFF_4CH_PRO_R3 V1.0 2020-04-13" is running Tasmota 9.3.1.1 and works.
http://ota.tasmota.com/tasmota/release/tasmota-DE.bin.gz

```
19:32:01.143 CMD: Backlog Template; Module
19:32:01.208 RSL: stat/tasmota_317F47/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
19:32:01.416 RSL: stat/tasmota_317F47/RESULT = {"Module":{"23":"Sonoff 4CH Pro"}}
```

ascillato2, 2021-03-22:
Hi, any news on this?

Leox89, 2021-03-29:
not for me

> Hi, any news on this?

ascillato, 2021-03-29:
> not for me

Please, can you reset your device with reset 6 command, and then update to latest dev version: http://ota.tasmota.com/tasmota/tasmota.bin.gz

And then, after that, set the module?

ascillato2, 2021-04-01:
Please, when you have time, tell us if you could perform the requested tests. Thanks.

closed 2021-04-01 by ascillato2