arendst/Tasmota #11259: Gosund SP112 (CSE7766 Model): Wrong total power readings

micw, 2021-03-08:
### PROBLEM DESCRIPTION

I'm running a Gosung SP112. It is calibrated and shows correct voltage, current and wattage. Connected to it are two waterbed heatings, each with 200 Watts and a power factor of 1. All readings looks correct, except the total/daily power consumptions which are about factor 5 too high.

Example:

[image]

This shows a usage of >9 kWh in 24 hours. In this time, the total power increased from 206 to 215 kWh. It shows peaks of >500 kWh per hour which is not possible with a 400 Watt workload.

I measured with a different device in parallel. This is showing me ~2 kWh of power usage in the same time (which is way more reasonable).

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Gosund SP112
- [X] Tasmota binary firmware version number used: 9.2.0
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: OTA (from previous version)

I will provided the following information soon (i bricked the device a few minutes ago when trying other templates).

- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE

Do a longer-term measurement of power consumption, compare with expected results

### EXPECTED BEHAVIOUR

Should be close to expected

### SCREENSHOTS

see above

### ADDITIONAL CONTEXT

none

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2021-03-08:
Please update to latest release version 9.3.1. Changes are made for the effects you encounter.

closed 2021-03-08 by ascillato2

micw, 2021-03-08:
I updated and will see the result in a few hours.

It looks like https://github.com/arendst/Tasmota/blame/master/tasmota/xnrg_02_cse7766.ino has not changed for a long while. Could you point me to the dupplicate issue or the code that changed?

Jason2866, 2021-03-08:
In energy driver https://github.com/arendst/Tasmota/blob/development/tasmota/xdrv_03_energy.ino

arendst, 2021-03-08:
Pls provide some logging by enabling ``weblog 4`` for 10 seconds if the load is around 400W

micw, 2021-03-08:
@arendst 9.3.1 actually does fix it for me. Readings are in sync with those from my power meter. If you still need logs, just let me know.

micw, 2021-03-08:
@Jason2866 Thank you. I did a brief review of the recent commits and still have no idea what exactly caused this issue and which change does it fix.

arendst, 2021-03-09:
No need for logs now. If it reoccurs pls send the logs.