arendst/Tasmota #11264: zbscan returns invalid jason

bmc-git, 2021-03-08:
Hi,

It seems that zbscan returns an invalid Json string:

{"ZbScan":["11":-43,"12":-44,"13":-78,"14":-33,"15":-21,"16":1,"17":5,"18":-44,"19":-78,"20":-33,"21":-85,"22":-40,"23":-45,"24":-87,"25":-88,"26":-56]}

instead of:

{"ZbScan":[{"11":-43},{"12":-44},{"13":-78},{"14":-33},{"15":-21},{"16":1},{"17":5},{"18":-44},{"19":-78},{"20":-33},{"21":-85},{"22":-40},{"23":-45},{"24":-87},{"25":-88},{"26":-56}]}

Is there missing { and } in the last ResponseAppend_P ?

void EnergyScanResults(void) {
  Response_P(PSTR("{\"" D_JSON_ZIGBEE_SCAN "\":["));
  for (uint32_t i = 0; i < USE_ZIGBEE_CHANNEL_COUNT; i++) {
    int8_t energy = zigbee.energy[i];

    if (i > 0) { ResponseAppend_P(PSTR(",")); }
    ResponseAppend_P(PSTR("\"%d\":%d"), i + USE_ZIGBEE_CHANNEL_MIN, energy);

ascillato, 2021-03-08:
Sorry. It is a valid JSON as it is.

bmc-git, 2021-03-08:
I did try different json parsers and they all give me an error, if I 
change the string as the proposed one if works.

jaymemaurice, 2021-03-08:
[ ] is an array... and array of objects would be as OP posted.
I think though the ["s are supposed to be {'s and the result is supposed to
be one object...

bmc-git, 2021-03-08:
Yes [ ] is an array, but I thought that each entry in the array would be 
[{"xyz":12}, {"twu": 11} ..... ]

I might be wrong, but parsers seems to agree ;o)

s-hadinger, 2021-03-08:
Indeed. The json is wrong. I will fix that.

sfromis, 2021-03-09:
Simplet way to make it valid would simply to replace the [] by {} without seeing it as an array

bmc-git, 2021-03-09:
Well in the loop it would be just as easy to add the { and }, would it 
not be 'correct' that it would be an array of channels ?

sfromis, 2021-03-09:
No clear "right" or "wrong" there. My view is that the "names" are already index numbers, making it slightly confusing to wrap it in an array with another set of index numbers.

s-hadinger, 2021-03-09:
My bad, I will just replace [] with {}

PR #11276 (2021-03-09): Fix ZbScan json output #11264
## Description:

**Related issue (if applicable):** fixes #11264
## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works on Tasmota core ESP8266 V.2.7.4.9
  - [x] The code change is tested and works with core ESP32 V.1.0.5
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreement-cla).

_NOTE: Th [... cut]
closed 2021-03-09 by s-hadinger