arendst/Tasmota #11296: Rule with long event parameter doesn't work, results in unkillable continuous beep on Sonoff Bridge

nagyrobi, 2021-03-11:
### PROBLEM DESCRIPTION
Trying to [create](https://github.com/arendst/Tasmota/discussions/11273#discussion-3256606) a rule to repeat a command with parameter, to run on Sonoff RF Bridge.

When that happens, the device keeps beeping until I remove power cord. Restarting from the web interface doesn't kill the beep.

### REQUESTED INFORMATION

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Bridge (RF)
- [x] Tasmota binary firmware version number used: 9.3.1.1
  - [x] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: Tasmotizer, OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
11:22:26.976 CMD: Backlog Template; Module; GPIO 255
11:22:27.003 MQT: stat/rf-bridge-2/RESULT = {"NAME":"Sonoff RF Bridge","GPIO":[32,3200,1,3232,1,1,0,0,1,320,1,0,0,0],"FLAG":0,"BASE":25}
11:22:27.224 MQT: stat/rf-bridge-2/RESULT = {"Module":{"0":"Sonoff RF Bridge"}}
11:22:27.481 MQT: stat/rf-bridge-2/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"3200":"Serial Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3232":"Serial Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
11:22:44.475 MQT: stat/rf-bridge-2/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":68,"Free":443,"Rules":"on event#rfrf do backlog RfRaw %value%; RfRaw %value%; RfRaw 0 endon"}
11:22:44.706 MQT: stat/rf-bridge-2/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
11:22:44.955 MQT: stat/rf-bridge-2/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
```
- [ ] Provide the output of this command: `Status 0`:
```lua
11:23:05.669 MQT: stat/rf-bridge-2/STATUS = {"Status":{"Module":0,"DeviceName":"RF 2","FriendlyName":["RF 2"],"Topic":"rf-bridge-2","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
11:23:05.678 MQT: stat/rf-bridge-2/STATUS1 = {"StatusPRM":{"Baudrate":19200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://dev-tasmobuilder.bubu/ota/tasmota.bin.gz","RestartReason":"Power On","Uptime":"0T00:20:00","StartupUTC":"2021-03-11T10:03:05","Sleep":50,"CfgHolder":4617,"BootCount":5,"BCResetTime":"2021-03-11T10:19:51","SaveCount":13,"SaveAddress":"F7000"}}
11:23:05.689 MQT: stat/rf-bridge-2/STATUS2 = {"StatusFWR":{"Version":"9.3.1.1(tasmota-robi)","BuildDateTime":"2021-03-11T10:17:18","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"392/699"}}
11:23:05.696 MQT: stat/rf-bridge-2/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["ZACUT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","000000C0","00006000","00000000"]}}
11:23:05.708 MQT: stat/rf-bridge-2/STATUS4 = {"StatusMEM":{"ProgramSize":419,"Free":584,"Heap":29,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","08000580","04000001","00000002","00000000","00002000","00000000","00000000","00000000"],"Drivers":"1,2,6,10","Sensors":"2"}}
11:23:05.717 MQT: stat/rf-bridge-2/STATUS5 = {"StatusNET":{"Hostname":"rf-bridge-2-6322","IPAddress":"172.22.215.212","Gateway":"172.22.215.254","Subnetmask":"255.255.255.0","DNSServer":"172.22.215.254","Mac":"84:0D:8E:66:18:B2","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
11:23:05.721 MQT: stat/rf-bridge-2/STATUS6 = {"StatusMQT":{"MqttHost":"172.22.215.250","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_6618B2","MqttUser":"hamqt","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
11:23:05.728 MQT: stat/rf-bridge-2/STATUS7 = {"StatusTIM":{"UTC":"2021-03-11T10:23:05","Local":"2021-03-11T11:23:05","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00"}}
11:23:05.736 MQT: stat/rf-bridge-2/STATUS10 = {"StatusSNS":{"Time":"2021-03-11T11:23:05"}}
11:23:05.745 MQT: stat/rf-bridge-2/STATUS11 = {"StatusSTS":{"Time":"2021-03-11T11:23:05","Uptime":"0T00:20:00","UptimeSec":1200,"Heap":29,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"ZACUT","BSSId":"F0:9F:C2:A4:1A:28","Channel":6,"RSSI":72,"Signal":-64,"LinkCount":1,"Downtime":"0T00:00:07"}}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
11:02:04.874 SRC: MQTT
11:02:04.877 CMD: Group 0, Index 1, Command "EVENT", Data "rfrf=AAB035050412FC061802C60186229CA481A3B2B2A3B2A3A3B2B2B2A3A3B2B2A3B2A3B2A3A3B2A...
11:02:04.884 MQT: stat/rf-bridge-2/RESULT = {"Event":"Done"}
11:02:04.904 RUL: EVENT#RFRF performs "backlog RfRaw AAB035050412FC061802C60186229CA481A3B2B2A3B2A3A3B2B2B2A3A3B2B2A3B2A3B2A3A3B2A3A3B2A3B2A3...
11:02:04.907 SRC: Rule
11:02:04.910 CMD: Group 0, Index 1, Command "BACKLOG", Data "RfRaw AAB035050412FC061802C60186229CA481A3B2B2A3B2A3A3B2B2B2A3A3B2B2A3B2A3B2A3A3...
11:02:04.922 SRC: Backlog
11:02:04.925 CMD: Group 0, Index 1, Command "RFRAW", Data "AAB035050412FC061802C60186229CA481A3B2B2A3B2A3A3B2B2B2A3A3B2B2A3B2A3B2A3A3B2A3A3B2...
11:02:04.934 MQT: stat/rf-bridge-2/RESULT = {"RfRaw":"ON"}
11:02:05.173 SRC: Backlog
11:02:05.176 CMD: Group 0, Index 1, Command "RFRAW", Data "AAB035050412FC061802C60186229CA481A3B2B2A3B2A3A3B2B2B2A3A3B2B2A3B2A3B2A3A3B2A3A3B2...
11:02:05.184 MQT: stat/rf-bridge-2/RESULT = {"RfRaw":"ON"}
11:02:05.444 SRC: Backlog
11:02:05.447 CMD: Group 0, Index 1, Command "RFRAW", Data "0"
11:02:05.455 MQT: stat/rf-bridge-2/RESULT = {"RfRaw":"OFF"}
```

### TO REPRODUCE
Passing to the `cmnd/rf-bridge-2/event` topic the payload `rfrt=AAB035050412FC061802C60186229CA481A3B2B2A3B2A3A3B2B2B2A3A3B2B2A3B2A3B2A3A3B2A3A3B2A3B2A3A3B2B2B2A3B2B2A3A3B2B2A355` (an RF code created earlier with BitBucketConverter.py)

On Sonoff RF Bridge have a rule `on event#rfrf do backlog RfRaw %value%; RfRaw %value%; RfRaw 0 endon`. The goal is to send twice the `RfRaw` command with the code specified in the event, and after that `RfRaw 0`.

### EXPECTED BEHAVIOUR
To run just like I manually enter `backlog RfRaw AAB035050412FC061802C60186229CA481A3B2B2A3B2A3A3B2B2B2A3A3B2B2A3B2A3B2A3A3B2A3A3B2A3B2A3A3B2B2B2A3B2B2A3A3B2B2A355; RfRaw AAB035050412FC061802C60186229CA481A3B2B2A3B2A3A3B2B2B2A3A3B2B2A3B2A3B2A3A3B2A3A3B2A3B2A3A3B2B2B2A3B2B2A3A3B2B2A355; RfRaw 0`, which works OK.

### SCREENSHOTS
Not applicable.

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2021-03-11:
A bit of testing (using event and publish) suggests a maximum length of name and value of ~100. Using a one-letter name, the value was truncated after 97 chars. Using a 4-letter name got me to 94 chars of value.

nagyrobi, 2021-03-11:
Well unfortunately these codes are longer, I have even 124 chars long codes...

sfromis, 2021-03-11:
That limitation (not a bug) has an easy workaround, splitting into two Event commands, first to store one half of the payload in a var, and second to issue the RfRaw referencing both the var and %value%.

nagyrobi, 2021-03-11:
I don't quite understand but can you help me fix my rule to handle this?

sfromis, 2021-03-11:
`on event#rfrf do backlog RfRaw %var1%%value%; RfRaw %var1%%value%; RfRaw 0 endon`
This assumes that you first set `var1` to the first part of the data, publishing a `var1` command, or doing it via an `Event` if you prefer.

ascillato2, 2021-03-11:
Closing this issue as it has been answered.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-03-11 by ascillato2

nagyrobi, 2021-03-11:
Unfortunately does not work. The symptom is the same.

sfromis, 2021-03-11:
If you want suggestions, it is never enough to just state "does not work".

nagyrobi, 2021-03-11:
What else can I provide? 
Here's the weblog4 output:
```
19:10:58.986 MQT: stat/rf-bridge-2/RESULT = {"WebLog":4}
19:10:59.640 CFG: Saved to flash at FA, Count 18, Bytes 4096
19:11:05.199 SRC: MQTT
19:11:05.202 CMD: Group 0, Index 1, Command "EVENT", Data "rfrf=AAB03505041306061802C6019022A6A481A3B2B2A3B2A3A3B2B2B2A3A3B2B2A3B2A3B2A3A3B2A...
19:11:05.208 MQT: stat/rf-bridge-2/RESULT = {"Event":"Done"}
19:11:05.226 RUL: EVENT#RFRF performs "backlog RfRaw AAB03505041306061802C6019022A6A481A3B2B2A3B2A3A3B2B2B2A3A3B2B2A3B2A3B2A3A3B2A3A3B2A3B2A3...
19:11:05.228 SRC: Rule
19:11:05.230 CMD: Group 0, Index 1, Command "BACKLOG", Data "RfRaw AAB03505041306061802C6019022A6A481A3B2B2A3B2A3A3B2B2B2A3A3B2B2A3B2A3B2A3A3...
19:11:05.248 SRC: Backlog
19:11:05.250 CMD: Group 0, Index 1, Command "RFRAW", Data "AAB03505041306061802C6019022A6A481A3B2B2A3B2A3A3B2B2B2A3A3B2B2A3B2A3B2A3A3B2A3A3B2...
19:11:05.256 MQT: stat/rf-bridge-2/RESULT = {"RfRaw":"ON"}
19:11:05.498 SRC: Backlog
19:11:05.501 CMD: Group 0, Index 1, Command "RFRAW", Data "AAB03505041306061802C6019022A6A481A3B2B2A3B2A3A3B2B2B2A3A3B2B2A3B2A3B2A3A3B2A3A3B2...
19:11:05.508 MQT: stat/rf-bridge-2/RESULT = {"RfRaw":"ON"}
19:11:05.749 SRC: Backlog
19:11:05.751 CMD: Group 0, Index 1, Command "RFRAW", Data "0"
19:11:05.757 MQT: stat/rf-bridge-2/RESULT = {"RfRaw":"OFF"}
19:11:07.074 DMP: A0 55
```

sfromis, 2021-03-11:
I specifically mentioned the need for splitting the data to have the first part in var1. I'm seeing nothing of that there.... Don't just ignore most of the suggestions you get ;-)

nagyrobi, 2021-03-11:
I'm sorry I used the rule you provided [above](https://github.com/arendst/Tasmota/issues/11296#issuecomment-796884332).
Can you please provide me with an example how to split the data?
The topic is: `topic: 'cmnd/rf-bridge-1/event'`
The payload currently is:
`rfrf=AAB035050412F2062202BC0190229CA481A3B2B2A3B2A3A3B2B2B2A3A3B2B2A3B2A3B2A3A3B2A3A3B2A3B2A3A3B2B2A3B2B2B2A3A3B2B2A355`
How should I split this?

nagyrobi, 2021-03-11:
???

nagyrobi, 2021-03-11:
I need to do this in a single MQTT payload...

nagyrobi, 2021-03-11:
Where in the code should I look for enlarging the size of the string to avoid truncation? I compile images for myself anyway.

sfromis, 2021-03-11:
You need to split it on the backend side. You can publish the Var1 command with the first 32 bytes, and the Event command in the same MQTT payload, using the Backlog command. Details depend on your backend environment. The rule I suggested was very specifically mentioned as depending on making that split. Again. do not ignore most of a post, to only do a bit of copy/paste.

sfromis, 2021-03-11:
The 100 bytes `event_data` buffer is defined in this struct:
https://github.com/arendst/Tasmota/blob/development/tasmota/xdrv_10_rules.ino#L164-L182

nagyrobi, 2021-03-12:
> You can publish the Var1 command with the first 32 bytes, and the Event command in the same MQTT payload

Unfortunately I don't quite understand how to do that, that's why I was specifically kindly asking for an example. Could you please provide me with one?

sfromis, 2021-03-12:
As I wrote, details depend on your backend environment. Outcome would be the single MQTT command:
Topic: `cmnd/rf-bridge-1/Backlog`
Payload: `var1 first-32-bytes-of-data; Event rfrf=the-rest-of data`
If you already store the Event command payload, you can just rewrite it to match the above.

nagyrobi, 2021-03-12:
Ok splitted to `var1 AAB035050412F2062202BC0190229CA4; Event 81A3B2B2A3B2A3A3B2B2B2A3A3B2B2A3B2A3B2A3A3B2A3A3B2A3B2A3A3B2B2A3B2B2B2A3A3B2B2A355`

Nothing happens... Result is 
```
16:03:02.690 MQT: stat/rf-bridge-2/RESULT = {"Var1":"AAB035050412F2062202BC0190229CA4"}
16:03:02.891 MQT: stat/rf-bridge-2/RESULT = {"Event":"Done"}
```

Rule is
```
16:04:53.531 CMD: rule1
16:04:53.538 MQT: stat/rf-bridge-2/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":80,"Free":431,"Rules":"on event#rfrf do backlog RfRaw %var1%%value%; RfRaw %var1%%value%; RfRaw 0 endon"}
```

sfromis, 2021-03-12:
Your Event command is invalid, you no longer include the event name. I already told you where the 100-character limit is in the code.

nagyrobi, 2021-03-12:
OK I see now. Thanks for your help.

I'll compile all this information in a readable and easy to understand cookbook.

nagyrobi, 2021-03-14:
Ok turns out this kind of repetition is useless. Problem still persists and is likely caused by switching back and forth the RfRaw sniffing with values and 0. 

Backlogs with `rfraw AAB0XXXXX....XXXXXXXXXX; rfraw 0` should be avoided as switching the sniff on and off quickly for every cover movement may cause issues. It's enough to send `rfraw 0` only once from HA system with some delay after all procedures related to cover movements finished.