arendst/Tasmota #11311: Modbus sends other register than requested

chrisbucher, 2021-03-12:
### PROBLEM DESCRIPTION
The modbus message sent by Tasmota is not the one expected. I am trying to send the "0103000E" command, but the sent command is "01 03 80 87 40 a0 1a fe". (Used another RS454 module and SerialPortMonitor on Windows to sniff )
I have tried changing the modbus ID, the function and the register. When doing so, the changed ID and function codes were correctly sent, but the register still is not the right one.

Used script:
----------------------------------------
>D
>B
=>sensor53 r
>M 1
+1,4,M,0,9600,modbus,5,25,0103000E
1,010308ffffffff@i0:1,Voltage L1,V,Voltage_L1,2
#
-----------------------------------------

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): wemos d1 mini
- [ ] Tasmota binary firmware version number used: v9.3.1
  - [ ] Pre-compiled
  - [x] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

gemu2015, 2021-03-12:
use last dev version and go to dump mode.
sensor53 d1
you will see bytes sent by driver marked by '>' and response from device

then report again

chrisbucher, 2021-03-12:
Thanks for helping! Ok, compiled using dev and reflashed.
It appears that the driver sends the right message, but something else is read by the rest of the devices.
I also get no response from the meter. But if I send the exact same bytes "01 03 00 0e 00 02 a5 c8" using Serial Port Monitor, I get a correct response from the meter.

Are the uart settings are ok? "M" should be 8E1. In Serial Port Monitor I used Databits:8, Stopbits:1, Parity:Even and got a correct response from the meter. 

[image]

gemu2015, 2021-03-12:
i assume you are using software serial on an esp8266 ?
8E1 is not supported here. you must use hardware serial.

chrisbucher, 2021-03-12:
Yes, I am using an esp8266 and had RX, TX on pins 4,5. I switched to using hardware serial (gpio 3,1) and it does work now, in mode 8E1
It would be a good idea to add this particularity to  [https://tasmota.github.io/docs/Smart-Meter-Interface/#meter-definition](url), as it is not mentioned anywhere.

Thanks for your help!

closed 2021-03-12 by chrisbucher