arendst/Tasmota #11335: Sonoff ZBBridge keeps rebooting after 17 seconds

ConfigMgrPro, 2021-03-13:
### PROBLEM DESCRIPTION
I flashed a new Sonoff ZBBridge using a Raspberry PI, the USB > Serial method did not work.
after configuring the WiFi (just the WiFi, no other config), the unit keeps rebooting after exactly 17 seconds.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): Sonoff Zigbee Bridge
- [ x] Tasmota binary firmware version number used: 9.3.1(zbbridge)
  - [x ] Pre-compiled
  - [- ] Self-compiled
- [ ] Flashing tools used: Tried USB Flashing with Tasmotizer, when connecting USB to the laptop the USB device crashed. Succesfully flashed using a Raspberry Pi 4b using esptool.
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
00:00:00.032 CFG: Loaded from flash at F7, Count 88
00:00:00.035 QPC: Count 1
00:00:00.038 ZIG: ZBBridge EEPROM found at address 0x50
00:00:00.039 SNS: Hardware Serial
00:00:00.040 SRC: Restart
00:00:00.040 Project tasmota Tasmota Version 9.3.1(zbbridge)-2_7_4_9(2021-03-09T16:13:51)
00:00:00.041 ZIG: Resetting EZSP device
00:00:00.192 WIF: Attempting connection...
00:00:00.528 WIF: Connecting to AP1 BlackHole Channel 6 BSSId 6C:99:61:E4:B6:66 in mode 11n as ZbBridge...
00:00:01.801 WIF: Attempting connection...
00:00:02.801 WIF: Connected
00:00:03.002 HTP: Web server active on ZbBridge with IP address 192.168.2.198
00:00:03.005 RSL: INFO1 = {"Module":"Sonoff ZbBridge","Version":"9.3.1(zbbridge)","FallbackTopic":"cmnd/DVES_A9C1B0_fb/","GroupTopic":"cmnd/tasmotas/"}
00:00:03.006 RSL: INFO2 = {"WebServerMode":"Admin","Hostname":"ZbBridge","IPAddress":"192.168.2.198"}
00:00:03.009 RSL: INFO3 = {"RestartReason":{"Exception":9,"Reason":"Exception","EPC":["4023ab37","00000000","00000000"],"EXCVADDR":"00004bca","DEPC":"00000000","CallChain":["4023a9b8","4020e85c","4020e855","4026afff","4026afff","401004a4","4026b04e","40268cdb","4026b132","40102b37","4026a0c4","4025375f","401062d3","40286c39","40244ae4","402470bd","40286c3e","40244ae4","40247346","40247764","401029fc","40105c8f","4024ddc1","4024e868","40103c1a","4024e8fa","4024efe7","4024b1b4","40100f3f","401063e9","4024b20b"]}}
00:00:04.483 NTP: No reply
00:00:04.505 RTC: UTC 2021-03-13T20:38:35, DST 2021-03-28T02:00:00, STD 2021-10-31T03:00:00
21:38:37.439 CMD: Backlog Template; Module; GPIO 255
21:38:37.440 SRC: WebConsole from 192.168.2.196
21:38:37.441 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
21:38:37.490 SRC: Backlog
21:38:37.491 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
21:38:37.493 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
21:38:37.535 QPC: Reset
21:38:37.698 SRC: Backlog
21:38:37.700 CMD: Group 0, Index 1, Command "MODULE", Data ""
21:38:37.702 RSL: RESULT = {"Module":{"75":"Sonoff ZbBridge"}}
21:38:37.949 SRC: Backlog
21:38:37.950 CMD: Group 0, Index 1, Command "GPIO", Data "255"
21:38:37.954 RSL: RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"3552":"Zigbee Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3584":"Zigbee Rx"},"GPIO4":{"5312":"Zigbee Rst"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"608":"I2C SCL"},"GPIO15":{"0":"None"},"GPIO16":{"32":"Button1"},"GPIO17":{"0":"None"}}
21:38:38.462 RSL: STATE = {"Time":"2021-03-13T21:38:38","Uptime":"0T00:00:09","UptimeSec":9,"Vcc":3.458,"Heap":30,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"xxxx","BSSId":"xx:xx:xx:xx:xx:xx","Channel":6,"RSSI":100,"Signal":-31,"LinkCount":1,"Downtime":"0T00:00:04"}}
21:38:39.450 APP: Boot Count 77
21:38:39.556 CFG: Saved to flash at 1FB, Count 89, Bytes 4096

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

00:00:03.256 CMD: status 0
00:00:03.257 SRC: WebConsole from 192.168.2.196
00:00:03.258 CMD: Group 0, Index 1, Command "STATUS", Data "0"
00:00:03.260 RSL: STATUS = {"Status":{"Module":75,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_A9C1B0","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
00:00:03.262 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Exception","Uptime":"0T00:00:04","StartupUTC":"","Sleep":50,"CfgHolder":4617,"BootCount":81,"BCResetTime":"2021-03-13T21:18:45","SaveCount":93,"SaveAddress":"FA000"}}
00:00:03.263 RSL: STATUS2 = {"StatusFWR":{"Version":"9.3.1(zbbridge)","BuildDateTime":"2021-03-09T16:13:51","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":160,"Hardware":"ESP8266EX","CR":"383/699"}}
[... 36 lines cut ...]
00:00:03.474 RTC: UTC 2021-03-13T20:41:34, DST 2021-03-28T02:00:00, STD 2021-10-31T03:00:00
21:41:37.479 QPC: Reset
21:41:38.442 RSL: STATE = {"Time":"2021-03-13T21:41:38","Uptime":"0T00:00:09","UptimeSec":9,"Vcc":3.458,"Heap":30,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":22,"MqttCount":0,"Wifi":{"AP":1,"SSId":"BlackHole","BSSId":"6C:99:61:E4:B6:66","Channel":6,"RSSI":100,"Signal":-31,"LinkCount":1,"Downtime":"0T00:00:04"}}
21:41:39.438 APP: Boot Count 88
21:41:39.593 CFG: Saved to flash at FA, Count 101, Bytes 4096

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Direct after flashing.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

No reboot

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2021-03-13:
Fixed in the latest `development` version. Upgrade from here: http://ota.tasmota.com/tasmota/

ConfigMgrPro, 2021-03-13:
Thank you!! Issue resolved!

closed 2021-03-13 by ConfigMgrPro