arendst/Tasmota #11337: Sonoff ZbBridge: Zigbee not started

ConfigMgrPro, 2021-03-13:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

When i click the button "zigbee Map" or enter the command 'ZbPermitJoin 1'

i get the message: "Zigbee not started" 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Sonoff ZBBridge
- [ ] Tasmota binary firmware version number used: 9.3.1.1(zbbridge)
  - [ x ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: ESPtool on Raspberry
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
22:49:41.197 CMD: Backlog Template; Module; GPIO 255
22:49:41.198 SRC: WebConsole from 192.168.2.196
22:49:41.200 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
22:49:41.264 SRC: Backlog
22:49:41.266 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
22:49:41.268 RSL: RESULT = {"NAME":"Sonoff ZbBridge","GPIO":[320,3552,0,3584,5312,0,0,0,640,576,608,0,32,0],"FLAG":0,"BASE":75}
22:49:41.473 SRC: Backlog
22:49:41.475 CMD: Group 0, Index 1, Command "MODULE", Data ""
22:49:41.477 RSL: RESULT = {"Module":{"0":"Sonoff ZbBridge"}}
22:49:41.725 SRC: Backlog
22:49:41.727 CMD: Group 0, Index 1, Command "GPIO", Data "255"
22:49:41.732 RSL: RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"3552":"Zigbee Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3584":"Zigbee Rx"},"GPIO4":{"5312":"Zigbee Rst"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"608":"I2C SCL"},"GPIO15":{"0":"None"},"GPIO16":{"32":"Button1"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
22:50:08.903 CMD: Status 0
22:50:08.905 SRC: WebConsole from 192.168.2.196
22:50:08.907 CMD: Group 0, Index 1, Command "STATUS", Data "0"
22:50:08.910 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"ZbBridge","FriendlyName":["ZbBridge"],"Topic":"tasmota_A9C1B0","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
22:50:08.912 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota-zbbridge.bin","RestartReason":"Software/System restart","Uptime":"0T00:09:33","StartupUTC":"2021-03-13T21:40:35","Sleep":50,"CfgHolder":4617,"BootCount":115,"BCResetTime":"2021-03-13T21:18:45","SaveCount":147,"SaveAddress":"F9000"}}
22:50:08.914 RSL: STATUS2 = {"StatusFWR":{"Version":"9.3.1.1(zbbridge)","BuildDateTime":"2021-03-13T14:10:49","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":160,"Hardware":"ESP8266EX","CR":"391/699"}}
22:50:08.916 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["BlackHole",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008001","2805C8000100060000005A00000000000000","00000080","00006000","00000000"]}}
22:50:08.920 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":574,"Free":1216,"Heap":29,"ProgramFlashSize":2048,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F1007C4","04400001","00000002","00000000","00000000","00020000","00000000","00000000"],"Drivers":"1,2,7,9,10,20,23,41","Sensors":""}}
22:50:08.923 RSL: STATUS5 = {"StatusNET":{"Hostname":"ZbBridge","IPAddress":"192.168.2.198","Gateway":"192.168.2.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.254","Mac":"84:CC:A8:A9:C1:B0","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
22:50:08.926 RSL: STATUS7 = {"StatusTIM":{"UTC":"2021-03-13T21:50:08","Local":"2021-03-13T22:50:08","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"07:07","Sunset":"18:51"}}
22:50:08.928 RSL: STATUS10 = {"StatusSNS":{"Time":"2021-03-13T22:50:08"}}
22:50:08.932 RSL: STATUS11 = {"StatusSTS":{"Time":"2021-03-13T22:50:08","Uptime":"0T00:09:33","UptimeSec":573,"Vcc":3.458,"Heap":29,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":21,"MqttCount":0,"Wifi":{"AP":1,"SSId":"BlackHole","BSSId":"6C:99:61:E4:B6:66","Channel":6,"RSSI":68,"Signal":-66,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
22:50:38.061 CMD: ZbPermitJoin 1
22:50:38.063 SRC: WebConsole from 192.168.2.196
22:50:38.065 CMD: Group 0, Index 1, Command "ZBPERMITJOIN", Data "1"
22:50:38.068 RSL: RESULT = {"ZbPermitJoin":"Zigbee not started"}
22:50:38.791 WIF: Checking connection...

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Run command: ZbPermitJoin 1

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2021-03-13:
Console messages at boot time should show details. Maybe you have not flashed firmware to the Zigbee chip, which is needed for Tasmota.

ConfigMgrPro, 2021-03-14:
I flashed using this method: https://zigbee.blakadder.com/Sonoff_ZBBridge.html

`00:00:00.032 CFG: Loaded from flash at FC, Count 151
00:00:00.035 QPC: Count 1
00:00:00.038 ZIG: ZBBridge EEPROM found at address 0x50
00:00:00.038 ZIG: GPIOs Rx:3 Tx:1
00:00:00.039 SNS: Hardware Serial
00:00:00.040 SRC: Restart
00:00:00.041 Project tasmota ZbBridge Version 9.3.1.1(zbbridge)-2_7_4_9(2021-03-13T14:10:49)
00:00:00.041 ZIG: Resetting EZSP device
00:00:00.191 WIF: Checking connection...
00:00:00.191 WIF: Attempting connection...
00:00:00.527 WIF: Connecting to AP1 BlackHole Channel 6 BSSId 6C:99:61:E4:B6:66 in mode 11n as ZbBridge...
00:00:01.753 WIF: Checking connection...
00:00:01.754 WIF: Attempting connection...
00:00:02.754 WIF: Checking connection...
00:00:02.754 WIF: Attempting connection...
00:00:03.754 WIF: Checking connection...
00:00:03.754 WIF: Connected
00:00:04.006 HTP: Web server active on ZbBridge with IP address 192.168.2.198
00:00:04.007 RSL: INFO1 = {"Module":"Sonoff ZbBridge","Version":"9.3.1.1(zbbridge)","FallbackTopic":"cmnd/DVES_A9C1B0_fb/","GroupTopic":"cmnd/tasmotas/"}
00:00:04.008 RSL: INFO2 = {"WebServerMode":"Admin","Hostname":"ZbBridge","IPAddress":"192.168.2.198"}
00:00:04.009 RSL: INFO3 = {"RestartReason":"Software/System restart"}
00:00:04.505 RTC: UTC 2021-03-14T06:36:51, DST 2021-03-28T02:00:00, STD 2021-10-31T03:00:00
07:36:53.495 QPC: Reset
07:36:54.460 HTP: Main Menu
07:36:55.417 APP: Boot Count 118
07:36:55.420 RSL: STATE = {"Time":"2021-03-14T07:36:55","Uptime":"0T00:00:10","UptimeSec":10,"Vcc":3.447,"Heap":30,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"BlackHole","BSSId":"6C:99:61:E4:B6:66","Channel":6,"RSSI":78,"Signal":-61,"LinkCount":1,"Downtime":"0T00:00:05"}}
07:36:55.581 CFG: Saved to flash at FB, Count 152, Bytes 4096
07:37:02.060 ZIG: Resetting EZSP device
07:37:02.436 ZbEZSPReceived: time-out, discarding FE0D0180020000000001010878
07:37:07.196 ZIG: timeout, goto label 99
07:37:07.198 RSL: RESULT = {"ZbState":{"Status":99,"Message":"Abort"}}
07:37:07.199 ZIG: Abort
07:37:07.199 ZIG: Stopping (99)
07:37:09.628 HTP: Console
07:37:10.756 WIF: Checking connection...`

i followed the "Flash zigbee module" part again:

`07:40:38.631 ZIG: {"ZbEZSPReceived":"100000"}
07:40:38.632 ZIG: ZbEZSPSend 55000003
07:40:38.633 ZIG: adding packet to_send, to_ack:1, to_send:1, to_end:1
07:40:38.676 ZIG: Something to_send, to_ack:1, to_send:1, to_end:2
07:40:38.678 ZIG: ZbEZSPSentRaw 1128000155000003
07:40:38.682 ZIG: {"ZbEZSPReceived2":"12288001550000"}
07:40:38.684 ZIG: new ack/data received, was 1 now 2
07:40:38.685 ZIG: freeing packet 1 from memory
07:40:38.687 ZIG: ZbEZSPSentRaw 82
07:40:38.688 ZIG: {"ZbEZSPReceived":"550000"}
07:40:38.689 ZIG: ZbEZSPSend 55000220
07:40:38.691 ZIG: adding packet to_send, to_ack:2, to_send:2, to_end:2
07:40:38.733 ZIG: Something to_send, to_ack:2, to_send:2, to_end:3
07:40:38.734 ZIG: ZbEZSPSentRaw 2229000155000220
07:40:38.738 ZIG: {"ZbEZSPReceived2":"23298001550000"}
07:40:38.740 ZIG: new ack/data received, was 2 now 3
07:40:38.741 ZIG: freeing packet 2 from memory
07:40:38.742 ZIG: ZbEZSPSentRaw 83
07:40:38.744 ZIG: {"ZbEZSPReceived":"550000"}
07:40:38.745 ZIG: ZbEZSPSend 55000330
07:40:38.746 ZIG: adding packet to_send, to_ack:3, to_send:3, to_end:3
07:40:38.789 ZIG: Something to_send, to_ack:3, to_send:3, to_end:4
07:40:38.791 ZIG: ZbEZSPSentRaw 332A000155000330
07:40:38.795 ZIG: {"ZbEZSPReceived2":"342A8001550000"}
07:40:38.797 ZIG: new ack/data received, was 3 now 4
07:40:38.798 ZIG: freeing packet 3 from memory
07:40:38.799 ZIG: ZbEZSPSentRaw 84
07:40:38.800 ZIG: {"ZbEZSPReceived":"550000"}
07:40:38.802 ZIG: ZbEZSPSend 55000551
07:40:38.803 ZIG: adding packet to_send, to_ack:4, to_send:4, to_end:4
[... 119 lines cut ...]
07:43:10.562 ZIG: freeing packet 4 from memory
07:43:10.564 ZIG: ZbEZSPSentRaw 85
07:43:10.565 ZIG: {"ZbEZSPReceived":"550000"}
07:43:10.566 ZIG: ZbEZSPSend 55000660
07:43:10.567 ZIG: adding packet to_send, to_ack:5, to_send:5, to_end:5
07:43:10.610 ZIG: Something to_send, to_ack:5, to_send:5, to_end:6
07:43:10.612 ZIG: ZbEZSPSentRaw 5515000155000660
07:43:10.616 ZIG: {"ZbEZSPReceived2":"56158001550000"}
07:43:10.618 ZIG: new ack/data received, was 5 now 6
07:43:10.619 ZIG: freeing packet 5 from memory
07:43:10.620 ZIG: ZbEZSPSentRaw 86
07:43:10.621 ZIG: {"ZbEZSPReceived":"550000"}
07:43:10.623 ZIG: ZbEZSPSend 17000000
07:43:10.624 ZIG: adding packet to_send, to_ack:6, to_send:6, to_end:6
07:43:10.667 ZIG: Something to_send, to_ack:6, to_send:6, to_end:7
07:43:10.668 ZIG: ZbEZSPSentRaw 6616000117000000
07:43:10.672 ZIG: {"ZbEZSPReceived2":"67168001170093"}
07:43:10.673 ZIG: new ack/data received, was 6 now 7
07:43:10.674 ZIG: freeing packet 6 from memory
07:43:10.675 ZIG: ZbEZSPSentRaw 87
07:43:10.677 ZIG: {"ZbEZSPReceived":"170093"}
07:43:10.679 RSL: RESULT = {"ZbState":{"Status":2,"Message":"Resetting configuration"}}
07:43:14.786 WIF: Checking connection...`

still no change :(

s-hadinger, 2021-03-14:
Please use loglevel 3. Parts of you logs are missing important information (the buffer overflowed before the web page refreshed).

Why is you PanID 0xFFFF ? This is not a valid PanID, this is why EZSP refuses to form the network.

ascillato2, 2021-03-22:
Hi, any news on this?

ConfigMgrPro, 2021-03-22:
I managed to get the zbbridge working.

1. Reset configuration, configure wifi
2. enter the following command: Backlog Weblog 3; so65 1; Module 75
3. upload the chipset firmware.

performing these steps 3 times over and over caused the zbbridge to work.

closed 2021-03-22 by ConfigMgrPro

jroake99, 2022-01-17:
Need to re-awaken this issue.
I cannot get the ZbBridge to start!
I have read this issue and variants and none of the solutions work. Have repeated many times and still no start.
Hope community can assist as must be missing the obvious.
I  purchased a ZbBridge and flashed to tasmota and followed all instructions as per : https://zigbee.blakadder.com/Sonoff_ZBBridge.html.

Sonoff ZbBridge
Tasmota
Program Version	10.1.0(zbbridge)
Build Date & Time	2021-12-08T14:47:37
Core/SDK Version	2_7_4_9/2.2.2-dev(38a443e)
Uptime	0T00:04:31
Flash write Count	79 at 0x1FB000
Boot Count	34
Restart Reason	Software/System restart
 
AP1 SSId (RSSI)	media-ap1 (86%, -57 dBm) 11n
Hostname	tasmota-946D39-3385
MAC Address	84:CC:A8:94:6D:39
IP Address (wifi)	192.168.1.107
....
ESP Chip Id	9727289 (ESP8266EX)
Flash Chip Id	0x1540A1
Flash Size	2048 kB
Program Flash Size	2048 kB
Program Size	671 kB
Free Program Space	1120 kB
Free Memory	26.8 kB

I successfully loaded tasmota using using both the esptool and Tasmotizer, configured AP, set module to 75.

06:52:28.819 CMD: module
06:52:28.820 SRC: WebConsole from 192.168.1.138
06:52:28.822 CMD: Grp 0, Cmnd 'MODULE', Idx 1, Len 0, Data ''
06:52:28.824 RSL: RESULT = {"Module":{"75":"Sonoff ZbBridge"}}

Went to file upgrade using : ncp-uart-sw_6.7.8_115200.ota 187kb from github

Set log levels on web 4, syslog 4.
Status before load:
06:56:34.780 CMD: status
06:56:34.781 SRC: WebConsole from 192.168.1.138
06:56:34.783 CMD: Grp 0, Cmnd 'STATUS', Idx 1, Len 0, Data ''
06:56:34.786 RSL: STATUS = {"Status":{"Module":75,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_946D39","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
06:56:47.234 WIF: Checking connection...

Upload started on web menu, then goes blank due to site not available in a few secs.

results in.

06:59:02.390 RSL: INFO1 = {"Info1":{"Module":"Sonoff ZbBridge","Version":"10.1.0(zbbridge)","FallbackTopic":"cmnd/DVES_946D39_fb/","GroupTopic":"cmnd/tasmotas/"}}
06:59:02.391 RSL: INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-946D39-3385","IPAddress":"192.168.1.107"}}
06:59:02.396 RSL: INFO3 = {"Info3":{"RestartReason":{"Exception":4,"Reason":"Software Watchdog","EPC":["400043f0","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["4010667f","4000410f","40004a3c","40106910","4020b888","402462fa","4020edf1","40234258","4024c334","402052f4","4024c334","4024c2d6","40205274","401000e1","40205274","402197a5","402480fd","402257b5","4023d620","40219774","4023d70e","40225c71","4022565e","4024c610","40219774","40226582","40248100","401014bb","40100da6","40228545","4024e050"]}}}
06:59:02.399 TFS: File 'autoexec.bat' not found
06:59:06.905 RSL: STATE = {"Time":"2022-01-17T06:59:06","Uptime":"0T00:00:07","UptimeSec":7,"Vcc":3.543,"Heap":29,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"media-ap1","BSSId":"60:E3:27:B5:9B:EC","Channel":6,"Mode":"11n","RSSI":90,"Signal":-55,"LinkCount":1,"Downtime":"0T00:00:01"}}
06:59:07.953 QPC: Reset
06:59:10.893 APP: Boot Count 35
06:59:11.194 CFG: Saved to flash at 1FB, Count 81, Bytes 4096
06:59:12.708 RSL: config = {"ip":"192.168.1.107","dn":"Tasmota","fn":["Tasmota",null,null,null,null,null,null,null],"hn":"tasmota-946D39-3385","mac":"84CCA8946D39","md":"Sonoff ZbBridge","ty":0,"if":0,"ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"10.1.0","t":"tasmota_946D39","ft":"%prefix%/%topic%/","tp":["cmnd","stat","tele"],"rl":[0,0,0,0,0,0,0,0],"swc":[-1,-1,-1,-1,-1,-1,-1,-1],"swn":[null,null,null,null,null,null,null,null],"btn":[0,0,0,0,0,0,0,0],"so":{"4":0,"11":0,"13":0,"17":0,"20":0,"30":0,"68":0,"73":0,"82":0,"114":0,"117":0},"lk":0,"lt_st":0,"sho":[0,0,0,0],"ver":1}
06:59:12.709 RSL: sensors = {"sn":{"Time":"2022-01-17T06:59:12"},"ver":1}
06:59:16.516 ZIG: Resetting EZSP device
06:59:16.906 ZbEZSPReceived: time-out, discarding FE0D0180020000000001010878
06:59:21.703 ZIG: timeout, goto label 99
06:59:21.705 RSL: RESULT = {"ZbState":{"Status":99,"Message":"Abort"}}
06:59:21.706 ZIG: Abort
06:59:21.706 ZIG: Stopping (99)
06:59:23.055 WIF: Checking connection...

Just to check again using Zigbee Permit Join command, result is:
07:01:09.910 HTP: Main Menu
07:01:12.551 SRC: WebGui from 192.168.1.138
07:01:12.552 CMD: Grp 0, Cmnd 'ZBPERMITJOIN', Idx 1, Len 0, Data ''
07:01:12.555 RSL: RESULT = {"ZbPermitJoin":"Zigbee not started"}

Any ideas what I'm missing or how to test?
How do I know if zigbee s/w loaded?
Can I test through command line?

I note that if I press the pairing button once the green led comes on, then off on next press resulting in:
07:14:33.899 WIF: Checking connection...
07:14:51.542 APP: Button1 multi-press 1
07:14:52.043 SRC: Button
07:14:52.045 RSL: RESULT = {"POWER":"ON"}
07:14:52.046 RSL: POWER = ON
07:14:52.893 CFG: Saved to flash at F7, Count 100, Bytes 4096
07:14:55.926 WIF: Checking connection...
07:14:59.511 APP: Button1 multi-press 1
07:15:00.014 SRC: Button
07:15:00.016 RSL: RESULT = {"POWER":"OFF"}
07:15:00.017 RSL: POWER = OFF
07:15:00.078 CFG: Saved to flash at 1FB, Count 101, Bytes 4096

Six quick pushes puts it into setting up the AP as expected.

I've got a bag full of sensors to attach, that worked with out fo the box Sonoff software waiting to get integrated into Home Assistant.

I'm new at this so apologies for any questions that might be obvious, I can usually nut them out but his has got very frustrating.

Thanks.

GO74, 2022-07-18:
> Need to re-awaken this issue. I cannot get the ZbBridge to start! I have read this issue and variants and none of the solutions work. Have repeated many times and still no start. Hope community can [...]

Same here! No clue what to do or try next.

s-hadinger, 2022-07-18:
> Same here! No clue what to do or try next.

Please re-open an issue with logs and required information

tracetechnical, 2022-08-09:
In my case, my NTP was failing to resolve, locking up the processor somehow and causing it to miss the Zigbee module reply. Fixed my DNS and the module booted as expected.

trichromic-ab, 2022-11-10:
I have just purchased another Sonoff bridge today (board says "v1.1 2020.02.28)
The latest tasmota-zbbridge.bin (v12.2.0) is failing to start Zigbee. The Zigbee Permit Join and the Zigbee Map buttons are disabled. The console says : 
`22:24:53.862 ZIG: Resetting EZSP device`
`22:24:59.017 ZIG: timeout, goto label 99`
`22:24:59.023 MQT: tele/tasmota-zb5/RESULT = {"ZbState":{"Status":99,"Message":"Abort"}}`
`22:24:59.024 ZIG: Abort`
`22:24:59.025 ZIG: Stopping (99)`

If I flash with the previous v12.1.1 then Zigbee starts fine.

Strangely though, all my other 4 Sonoff bridges were updated to v12.2.0 last week and are working fine.

trichromic-ab, 2022-11-10:
> If I flash with the previous v12.1.1 then Zigbee starts fine.

Sorry, I'm wrong. Zigbee is not starting with v12.1.1 either and I've gone back to 11.0.0 and its the same situation.
The only difference with 12.2.0 is that the Zigbee buttons are black and disabled, whereas with previous versions, they remain blue. I guess this is a new feature to indicate the Zigbee status.

So my issue is something else....

trichromic-ab, 2022-11-10:
Ok, for anyone that comes here after me. This was all down to me completely forgetting to do the Zigbee module flash part (ncp-uart-sw_6.7.9_115200.ota) , **_DOH!_**
Note to self - follow the very well written guide here and don't just rely on poor memory of previous device set up procedure.
https://zigbee.blakadder.com/Sonoff_ZBBridge.html

All working now :-)

oywino, 2023-10-30:
Well, all these fine guides tells you exactly WHAT to do, but they don't always tell you WHY.
I.e. is it necessary to add the ZHA Home Assistant template if the Bridge is only going to be used with an MQTT broker using Z2M?

sfromis, 2023-10-30:
If the bridge is going communicate via a MQTT broker, that cannot be Z2M, but only Z2T. If what you mean is instead that the bridge should connect to Z2M via TCP/IP, that is indeed the same as with ZHA. As a ser2net tunnel to the serial pins of the Zigbee chip is used via TCP/IP, it makes no difference to Tasmota what's on the other end.

oywino, 2023-10-31:
Z2T ???  And what might that be?

sfromis, 2023-10-31:
Zigbee2Tasmota, where Tasmota is handling Zigbee via MQTT publish/subscribe, somewhat similar to Z2M.

oywino, 2023-10-31:
The signal chain that I had in mind is:  Zigbee --> Tasmota Bridge --> Z2M ----> MQTT ---> Home Assistant
Are you saying that the following as an optional path:  Zigbee ---> Tasmota Bridge ---> MQTT ----> Home Assistant
Which implies dropping Z2M from the chain ?

sfromis, 2023-10-31:
Yes, both paths are viable. And there is also a third path: Zigbee -> Tasmota Bridge -> Home Assistant with ZHA.
(With both ZHA and Z2M, Tasmota is providing nothing more than a ser2net tunnel to the Zigbee chip).

Do note that the paths with ZHA and Z2M have integrations in HA. The path with Z2T has no ready to use HA integration, thus would take manual configuration.

I'm using Z2T, as my setup does not include HA. Then adding Z2M makes the path longer, and Z2T aims at producing similar payloads to Z2M. Still, Z2M does have more end devices recognized out of the box, where Z2T sometimes takes extra steps to get stuff well integrated.