arendst/Tasmota #11347: ESP32 string length overflow bug in xdrv_12_home_assistant.ino

vic42, 2021-03-15:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

ESP32 string length overflow bug in xdrv_12_home_assistant.ino

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Here's the invalid JSON that tasmota was sending (extra comma after last null):
2021-03-13 11:10:21 WARNING (MainThread) [hatasmota.discovery] Invalid discovery message 3C6105******: 
```lua
'{"ip":"192.168.1.220","dn":"Tasmota-Heltec","fn":["Tasmota1",null,null,null,null,null,null,null],"hn":"HeltecTopic-0724","mac":"3C6105******","md":"Heltec WiFi Kit 32","ty":0,"if":0,"ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"9.3.1","t":"HeltecTopic","ft":"%prefix%/%topic%/","tp":["cmnd","stat","tele"],"rl":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"swc":[-1,1,1,1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1],"swn":[null,"Switch2","Switch3","Switch4",null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,],"btn":[0,0,0,0,0,0,0,0],"so":{"4":0,"11":0,"13":0,"17":0,"20":0,"30":0,"68":0,"73":0,"82":0,"114":1,"117":0},"lk":1,"lt_st":0,"sho":[0,0,0,0],"ver":1}'
```
**Note the end of the `"swn"` value: `null,null,],` with the extra comma before the bracket**

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

A valid and complete JSON string

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

I am only reporting this as a proxy from Tasmota Discord chat. See thread here: 
https://discord.com/channels/479389167382691863/790185752855052288/820353710185906219
There is a good description of what is actually going wrong. I checked the code and found the cause of the problem. It was introduced when extending the number of possible switches in tasmota32.

stemp4 is only TOPSZ (151 Bytes) long. The resulting JSON string in this code becomes longer than that on a regular basis:
https://github.com/arendst/Tasmota/blob/development/tasmota/xdrv_12_home_assistant.ino#L302-L308

**(Please, remember to close the issue when the problem has been addressed)**

scottcopus, 2021-03-15:
@vic42 Thanks for opening this bug. I finally had some time to submit this bug report when I saw you beat me to the punch.   👍

arendst, 2021-03-15:
Thx for documented reporting.

closed 2021-03-15 by vic42