arendst/Tasmota #11349: TuyaMCU dimmer turns back on when turning off using hardware button

mvanlijden, 2021-03-15:
### PROBLEM DESCRIPTION
When using the dimmer function, turning off the lights results in a flash of the power and connection light and the light turning back on immediately. Running the latest Tasmota which should have this fixed according to, but does not provide a solution. https://github.com/arendst/Tasmota/issues/9864

The strange part is that i have two exactly the same dimmers connected to rows of dimmable LED lights. Both have received the same config via copy paste, with only a name difference. Only one of the dimmers is showing this behavior. Two other dimmers in the house also has this issue.

A workaround for this seems to be to change the dimmer setting using the hardware buttons up and down once. After that the on/off switch will work as it should.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): WF-DS01
- [X] Tasmota binary firmware version number used: 9.3.1.1
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: Tuya-Convert / OTA
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
08:53:12.629 CMD: Backlog Template; Module; GPIO 255
08:53:12.632 SRC: WebConsole from 192.168.142.156
08:53:12.636 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
08:53:12.693 SRC: Backlog
08:53:12.696 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
08:53:12.704 MQT: stat/werkkamer-licht-kast_003DFE/RESULT = {"NAME":"Dimmer WF-DS01","GPIO":[1,1,1,1,1,1,0,0,1,1,290,1,1,0],"FLAG":0,"BASE":54}
08:53:12.920 SRC: Backlog
08:53:12.923 CMD: Group 0, Index 1, Command "MODULE", Data ""
08:53:12.928 MQT: stat/werkkamer-licht-kast_003DFE/RESULT = {"Module":{"0":"Dimmer WF-DS01"}}
08:53:13.169 SRC: Backlog
08:53:13.173 CMD: Group 0, Index 1, Command "GPIO", Data "255"
08:53:13.181 MQT: stat/werkkamer-licht-kast_003DFE/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"2272":"Tuya Tx"},"GPIO2":{"0":"None"},"GPIO3":{"2304":"Tuya Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"290":"Led3"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
08:54:31.492 CMD: status 0
08:54:31.495 SRC: WebConsole from 192.168.142.156
08:54:31.498 CMD: Group 0, Index 1, Command "STATUS", Data "0"
08:54:31.506 MQT: stat/werkkamer-licht-kast_003DFE/STATUS = {"Status":{"Module":0,"DeviceName":"Werkkamer Licht Kast","FriendlyName":["Werkkamer Licht Kast"],"Topic":"werkkamer-licht-kast_003DFE","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
08:54:31.513 MQT: stat/werkkamer-licht-kast_003DFE/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"9T21:36:06","StartupUTC":"2021-03-05T10:18:25","Sleep":50,"CfgHolder":4617,"BootCount":9,"BCResetTime":"2021-03-04T10:09:07","SaveCount":335,"SaveAddress":"FB000"}}
08:54:31.521 MQT: stat/werkkamer-licht-kast_003DFE/STATUS2 = {"StatusFWR":{"Version":"9.3.1.1(tasmota)","BuildDateTime":"2021-03-05T08:56:06","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"425/699"}}
08:54:31.526 MQT: stat/werkkamer-licht-kast_003DFE/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["RoseMeshSmart-2G",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000280","00006000","00000000"]}}
08:54:31.539 MQT: stat/werkkamer-licht-kast_003DFE/STATUS4 = {"StatusMEM":{"ProgramSize":593,"Free":408,"Heap":22,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
08:54:31.549 MQT: stat/werkkamer-licht-kast_003DFE/STATUS5 = {"StatusNET":{"Hostname":"werkkamer-licht-kast_003DFE-767","IPAddress":"192.168.142.151","Gateway":"192.168.142.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.142.1","Mac":"D8:BF:C0:00:3D:FE","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
08:54:31.554 MQT: stat/werkkamer-licht-kast_003DFE/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.142.10","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_003DFE","MqttUser":"tasmota","MqttCount":313,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
08:54:31.562 MQT: stat/werkkamer-licht-kast_003DFE/STATUS7 = {"StatusTIM":{"UTC":"2021-03-15T07:54:31","Local":"2021-03-15T08:54:31","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"07:02","Sunset":"18:54"}}
08:54:31.569 MQT: stat/werkkamer-licht-kast_003DFE/STATUS10 = {"StatusSNS":{"Time":"2021-03-15T08:54:31"}}
08:54:31.577 MQT: stat/werkkamer-licht-kast_003DFE/STATUS11 = {"StatusSTS":{"Time":"2021-03-15T08:54:31","Uptime":"9T21:36:06","UptimeSec":855366,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":45,"MqttCount":313,"POWER":"OFF","Dimmer":74,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"RoseMeshSmart-2G","BSSId":"E6:63:DA:AD:18:CC","Channel":1,"RSSI":88,"Signal":-56,"LinkCount":6,"Downtime":"0T03:22:32"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
08:59:14.378 {"TuyaReceived":{"Data":"55AA0007000501010001000E","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
08:59:14.381 TYA: fnId=11 is set for dpId=1
08:59:14.383 TYA: RX Relay-1 --> MCU State: Off Current State:On
08:59:14.386 SRC: Switch
08:59:14.404 TYA: Send "55aa00060008020200040000004a5f"
08:59:14.414 MQT: tele/werkkamer-licht-kast_003DFE/STATE = {"Time":"2021-03-15T08:59:14","Uptime":"9T21:40:49","UptimeSec":855649,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":99,"MqttCount":313,"POWER":"OFF","Dimmer":74,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"RoseMeshSmart-2G","BSSId":"E6:63:DA:AD:18:CC","Channel":1,"RSSI":82,"Signal":-59,"LinkCount":6,"Downtime":"0T03:22:32"}}
08:59:14.419 MQT: stat/werkkamer-licht-kast_003DFE/RESULT = {"POWER":"OFF"}
[... 12 lines cut ...]
08:59:14.464 MQT: stat/werkkamer-licht-kast_003DFE/RESULT = {"POWER":"ON"}
08:59:14.468 MQT: stat/werkkamer-licht-kast_003DFE/POWER = ON
08:59:14.471 {"TuyaReceived":{"Data":"55AA0007000802020004000000485E","Cmnd":7,"CmndData":"0202000400000048","DpType2Id2":72,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000048"}}}
08:59:14.474 TYA: fnId=21 is set for dpId=2
08:59:14.477 TYA: RX value 72 from dpId 2 

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Restart dimmer
Turn on using software
Turn off using hardware button

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Light stays off

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

jaymemaurice, 2021-03-17:
Interesting you have different behaviour with same configuration. I wonder if we can get the TuyaMCU firmware version... 

Can you send just the TuyaMCU command outputs from both dimmers and TuyaReceived command output while attempting to reproduce on the working dimmer?

jaymemaurice, 2021-03-17:
Also I think your weblog 4 logs might be incomplete as compared to your instructions to reproduce.

From the logs it seems that you start with the light on, or at least you started by restarting the light while it was previously in the on state... then it went off, it's brightness was then increased while it was turned on.... 

08:59:14.378 {"TuyaReceived":{"Data":"55AA0007000501010001000E","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":
MCU says relay is now off... Tasmota thought it was previously on.... 

15T08:59:14","Uptime":"9T21:40:49","UptimeSec":855649,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":99,"MqttCount":313,"POWER":"OFF","Dimmer":74,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"RoseMeshSmart-2G","BSSId":"E6:63:DA:AD:18:CC","Channel":1,"RSSI":82,"Signal":-59,"LinkCount":6,"Downtime":"0T03:22:32"}}
08:59:14.419 MQT: stat/werkkamer-licht-kast_003DFE/RESULT = {"POWER":"OFF"}
08:59:14.423 MQT: stat/werkkamer-licht-kast_003DFE/POWER = OFF
MQTT message sent from TASMOTA saying switch is OFF

08:59:14.427 {"TuyaReceived":{"Data":"55AA0007000802020004000000485E","Cmnd":7,"CmndData":"0202000400000048","DpType2Id2":72,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000048"}}}
08:59:14.430 TYA: fnId=21 is set for dpId=2
08:59:14.432 TYA: RX value 72 from dpId 2 
MCU reports Dimmer value is 72. Probably irrelevant

08:59:14.435 {"TuyaReceived":{"Data":"55AA00070008020200040000004A60","Cmnd":7,"CmndData":"020200040000004A","DpType2Id2":74,"2":{"DpId":2,"DpIdType":2,"DpIdData":"0000004A"}}}
08:59:14.438 TYA: fnId=21 is set for dpId=2
08:59:14.440 TYA: RX value 74 from dpId 2 
MCU reports Dimmer value is 74 - Looks like someone pushed the dimmer button

08:59:14.443 {"TuyaReceived":{"Data":"55AA0007000501010001010F","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
08:59:14.446 TYA: fnId=11 is set for dpId=1
08:59:14.448 TYA: RX Relay-1 --> MCU State: On Current State:Off
08:59:14.451 SRC: Switch
MCU reports state is now ON

08:59:14.459 MQT: tele/werkkamer-licht-kast_003DFE/STATE = {"Time":"2021-03-15T08:59:14","Uptime":"9T21:40:49","UptimeSec":855649,"Heap":23,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":313,"POWER":"ON","Dimmer":74,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"RoseMeshSmart-2G","BSSId":"E6:63:DA:AD:18:CC","Channel":1,"RSSI":82,"Signal":-59,"LinkCount":6,"Downtime":"0T03:22:32"}}
08:59:14.464 MQT: stat/werkkamer-licht-kast_003DFE/RESULT = {"POWER":"ON"}
08:59:14.468 MQT: stat/werkkamer-licht-kast_003DFE/POWER = ON
Tasmota sends state to MQTT that switch is now ON

08:59:14.471 {"TuyaReceived":{"Data":"55AA0007000802020004000000485E","Cmnd":7,"CmndData":"0202000400000048","DpType2Id2":72,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000048"}}}
08:59:14.474 TYA: fnId=21 is set for dpId=2
MCU reports to Tasmota that dimmer is value 72

mvanlijden, 2021-03-18:
> Interesting you have different behaviour with same configuration. I wonder if we can get the TuyaMCU firmware version...
> 
> Can you send just the TuyaMCU command outputs from both dimmers and TuyaReceived command output while attempting to reproduce on the working dimmer?

And now somehow both my dimmers have this behavior. I did see a difference in the dim level between both (don't use it that much to be honest), so that might be something that explains the difference. But currently both act wrong

mvanlijden, 2021-03-18:
Here is the output from the dimmer that most frequently had the issue. Steps I took (if i got i right..)
- Deactive light via MQTT command using Tasmota MQTT app on Homey (so software off)
- Activate light via MQTT command using Tasmota MQTT app on Homey (so software on)
- Deactivate switch with one push

Home Assistant (192,.168.142.14) might be polluting the logs as the seem to check status very often.
```
10:49:42.540 MQT: stat/werkkamer-licht-kast_003DFE/STATUS10 = {"StatusSNS":{"Time":"2021-03-18T10:49:42"}}
10:49:42.547 MQT: stat/werkkamer-licht-kast_003DFE/STATUS11 = {"StatusSTS":{"Time":"2021-03-18T10:49:42","Uptime":"0T01:18:07","UptimeSec":4687,"Heap":24,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":2,"POWER":"ON","Dimmer":94,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"RoseMeshSmart-2G","BSSId":"E6:63:DA:AD:18:CC","Channel":1,"RSSI":94,"Signal":-53,"LinkCount":1,"Downtime":"0T00:00:02"}}}
10:49:46.310 HTP: Console
10:49:47.385 HTP: Command
10:49:47.390 SRC: WebCommand from 192.168.142.14
10:49:47.393 CMD: Group 0, Index 1, Command "STATUS", Data "0"
10:49:47.399 MQT: stat/werkkamer-licht-kast_003DFE/STATUS = {"Status":{"Module":0,"DeviceName":"Werkkamer Licht Kast","FriendlyName":["Werkkamer Licht Kast"],"Topic":"werkkamer-licht-kast_003DFE","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
10:49:47.407 MQT: stat/werkkamer-licht-kast_003DFE/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T01:18:12","StartupUTC":"2021-03-18T08:31:35","Sleep":50,"CfgHolder":4617,"BootCount":13,"BCResetTime":"2021-03-04T10:09:07","SaveCount":393,"SaveAddress":"FB000"}}
10:49:47.413 MQT: stat/werkkamer-licht-kast_003DFE/STATUS2 = {"StatusFWR":{"Version":"9.3.1.1(tasmota)","BuildDateTime":"2021-03-05T08:56:06","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"425/699"}}
10:49:47.418 MQT: stat/werkkamer-licht-kast_003DFE/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["RoseMeshSmart-2G",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000280","00006000","00000000"]}}
10:49:47.429 MQT: stat/werkkamer-licht-kast_003DFE/STATUS4 = {"StatusMEM":{"ProgramSize":593,"Free":408,"Heap":23,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
10:49:47.435 MQT: stat/werkkamer-licht-kast_003DFE/STATUS5 = {"StatusNET":{"Hostname":"werkkamer-licht-kast_003DFE-767","IPAddress":"192.168.142.151","Gateway":"192.168.142.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.142.1","Mac":"D8:BF:C0:00:3D:FE","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
10:49:47.439 MQT: stat/werkkamer-licht-kast_003DFE/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.142.10","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_003DFE","MqttUser":"tasmota","MqttCount":2,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
10:49:47.447 MQT: stat/werkkamer-licht-kast_003DFE/STATUS7 = {"StatusTIM":{"UTC":"2021-03-18T09:49:47","Local":"2021-03-18T10:49:47","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"06:56","Sunset":"18:59"}}
10:49:47.453 MQT: stat/werkkamer-licht-kast_003DFE/STATUS10 = {"StatusSNS":{"Time":"2021-03-18T10:49:47"}}
10:49:47.459 MQT: stat/werkkamer-licht-kast_003DFE/STATUS11 = {"StatusSTS":{"Time":"2021-03-18T10:49:47","Uptime":"0T01:18:12","UptimeSec":4692,"Heap":23,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":2,"POWER":"ON","Dimmer":94,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"RoseMeshSmart-2G","BSSId":"E6:63:DA:AD:18:CC","Channel":1,"RSSI":94,"Signal":-53,"LinkCount":1,"Downtime":"0T00:00:02"}}}
10:49:52.170 TYA: Send "55aa00000000ff"
10:49:52.179 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
10:49:52.182 TYA: Heartbeat
10:49:52.364 HTP: Command
10:49:52.370 SRC: WebCommand from 192.168.142.14
10:49:52.373 CMD: Group 0, Index 1, Command "STATUS", Data "0"
10:49:52.380 MQT: stat/werkkamer-licht-kast_003DFE/STATUS = {"Status":{"Module":0,"DeviceName":"Werkkamer Licht Kast","FriendlyName":["Werkkamer Licht Kast"],"Topic":"werkkamer-licht-kast_003DFE","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
10:49:52.388 MQT: stat/werkkamer-licht-kast_003DFE/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T01:18:17","StartupUTC":"2021-03-18T08:31:35","Sleep":50,"CfgHolder":4617,"BootCount":13,"BCResetTime":"2021-03-04T10:09:07","SaveCount":393,"SaveAddress":"FB000"}}
[... cut ...]
10:51:08.502 MQT: stat/werkkamer-licht-kast_003DFE/STATUS11 = {"StatusSTS":{"Time":"2021-03-18T10:51:08","Uptime":"0T01:19:33","UptimeSec":4773,"Heap":21,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":2,"POWER":"ON","Dimmer":94,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"RoseMeshSmart-2G","BSSId":"E6:63:DA:AD:18:CC","Channel":1,"RSSI":90,"Signal":-55,"LinkCount":1,"Downtime":"0T00:00:02"}}}
10:51:09.165 TYA: Send "55aa00000000ff"
10:51:09.175 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
10:51:09.177 TYA: Heartbeat
```

mvanlijden, 2021-03-18:
Here is the log from the other dimmer, with almost the same steps.
When I tried to turn it off, first time it did turn off correctly. When turning back on with physical switch, and turning off with physical switch then it refused.

Sorry for the long logs :)

```
10:49:36.528 MQT: stat/werkkamer-licht-raam_CC979E/STATUS10 = {"StatusSNS":{"Time":"2021-03-18T10:49:36"}}
10:49:36.534 MQT: stat/werkkamer-licht-raam_CC979E/STATUS11 = {"StatusSTS":{"Time":"2021-03-18T10:49:36","Uptime":"0T01:18:04","UptimeSec":4684,"Heap":22,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":72,"MqttCount":4,"POWER":"ON","Dimmer":94,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"RoseMeshSmart-2G","BSSId":"E6:63:DA:AD:18:CC","Channel":1,"RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:02"}}}
10:49:37.260 HTP: Command
10:49:37.265 SRC: WebCommand from 192.168.142.14
10:49:37.268 CMD: Group 0, Index 1, Command "STATUS", Data "0"
10:49:37.276 MQT: stat/werkkamer-licht-raam_CC979E/STATUS = {"Status":{"Module":0,"DeviceName":"Werkkamer Licht Raam","FriendlyName":["Werkkamer Licht Raam"],"Topic":"werkkamer-licht-raam_CC979E","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
10:49:37.283 MQT: stat/werkkamer-licht-raam_CC979E/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T01:18:05","StartupUTC":"2021-03-18T08:31:32","Sleep":50,"CfgHolder":4617,"BootCount":11,"BCResetTime":"2021-03-04T10:01:33","SaveCount":151,"SaveAddress":"F9000"}}
10:49:37.289 MQT: stat/werkkamer-licht-raam_CC979E/STATUS2 = {"StatusFWR":{"Version":"9.3.1.1(tasmota)","BuildDateTime":"2021-03-05T08:56:06","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"417/699"}}
10:49:37.294 MQT: stat/werkkamer-licht-raam_CC979E/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["RoseMeshSmart-2G",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000280","00006000","00000000"]}}
10:49:37.303 MQT: stat/werkkamer-licht-raam_CC979E/STATUS4 = {"StatusMEM":{"ProgramSize":593,"Free":408,"Heap":23,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
10:49:37.310 MQT: stat/werkkamer-licht-raam_CC979E/STATUS5 = {"StatusNET":{"Hostname":"werkkamer-licht-raam_CC979E-604","IPAddress":"192.168.142.229","Gateway":"192.168.142.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.142.1","Mac":"A4:CF:12:CC:97:9E","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
10:49:37.316 MQT: stat/werkkamer-licht-raam_CC979E/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.142.10","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_CC979E","MqttUser":"tasmota","MqttCount":4,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
10:49:37.324 MQT: stat/werkkamer-licht-raam_CC979E/STATUS7 = {"StatusTIM":{"UTC":"2021-03-18T09:49:37","Local":"2021-03-18T10:49:37","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"06:56","Sunset":"18:59"}}
10:49:37.330 MQT: stat/werkkamer-licht-raam_CC979E/STATUS10 = {"StatusSNS":{"Time":"2021-03-18T10:49:37"}}
10:49:37.336 MQT: stat/werkkamer-licht-raam_CC979E/STATUS11 = {"StatusSTS":{"Time":"2021-03-18T10:49:37","Uptime":"0T01:18:05","UptimeSec":4685,"Heap":23,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":4,"POWER":"ON","Dimmer":94,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"RoseMeshSmart-2G","BSSId":"E6:63:DA:AD:18:CC","Channel":1,"RSSI":72,"Signal":-64,"LinkCount":1,"Downtime":"0T00:00:02"}}}
10:49:40.184 HTP: Console
10:49:41.170 TYA: Send "55aa00000000ff"
10:49:41.179 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
10:49:41.181 TYA: Heartbeat
10:49:43.171 HTP: Command
10:49:43.176 SRC: WebCommand from 192.168.142.14
10:49:43.179 CMD: Group 0, Index 1, Command "STATUS", Data "0"
10:49:43.186 MQT: stat/werkkamer-licht-raam_CC979E/STATUS = {"Status":{"Module":0,"DeviceName":"Werkkamer Licht Raam","FriendlyName":["Werkkamer Licht Raam"],"Topic":"werkkamer-licht-raam_CC979E","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
10:49:43.193 MQT: stat/werkkamer-licht-raam_CC979E/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T01:18:11","StartupUTC":"2021-03-18T08:31:32","Sleep":50,"CfgHolder":4617,"BootCount":11,"BCResetTime":"2021-03-04T10:01:33","SaveCount":151,"SaveAddress":"F9000"}}
[... cut ...]
10:50:49.788 MQT: stat/werkkamer-licht-raam_CC979E/STATUS10 = {"StatusSNS":{"Time":"2021-03-18T10:50:49"}}
10:50:49.796 MQT: stat/werkkamer-licht-raam_CC979E/STATUS11 = {"StatusSTS":{"Time":"2021-03-18T10:50:49","Uptime":"0T01:19:17","UptimeSec":4757,"Heap":22,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":117,"MqttCount":6,"POWER":"ON","Dimmer":92,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"RoseMeshSmart-2G","BSSId":"E6:63:DA:AD:18:CC","Channel":1,"RSSI":82,"Signal":-59,"LinkCount":1,"Downtime":"0T00:00:02"}}}
```

beren12, 2021-03-28:
It may be related to this: https://github.com/arendst/Tasmota/issues/7926

Someone rewrote the TuyaMCU code to be more robust, seems tasmota wasn't fully following the specs/recommendations on talking to the chip, hopefully he'll have time to update to 9.3.1 and push a PR soon. Or anyone else ;-)

beren12, 2021-03-28:
If you don't mind downgrading you can give his build a try and see if the issue still happens

mvanlijden, 2021-03-29:
> If you don't mind downgrading you can give his build a try and see if the issue still happens

Seems like the update to the communication to the TuyaMCU he did mainly focuses on synchronization of statuses and indeed making communication more robust. This seems very related to my issues. Problem for me is getting real clear logs. Most times I need one more action to reproduce compared to last time. Also timing seems like an issue. So difficult to give devs her what they need. But a solution like he had created with the message queue, seems like a robust idea.

I tried downgrading from 9.3.1.1 to his latest 8.5.0 based release, but haven't had any success... Not to happy to do a serial flash instead of an OTA downgrade. So haven't been able to verify anything on it.

beren12, 2021-03-31:
@jaymemaurice you seem to understand the code better than I, have you had a chance to see TheEebb's work on the TuyaMCU code? I linked it in #7926. I've been having issues with my Feit dimmer as well with stock tasmota.

mvanlijden, 2021-04-25:
Issue still has not been resolved

jaymemaurice, 2021-05-21:
Sorry, I've been MIA for a while. The current TuyaMCU driver seems pretty hacky. It works well enough but I think there is room for a re-write...
What if instead of changing the current TuyaMCU driver, we help TheEebb merge his re-write as a new driver. Once we have feature parity we prune out the lesser driver from the main build target?

rb0135, 2021-05-24:
Hi,

I would like to add I have the same issue with a Tuya Dimmer, WF-DS01. Worked perfectly on 8.5.1 but I upgraded to 9.4.0 as I wanted to use the HA Integration rather than MQTT.

What I discovered is:
1) Using Software only works fine (but the dimmer value is not transmitted back to the software). IE. Lights turn on and off using the software (HA in this case).
2) If I use the hardware button to turn on, when I try to turn off by hardware button, the lights turn back on instantly.
3) If I use the hardware button to turn on, when I try to turn off by software, the lights turn off
4) If I use the hardware button to turn on, press the dim down button, then try to turn off by hardware button, the lights turn back on instantly
5) If I use the hardware button to turn on, press the dim down button, then try to turn off by software, the lights turn off.
6) If I use the hardware button to turn on, press the dim up button, then try to turn off by hardware button, the lights turn OFF

Here is my weblog4 output when turning on the lights then trying to turn them off with the hardware button:

> 08:28:43.316 MQT: stat/SW_LoungeLights/RESULT = {"WebLog":4}
08:28:43.597 CFG: Saved to flash at F6, Count 694, Bytes 4096
08:28:46.485 TYA: Send "55aa00000000ff"
08:28:46.494 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
08:28:46.497 TYA: Heartbeat
08:28:57.506 TYA: Send "55aa00000000ff"
08:28:57.515 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
08:28:57.518 TYA: Heartbeat
08:29:02.765 WIF: Checking connection...
08:29:08.515 TYA: Send "55aa00000000ff"
08:29:08.525 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
08:29:08.528 TYA: Heartbeat
08:29:15.495 TYA: Send "55aa001c000801150519081d0f028d"
08:29:19.483 TYA: Send "55aa00000000ff"
08:29:19.493 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
08:29:19.496 TYA: Heartbeat
08:29:22.756 WIF: Checking connection...
08:29:30.483 TYA: Send "55aa00000000ff"
08:29:30.492 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
08:29:30.495 TYA: Heartbeat
08:29:42.757 CMD: weblog 2
08:29:42.769 MQT: stat/SW_LoungeLights/RESULT = {"WebLog":2}
08:30:25.481 MQT: tele/SW_LoungeLights/STATE = {"Time":"2021-05-25T08:30:25","Uptime":"0T16:50:10","UptimeSec":60610,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Dimmer":7,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"SAPTEN","BSSId":"78:8A:20:81:2F:EC","Channel":6,"RSSI":100,"Signal":-37,"LinkCount":1,"Downtime":"0T00:00:04"}}
08:34:55.692 CMD: weblog 4
08:34:55.699 MQT: stat/SW_LoungeLights/RESULT = {"WebLog":4}
08:34:56.089 CFG: Saved to flash at F4, Count 696, Bytes 4096
08:35:00.474 TYA: Send "55aa00000000ff"
08:35:00.483 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
08:35:00.485 TYA: Heartbeat
08:35:03.272 WIF: Checking connection...
08:35:03.618 {"TuyaReceived":{"Data":"55AA0007000501010001010F","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
08:35:03.621 TYA: fnId=11 is set for dpId=1
08:35:03.623 TYA: RX Relay-1 --> MCU State: On Current State:Off
08:35:03.626 SRC: Switch
08:35:03.644 TYA: Send "55aa00060008020200040000000b20"
08:35:03.652 MQT: tele/SW_LoungeLights/STATE = {"Time":"2021-05-25T08:35:03","Uptime":"0T16:54:48","UptimeSec":60888,"Heap":26,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Dimmer":7,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"SAPTEN","BSSId":"78:8A:20:81:2F:EC","Channel":6,"RSSI":100,"Signal":-35,"LinkCount":1,"Downtime":"0T00:00:04"}}
08:35:03.659 MQT: stat/SW_LoungeLights/RESULT = {"POWER":"ON"}
08:35:03.664 MQT: stat/SW_LoungeLights/POWER = ON (retained)
08:35:03.668 {"TuyaReceived":{"Data":"55AA00070008020200040000000016","Cmnd":7,"CmndData":"0202000400000000","DpType2Id2":0,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000000"}}}
08:35:03.671 TYA: fnId=21 is set for dpId=2
08:35:03.674 TYA: RX value 0 from dpId 2 
08:35:03.677 {"TuyaReceived":{"Data":"55AA00070008020200040000000B21","Cmnd":7,"CmndData":"020200040000000B","DpType2Id2":11,"2":{"DpId":2,"DpIdType":2,"DpIdData":"0000000B"}}}
08:35:03.681 TYA: fnId=21 is set for dpId=2
08:35:03.683 TYA: RX value 11 from dpId 2 
08:35:03.686 {"TuyaReceived":{"Data":"55AA0007000501010001010F","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
08:35:03.690 TYA: fnId=11 is set for dpId=1
08:35:03.693 TYA: RX Relay-1 --> MCU State: On Current State:On
08:35:03.696 {"TuyaReceived":{"Data":"55AA0007000802020004000000081E","Cmnd":7,"CmndData":"0202000400000008","DpType2Id2":8,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000008"}}}
08:35:03.700 TYA: fnId=21 is set for dpId=2
08:35:03.702 TYA: RX value 8 from dpId 2 
08:35:04.070 CFG: Saved to flash at FB, Count 697, Bytes 4096
08:35:11.191 {"TuyaReceived":{"Data":"55AA0007000501010001000E","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
08:35:11.194 TYA: fnId=11 is set for dpId=1
08:35:11.197 TYA: RX Relay-1 --> MCU State: Off Current State:On
08:35:11.200 SRC: Switch
08:35:11.218 TYA: Send "55aa00060008020200040000000b20"
[... 82 lines cut ...]
08:38:41.881 TYA: RX value 5 from dpId 2 
08:38:41.940 {"TuyaReceived":{"Data":"55AA0007000501010001000E","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
[... cut ...]
08:38:42.082 {"TuyaReceived":{"Data":"55AA00070008020200040000000016","Cmnd":7,"CmndData":"0202000400000000","DpType2Id2":0,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000000"}}}
08:38:42.086 TYA: fnId=21 is set for dpId=2
08:38:42.089 TYA: RX value 0 from dpId 2 
08:38:43.262 WIF: Checking connection...
08:38:51.472 TYA: Send "55aa00000000ff"
08:38:51.482 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
08:38:51.485 TYA: Heartbeat
08:39:02.503 TYA: Send "55aa00000000ff"
08:39:02.513 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
08:39:02.516 TYA: Heartbeat

Quiet happy to help test ideas.

Rob

wouterbouvy, 2021-05-26:
I can confirm this same issue with my RJWF-02A Dimmer Module since Tasmota 9+. I rolled back to Tasmota 8.5.1 which works without issues

rb0135, 2021-06-20:
Issue still seems not to have been resolved in the latest 9.5.

Will this issue ever be resolved/looked at? I mean, it worked perfectly in 8.5, but from 9.0, there have been issues.

Thanks, Rob

L0rdCha0s, 2021-07-15:
Bump - still occurring on latest build

ascillato2, 2021-07-20:
Added the label Help Needed to see if someone in the community with Tuya knowledge and hardware can assist on this.

L0rdCha0s, 2021-07-21:
I'm actually having a hack on this now - pain to reproduce (seems to be time based).

rb0135, 2021-07-21:
> I'm actually having a hack on this now - pain to reproduce (seems to be time based).

I can reproduce constantly and consistently (see my post above with all scenarios). Glad to try things to help out. 

Would be interesting if someone would actually say why it was broken from version 8.5 to 9 rather than being silent on the matter. Apparently @TheEebb worked on a fix and has a version that works, but it has never been implemented in the mainstream Tasmota version, but his fork is based on 8.5 which is working fine.

rb0135, 2021-08-16:
Bump - still occurring on latest build

jeroenst, 2021-08-17:
I think I have the same issue: https://github.com/arendst/Tasmota/discussions/12904

beren12, 2021-09-08:
No but his changes from before the big code update in tasmota last year are visible to anyone. I don't understand it enough to adapt it, maybe the person who submitted the additions to the tuya code could look at it?

Jason2866, 2021-09-08:
@beren12 It is a project done from volunteers. Life goes on and the people who did the main part of the Tuya code are not contributing anymore to Tasmota.

wouterbouvy, 2021-09-08:
I had this same issue with my Tuya dimmer, for which I moved to ESPHome. Which works fine with my Home Assistant setup

cyber-junkie9, 2021-09-10:
> moved to ESPHome. 
mind sharing yaml ?
i have PS-16-DZ dimmer

wouterbouvy, 2021-09-21:
> > moved to ESPHome.
> 
> mind sharing yaml ?
> i have PS-16-DZ dimmer

Sorry for the late response. Here you go:
`esphome:
  name: tuya-transistorlamp
  platform: ESP8266
  board: esp01_1m

# Enable logging
logger:
  baud_rate: 0

# Enable Home Assistant API
api:
  password: !secret api_ota

ota:
  password: !secret api_ota

wifi:
  ssid: !secret ssid
  password: !secret ssid_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
#  ap:
#    ssid: "Tuya-Transistorlamp"
#    password: "jEv6yibQvQGg"

#captive_portal:
  
uart:
  rx_pin: GPIO3
  tx_pin: GPIO1
  baud_rate: 9600

tuya:

light:
  - platform: "tuya"
    name: "Transistorlamp"
    id: dimmer
    dimmer_datapoint: 2
#    min_value_datapoint: 2
    switch_datapoint: 1
    min_value: 63

binary_sensor:
  - platform: gpio
    id: button
    internal: true
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    on_press:
      - light.toggle: dimmer

status_led:
  pin:
    number: GPIO14`

rb0135, 2021-10-15:
> @beren12 It is a project done from volunteers. Life goes on and the people who did the main part of the Tuya code are not contributing anymore to Tasmota. The badge `help needed` has it reasons.

@Jason2866 - And when help is given, it is ignored. I have voluntered a lot in the last 20 years, and I still maintain the code. Sure, it is run by volunteers, but these volunteers are now ruining a good piece of software because "life goes on". Pretty poor attitude to have. If this never worked before, then yes, that is different. This did work before and has been completely broken. No wonder people are leaving Tasmota for Esphome with that attitude.

gwietze, 2021-10-17:
Hello, I have the same problems with my WF-DC01 dimmer.
Maybe it isn't relevante but when I do tuyamcu 21 so that the tuyamcu 21,2 is gone and the interface of tasmota doesn't show the dimmer anymore then the buttons on de device is working fine. 
Only is the dimmer function  posible on the device.

TheEebb, 2021-10-21:
The last time I checked for these devices was a few years ago and there weren't any. Bad timing I guess. I just ordered a battery PIR sensor from amazon and I'll test against that. I'll be updating my fork first with the changes if anyone wants to test it out with their dimmer and battery devices to confirm functionality.

nickg90, 2021-11-05:
> ating my fork first with the changes if anyone wants to test it out with their dimmer and battery devices to confir

@TheEebb happy to test it out, let me know when available. not sure how to get it onto my device, is there a link I can use which would l could paste in the firmware update textbox in the device UI?

TheEebb, 2021-11-06:
You load the firmware the same way you would a new update of tasmota. [Links](https://github.com/TheEebb/tasmota-tuyamcu-fix) to releases are on my testing fork.

I'm still testing the code with the battery device I just received from china. All the devices I was able to purchase in US are unfortunately TR3 modules. Its a low power arm chip so it might be worth creating a port for this. But that's a separate discussion.

nickg90, 2021-11-06:
[my device](https://www.tlc-direct.co.uk/Products/QURW03.html) is not a battery device but @ascillato2 believes my device has the root cause is the same issue. It reports the state of the device correctly when the swich is physically toggled but is not able to send toggle commands from the Tasmota UI.

I usually update my devices using the tasmota ota link a complied [tasmota.bin](http://ota.tasmota.com/tasmota/release/tasmota.bin) file. im not sure how to go about compling the files into a .bin myself but if there is a .bin available I would be happy to test when you are ready.

nickg90, 2021-11-11:
not sure if anything has changed by my switch has started responding again though I havnt changed the firmware in a while and it wasnt working last time I tried. now on 9.1.0

Tortue95, 2021-11-24:
Hello everyone, I have the same type of problem, at a certain moment when you ask to turn off the light, the light comes back on instantly

in order to highlight the problem I use the following configuration: 

```
TuyaMCU 11,1 # on/off
TuyaMCU 21,2 # dimmer
TuyaSend2 3,10 # set min value dimmer with hardwarebtn (10)
DimmerRange 297,1000 # set Dimmer range (I voluntarily use a high min value to better show the problem)
```

in order to reproduce the problem I will realize this scenario :
- set dimmer 100%
- set dimmer 1% # you will already see a first bug appears here not blocking 
- set dimmer 1% # we do it again to cause the blocking bug 
- push hardware btn # it turns off and on instantly 
- push software btn # it turns off and on instantly too 

let's look at this in detail :

# set dimmer 100%

log mqtt:
`cmnd/tasmota_ECA65B/Dimmer 100`

log tasmota:
```
22:12:20.126 SRC: MQTT
22:12:20.129 CMD: Grp 0, Cmnd 'DIMMER', Idx 1, Len 3, Data '100'
22:12:20.136 MQT: stat/tasmota_ECA65B/RESULT = {"POWER":"ON","Dimmer":100}
22:12:20.155 TYA: Send "55aa0006000802020004000003e800"
22:12:20.173 {"TuyaReceived":{"Data":"55AA0307000802020004000003E804","Cmnd":7,"CmndData":"02020004000003E8","DpType2Id2":1000,"2":{"DpId":2,"DpIdType":2,"DpIdData":"000003E8"}}}
22:12:20.175 TYA: fnId=21 is set for dpId=2
22:12:20.177 TYA: RX value 1000 from dpId 2 
22:12:20.354 CFG: Saved to flash at F7, Count 637, Bytes 4096
22:12:23.392 {"TuyaReceived":{"Data":"55AA03070005010100010112","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
22:12:23.394 TYA: fnId=11 is set for dpId=1
22:12:23.395 TYA: RX Relay-1 --> MCU State: On Current State:On
22:12:23.407 {"TuyaReceived":{"Data":"55AA0307000802020004000003E804","Cmnd":7,"CmndData":"02020004000003E8","DpType2Id2":1000,"2":{"DpId":2,"DpIdType":2,"DpIdData":"000003E8"}}}
22:12:23.409 TYA: fnId=21 is set for dpId=2
22:12:23.411 TYA: RX value 1000 from dpId 2 
22:12:23.423 {"TuyaReceived":{"Data":"55AA03070008690200040000000080","Cmnd":7,"CmndData":"6902000400000000","DpType2Id105":0,"105":{"DpId":105,"DpIdType":2,"DpIdData":"00000000"}}}
22:12:23.426 TYA: fnId=0 is set for dpId=105
22:12:23.428 TYA: RX value 0 from dpId 105
```

here no worries everything is fine 

# set dimmer 1%

log mqtt:
```
cmnd/tasmota_ECA65B/Dimmer 1
stat/tasmota_ECA65B/RESULT {"POWER":"ON","Dimmer":1}
stat/tasmota_ECA65B/RESULT {"POWER":"ON","Dimmer":30}
```

log tasmota:
```
22:12:35.074 SRC: MQTT
22:12:35.077 CMD: Grp 0, Cmnd 'DIMMER', Idx 1, Len 1, Data '1'
22:12:35.083 MQT: stat/tasmota_ECA65B/RESULT = {"POWER":"ON","Dimmer":1}
22:12:35.102 TYA: Send "55aa0006000802020004000001293f"
22:12:35.120 {"TuyaReceived":{"Data":"55AA03070008020200040000012943","Cmnd":7,"CmndData":"0202000400000129","DpType2Id2":297,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000129"}}}
22:12:35.123 TYA: fnId=21 is set for dpId=2
22:12:35.125 TYA: RX value 297 from dpId 2 
22:12:35.358 CFG: Saved to flash at F6, Count 638, Bytes 4096
22:12:38.343 {"TuyaReceived":{"Data":"55AA03070005010100010112","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
22:12:38.346 TYA: fnId=11 is set for dpId=1
22:12:38.348 TYA: RX Relay-1 --> MCU State: On Current State:On
[... 174 lines cut ...]
22:13:19.429 {"TuyaReceived":{"Data":"55AA03070008020200040000012943","Cmnd":7,"CmndData":"0202000400000129","DpType2Id2":297,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000129"}}}
22:13:19.431 TYA: fnId=21 is set for dpId=2
22:13:19.433 TYA: RX value 297 from dpId 2 
22:13:19.445 {"TuyaReceived":{"Data":"55AA03070008690200040000000080","Cmnd":7,"CmndData":"6902000400000000","DpType2Id105":0,"105":{"DpId":105,"DpIdType":2,"DpIdData":"00000000"}}}
22:13:19.447 TYA: fnId=0 is set for dpId=105
22:13:19.449 TYA: RX value 0 from dpId 105 
```

with an OFF message in mqtt we have the same problem but worse because we receive 4 mqtt "off / on / off / on" messages while all the messages sent in mqtt are not present in the tasmota logs ....
(there is no lack of log tasmota after) 

# Bypass Issue temporarily
for those who would like to bypass the problem temporarily while solving this bug, you can perform this configuration: 
```
TuyaMCU 11,1 
TuyaMCU 21,2 
TuyaSend2 3,150 # set dimmer min value for hardware btn to 15%, like rule1 above (optional)
DimmerRange 0,1000 
rule1 ON Dimmer#State<15 DO Dimmer 15 ENDON 
rule1 1 
```

This working for me but I can't use the first 15% of the dimmer

I hope that my research will help you

Tortue95, 2021-11-26:
Hi
I found the solution :), 
you miss the `dimmer_hw_min` variable on `changeUIntScale()`

on fonction `LightSerialDuty() :`
you need to modify all:
`duty = changeUIntScale(duty, 0, 100, 0, Settings->dimmer_hw_max);`
to:
`duty = changeUIntScale(duty, 0, 100, Settings->dimmer_hw_min, Settings->dimmer_hw_max);`

and on fonction: `TuyaProcessStatePacket()`
you need to modify:
`Tuya.Levels[dimIndex] = changeUIntScale(packetValue, 0, Settings->dimmer_hw_max, 0, 100);`
to:
`Tuya.Levels[dimIndex] = changeUIntScale(packetValue, Settings->dimmer_hw_min, Settings->dimmer_hw_max, 0, 100);`

I'm not a developer, it's not easy to get into an existing project ^^ but it's fun :)
with that all my problems seem to be fixed.

if I have time this weekend I would do a PR, something I have never done yet ^^

Tortue95, 2021-11-28:
HI, with my previous patc, I have one new error:

when I apply the dimmer to 0 (which sets relay1 to OFF automatically) when I ask to turn it on via the physical btn
nothing is to send in MQTT the fact that the relay is activated.

in order to correct this I have replaced the test in the `TuyaProcessStatePacket()` function
`if ((TasmotaGlobal.power || Settings->light_dimmer > 0) && (Tuya.buffer[dpidStart + 4] != bitRead(TasmotaGlobal.power, fnId - TUYA_MCU_FUNC_REL1))) {  `
through
`if (Tuya.buffer[dpidStart + 4] != bitRead(TasmotaGlobal.power, fnId - TUYA_MCU_FUNC_REL1)) { `

I do not understand the reason for the presence of the test `(TasmotaGlobal.power || Settings-> light_dimmer> 0)` which moreover is not present in the other tests for the `RELx_INV` or the `SWTx`. 

With all these changes the expected behavior is working fine for me.

closed 2021-11-28 by ascillato2

ascillato, 2021-11-28:
PR merged. Please, test latest version from development branch. Thanks.

mvanlijden, 2022-01-17:
Tried the development build but did not fix the issue. Light switching is still glitchy.