arendst/Tasmota #11352: Blitzwolf SHP6 -> Tasmota flashed -> over 10m not connecting to Wifi

joselit00, 2021-03-15:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
WiFi range limited to 3-4 Meters around Router. After that no connection can be established by the Blitzwolf SHP6 flashed with Tasmota

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ done] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [done ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [done ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ done] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ done] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): BlitzWolf BW-SHP6 15A Power Monitoring Plug
- [ ] Tasmota binary firmware version number used: 9.3.1(tasmota)
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: first tuya convert to 9.2.0 and then OTA to 9.3.1
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
13:10:49.737 CMD: Backlog Template; Module; GPIO 255
13:10:49.815 RSL: stat/Solarplug/RESULT = {"NAME":"BlitzWolf SHP6","GPIO":[320,1,576,1,2656,2720,0,0,2624,32,0,224,0,0],"FLAG":0,"BASE":45}
13:10:50.026 RSL: stat/Solarplug/RESULT = {"Module":{"0":"BlitzWolf SHP6"}}
13:10:50.282 RSL: stat/Solarplug/RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"0":"None"},"GPIO2":{"576":"LedLink_i"},"GPIO3":{"0":"None"},"GPIO4":{"2656":"HLWBL CF1"},"GPIO5":{"2720":"BL0937 CF"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"2624":"HLWBL SEL_i"},"GPIO13":{"32":"Button1"},"GPIO14":{"0":"None"},"GPIO15":{"224":"Relay1"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
no rules

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
13:12:36.664 CMD: Status 0
13:12:36.675 RSL: stat/Solarplug/STATUS = {"Status":{"Module":0,"DeviceName":"Solarplug","FriendlyName":["Solarplug"],"Topic":"Solarplug","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
13:12:36.680 RSL: stat/Solarplug/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"External System","Uptime":"0T02:41:15","StartupUTC":"2021-03-15T09:31:21","Sleep":50,"CfgHolder":4617,"BootCount":49,"BCResetTime":"2021-03-13T19:32:41","SaveCount":134,"SaveAddress":"F5000"}}
13:12:36.686 RSL: stat/Solarplug/STATUS2 = {"StatusFWR":{"Version":"9.3.1(tasmota)","BuildDateTime":"2021-03-09T16:12:28","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"396/699"}}
13:12:36.690 RSL: stat/Solarplug/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":3,"LogHost":"","LogPort":514,"SSId":["Marbella",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600001E5A0A000000000000","00000080","00006000","00000000"]}}
13:12:36.698 RSL: stat/Solarplug/STATUS4 = {"StatusMEM":{"ProgramSize":593,"Free":408,"Heap":27,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
13:12:36.703 RSL: stat/Solarplug/STATUS5 = {"StatusNET":{"Hostname":"Solarplug-3220","IPAddress":"192.168.1.33","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.250","Mac":"70:03:9F:61:0C:94","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
13:12:36.708 RSL: stat/Solarplug/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_610C94","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
13:12:36.715 RSL: stat/Solarplug/STATUS7 = {"StatusTIM":{"UTC":"2021-03-15T12:12:36","Local":"2021-03-15T13:12:36","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"07:02","Sunset":"18:54"}}
13:12:36.719 RSL: stat/Solarplug/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
13:12:36.727 RSL: stat/Solarplug/STATUS10 = {"StatusSNS":{"Time":"2021-03-15T13:12:36","ENERGY":{"TotalStartTime":"2021-03-13T19:32:41","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
13:12:36.733 RSL: stat/Solarplug/STATUS11 = {"StatusSTS":{"Time":"2021-03-15T13:12:36","Uptime":"0T02:41:15","UptimeSec":9675,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Marbella","BSSId":"C8:0E:14:E6:D3:F8","Channel":6,"RSSI":30,"Signal":-85,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Plug device 10 meters away from Router

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Plug should connect to wifi without issues

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
Original device software connects without issue at the place in my house. With tasmota it won´t connect.
When plugin it in next to the router no issue. 

6 Meters away:
Tasmota Information:
AP1 SSId (RSSI) | Marbella (32%, -84 dBm)

Router (Fritzbox 6490 Cable) for the device
Signalstärke-56dBm

Recalibrating with "Reset 3" and restart did not help
SetOption41 was not tested as the initial wifi connection cannot be established.

Working fine with the original firmware. This is why i exclude a hardware issue

-- | --

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2021-03-15:
Tried tweaking [WifiPower](https://tasmota.github.io/docs/WifiPower/)?

joselit00, 2021-03-15:
> Tried tweaking [WifiPower](https://tasmota.github.io/docs/WifiPower/)?
It is currently set to default and also max. Reducing this value wouldn´t make any sense to me as it hardly works now.

16:35:29.416 CMD: wifipower
16:35:29.426 RSL: stat/Solarplug/RESULT = {"WifiPower":"17.0"}

ascillato2, 2021-04-01:
Hi,

This can be 2 things. A device flashed without erasing the flash first (or without performing reset 6 after flashing - that leaves an incorrect internal wifi calibration of the radio), or the wifi antenna damaged.

Please, try command reset 6 on the console and try again. If that don't improve your connection, I'm afraid that the antenna is broken.

If you want to rule out the software, try ESPHome, ESPEasy or ESPurna.

Please, let us know your tests. Thanks.

closed 2021-04-01 by ascillato2

joselit00, 2021-04-06:
I´ve tested now with 6 different devices all flashed with Tasmota. Where the original firmware reaches around -70dbm Tasmota only gets -78 or -80 dbm. 
I fully understand that the design is not the best of the hadware but it really seems to have something to do with tasmota.

jddonovan, 2021-10-06:
I haven't had any issues with range. I've used Tuya convert. Solid link between AP and SHP10 despite 10M+ and three dry walls in between. Solid link with SHP-6 (original 10A version) with 3-4 meter distance with the plug being in a suboptimal place -- signal to noise: -57/-91 dBm. Devices were upgraded around June-August.
 

Manual flash instructions spesifically state that one needs to erase flash to avoid issues with dropping WiFi:

> Step 7~
> Erase flash to avoid issues with dropping WiFi connection.
> Example using esptool:
> esptool.py --port COM3 erase_flash

I'm not sure if you can erase the wifi part without reflashing the entire chip, ie. with wires, not using OTA.

It's entirely possible that Tuya Convert script has somehow skipped the proper flash routine, but even then this is quite weird. This is just some info I can provide -- not sure if I can be of help.

reddwarf55555, 2023-03-04:
I had the same problem. I changed the network type on Tasmota from 802.11n to 802.11g and it works as before.
Try the command "Wifi 3" for 11g