arendst/Tasmota #11357: MCP23017 issue

ahmaddxb, 2021-03-16:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): __d1mini___
- [ ] Tasmota binary firmware version number used: __9.3.1___
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _Tasmotizer____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
12:04:29.022 SRC: Backlog
12:04:29.025 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
12:04:29.033 MQT: stat/sonoff/irrigation_16ch/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,640,608,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
12:04:29.247 SRC: Backlog
12:04:29.250 CMD: Group 0, Index 1, Command "MODULE", Data ""
12:04:29.256 MQT: stat/sonoff/irrigation_16ch/RESULT = {"Module":{"18":"Generic"}}
12:04:29.498 SRC: Backlog
12:04:29.502 CMD: Group 0, Index 1, Command "GPIO", Data "255`: ```lua"

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
12:05:55.874 MQT: stat/sonoff/irrigation_16ch/STATUS = {"Status":{"Module":18,"DeviceName":"Irrigation 16CH","FriendlyName":["Tasmota","Tasmota2","Tasmota3","Tasmota4","Tasmota5","Tasmota6","Tasmota7","Tasmota8"],"Topic":"sonoff/irrigation_16ch","ButtonTopic":"0","Power":32768,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
12:05:55.881 MQT: stat/sonoff/irrigation_16ch/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"sonoffs","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:03:23","StartupUTC":"2021-03-16T08:02:32","Sleep":50,"CfgHolder":4617,"BootCount":27,"BCResetTime":"2021-01-24T17:28:33","SaveCount":292,"SaveAddress":"F8000"}}
12:05:55.886 MQT: stat/sonoff/irrigation_16ch/STATUS2 = {"StatusFWR":{"Version":"9.3.1(sensors)","BuildDateTime":"2021-03-16T06:33:33","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"404/699"}}
12:05:55.892 MQT: stat/sonoff/irrigation_16ch/STATUS3 = {"StatusLOG":{"SerialLog":1,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["ASUS",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000AA009","2805C8000100060000005A0A000000000000","000082C8","00006000","00000000"]}}
12:05:55.905 MQT: stat/sonoff/irrigation_16ch/STATUS4 = {"StatusMEM":{"ProgramSize":651,"Free":352,"Heap":18,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDA8787","0415A005","B7FFBFCF","01DA9BF5","64367CC7","00084052","20000000","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,26,28,29,31,34,37,39,40,42,43,51,52,55,56,58,59,62,64,66,67,74"}}
12:05:55.913 MQT: stat/sonoff/irrigation_16ch/STATUS5 = {"StatusNET":{"Hostname":"sonoff/irrigation_16ch-4015","IPAddress":"192.168.1.102","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.104","Mac":"2C:F4:32:14:6F:AF","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
12:05:55.919 MQT: stat/sonoff/irrigation_16ch/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.235","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_146FAF","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
12:05:55.930 MQT: stat/sonoff/irrigation_16ch/STATUS7 = {"StatusTIM":{"UTC":"2021-03-16T08:05:55","Local":"2021-03-16T12:05:55","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+04:00","Sunrise":"10:00","Sunset":"21:56"}}
12:05:55.943 MQT: stat/sonoff/irrigation_16ch/STATUS10 = {"StatusSNS":{"Time":"2021-03-16T12:05:55","BME280":{"Temperature":33.8,"Humidity":37.7,"DewPoint":17.3,"Pressure":1012.6},"MCP230XX":{"D0":1,"D1":1,"D2":1,"D3":1,"D4":1,"D5":1,"D6":1,"D7":1,"D8":1,"D9":1,"D10":1,"D11":1,"D12":1,"D13":1,"D14":1,"D15":0,"MCP230_OUT":{"OUT_D0":"OFF","OUT_D1":"OFF","OUT_D2":"OFF","OUT_D3":"OFF","OUT_D4":"OFF","OUT_D5":"OFF","OUT_D6":"OFF","OUT_D7":"OFF","OUT_D8":"OFF","OUT_D9":"OFF","OUT_D10":"OFF","OUT_D11":"OFF","OUT_D12":"OFF","OUT_D13":"OFF","OUT_D14":"OFF","OUT_D15":"ON","END":1}},"PressureUnit":"hPa","TempUnit":"C"}}
12:05:55.954 MQT: stat/sonoff/irrigation_16ch/STATUS11 = {"StatusSTS":{"Time":"2021-03-16T12:05:55","Uptime":"0T00:03:23","UptimeSec":203,"Heap":18,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","POWER6":"OFF","POWER7":"OFF","POWER8":"OFF","POWER9":"OFF","POWER10":"OFF","POWER11":"OFF","POWER12":"OFF","POWER13":"OFF","POWER14":"OFF","POWER15":"OFF","POWER16":"ON","Wifi":{"AP":1,"SSId":"ASUS","BSSId":"B4:FB:E4:71:42:B2","Channel":11,"RSSI":66,"Signal":-67,"LinkCount":1,"Downtime":"0T00:00:07"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2021-03-17:
Please try latest development version. Changes are made in MCP230xx driver

theOzzieRat, 2021-03-17:
I am not sure if the change I made today will affect this, but it may. Interested to see the feedback.

ahmaddxb, 2021-03-18:
Compiled and tested 9.3.1.1. Now is seems the relays choose to be either on or off randomly after a reboot. If i turn them all off the wont be the way i left them after i reboot the device.

theOzzieRat, 2021-03-18:
I have done a bit more investigation and there appears to be a problem when so0 is set to 1. When so0 is set to 0 after a reboot the output values are all set as per the values specified by the sensor29 x,[5|6],y commands, but when  so0 is 1 the output values aren't random but are not what is expected (ie. not the values that were present at reboot OR the values specified in the sensor29 commands).

theOzzieRat, 2021-03-19:
There is a definite bug when setoption0 is 1. I am currently working on a fix.

closed 2021-03-19 by arendst

theOzzieRat, 2021-03-19:
@ahmaddxb when you get a chance please test the latest development version of 9.3.1.2 and let me know if it fixes your issues or not.

ahmaddxb, 2021-03-19:
The state is remembers after reboot thanks, I will let you know if the initial issue occurs still after some long term testing.

ahmaddxb, 2021-03-23:
I dont think this is fixed initial issue is still there and all so sometime if i cut the power and plug it back in it will want to start up to the state below. Strange

[image]

theOzzieRat, 2021-03-23:
I am unable to replicate your problem. The only thing that I can find that now doesn't work is enabling INTERLOCK, the code for that looks broken to me and turning it on I get unexpected/weird results.
If you want me to investigate further I will need your full configuration, not just the sensor29 commands.

ahmaddxb, 2021-03-24:
09:42:46.478 MQT: stat/sonoff/irrigation_16ch/RESULT = {"Interlock":"OFF","Groups":"1,2,3,4,5,6,7,8"}

ahmaddxb, 2021-03-24:
how do I send my full config

ahmaddxb, 2021-03-24:
[Config_sonoff_irrigation_16ch_4015_9.3.1.2.zip](https://github.com/arendst/Tasmota/files/6194673/Config_sonoff_irrigation_16ch_4015_9.3.1.2.zip)

theOzzieRat, 2021-03-24:
I suspect that config file contains your wifi credentials. Are you on discord? If so try and message me privately on there and we can try and sort things out. My username is theRat#2205.

theOzzieRat, 2021-03-24:
I uploaded your config onto a d1-mini with an MCP23007 attached and cannot get it to misbehave (multiple restarts and random power cycles). Are you certain that something external (ie. home assistant) isn't changing the state? Without further diagnostics from your end I am going to have to assume it is something wrong in your environment and not the MCP driver.

ahmaddxb, 2021-03-25:
I will try do more testing with mqtt disabled and rule that out but I don't think that is the case. 

I think it may be a power supply issue running the mcp23007 and the 5v relays from the d1 mini board, the USB charger maybe can't supply enough amps. Going to try power the 16ch relay from a separate 5v supply and see.

If you have handy the compile firmware you used for testing I could try load that onto my esp as another thing to test.

ahmaddxb, 2021-03-30:
can you send be a picture of the MCP23007 board you are using

ahmaddxb, 2021-03-30:
also can you send me how you wired it to the esp

rapi3, 2021-04-01:
Do you have MCP RST pin to VCC ? if not then strange problems appear.

theOzzieRat, 2021-04-01:
Both the Tasmota doc's and the Microchip datasheets tell you that RESET must be pulled to VCC for the device to operate. The RESET pin and the address pins (A0,A1,A2) CAN NOT BE LEFT FLOATING!

rapi3, 2021-04-02:
There is a problem if Pulsetime it is activated for relays 1-8; it will interfere with relays 9-16
and on console for relays 9-16 pulsetime is unavailable
15:38:41.212 CMD: pulsetime9
15:38:41.217 MQT: Wemos/stat/W11/RESULT = {"Command":"Error"}

It easy to reproduce:
Activate PulseTime for one or more relays 1-8 and you can see his effect on correspondent relay 9-16.

Jason2866, 2021-04-02:
As designed.

rapi3, 2021-04-02:
Good to know now because it is not mentioned anywhere on MCP230xx page.
https://tasmota.github.io/docs/MCP230xx/

Jason2866, 2021-04-02:
It is a general solution for pulsetime, not only for MCP230XX

rapi3, 2021-04-02:
Then it will be great if somebody can update the info here:
https://tasmota.github.io/docs/Commands/#pulsetime

meingraham, 2021-04-02:
Tasmota Docs are open source as well. All users are able to contribute edits and new content. Click on the pencil icon at the top right of the article, make the edits, and submit the docs PR.

Thanks.

rapi3, 2021-04-02:
Thank you for possibility to contribute. Note added.

Jason2866, 2021-04-02:
The corresponding commit https://github.com/arendst/Tasmota/commit/a9630b9d053036960ac03d2de74c97fb4f7a400d

PR #11385 (2021-03-19): MCP230xx fix savestate for pinmode=6 and init with so0=1
## Description:
When so0=1 and MCP230xx pinmode=6 the state was being saved inverted, resulting in the wrong value being restored on restart.
The saved state was not being restored correctly for the 2nd port of the MCP230xx, this has also been fixed.

**Related issue (if applicable):** fixes #11357

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change  [... cut]