arendst/Tasmota #11358: tasmota-display build isn't enabling TM1637 driver without enabling SPI.

ajithvasudevan, 2021-03-16:
### PROBLEM DESCRIPTION
Tasmota's _Display_ build (_tasmota-display.bin_) is supposed to support **TM1637** in release **9.3.1**. However, this is not happening if we configure just the **TM1637** specific GPIO pins (TM1637 DIO and TM1637 CLK). 

After configuring the _TM1637 DIO_ and _TM1637 CLK_ pins, none of the ```DISPLAY-``` commands are available, indicating that ```xdrv_13_display``` is not executed.

I have traced the issue (in the latest development branch) to the following in ```xdrv_13_display.ino``` :

```
bool Xdrv13(uint8_t function)
{
  bool result = false;

  if ((TasmotaGlobal.i2c_enabled || TasmotaGlobal.spi_enabled || TasmotaGlobal.soft_spi_enabled) && XdspPresent()) {
```
The above condition is false even if **TM1637** pins are configured 

This could be because of ```TasmotaGlobal.soft_spi_enabled``` not being ```true``` from the following code in ```support_tasmota.ino```

```
    bool valid_cs = (ValidSpiPinUsed(GPIO_SPI_CS) ||
                     ValidSpiPinUsed(GPIO_RC522_CS) ||
                     (ValidSpiPinUsed(GPIO_NRF24_CS) && ValidSpiPinUsed(GPIO_NRF24_DC)) ||
                     ValidSpiPinUsed(GPIO_ILI9341_CS) ||
                     ValidSpiPinUsed(GPIO_ILI9341_DC) || // there are also boards without cs
                     ValidSpiPinUsed(GPIO_EPAPER29_CS) ||
                     ValidSpiPinUsed(GPIO_EPAPER42_CS) ||
                     ValidSpiPinUsed(GPIO_ILI9488_CS) ||
                     ValidSpiPinUsed(GPIO_SSD1351_CS) ||
                     ValidSpiPinUsed(GPIO_RA8876_CS) ||
                     ValidSpiPinUsed(GPIO_ST7789_DC) ||  // ST7789 CS may be omitted so chk DC too
                     ValidSpiPinUsed(GPIO_ST7789_CS) ||
                     (ValidSpiPinUsed(GPIO_SSD1331_CS) && ValidSpiPinUsed(GPIO_SSD1331_DC)) ||
                     ValidSpiPinUsed(GPIO_SDCARD_CS)
                    );
    // If SPI_CS and/or SPI_DC is used they must be valid
    TasmotaGlobal.spi_enabled = (valid_cs) ? SPI_MOSI_MISO : SPI_NONE;
```

As a fix, one could add  ```(ValidSpiPinUsed(GPIO_TM1637DIO) && ValidSpiPinUsed(GPIO_TM1637CLK))```  to ```valid_cs``` above, but then this will automatically assign 3 **SPI** pins to GPIOs unnecessarily. I had pointed out this issue in the **TM1638** PR and had provided a fix there.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Wemos D1 Mini
- [x] Tasmota binary firmware version number used: 9.3.1 and latest development (Display build)
  - [x] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: tasmotizer,  visual code (platformIO)
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
18:11:01.200 MQT: stat/TM1637/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
18:11:01.427 MQT: stat/TM1637/RESULT = {"Module":{"18":"Generic"}}
18:11:01.683 MQT: stat/TM1637/RESULT = {"GPIO0":{"6624":"TM1637 CLK"},"GPIO1":{"0":"None"},"GPIO2":{"6656":"TM1637 DIO"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:

- [ ] Provide the output of this command: `Status 0`:
```lua
18:12:36.706 MQT: stat/TM1637/STATUS = {"Status":{"Module":18,"DeviceName":"Tasmota","FriendlyName":["TM1637"],"Topic":"TM1637","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
18:12:36.735 MQT: stat/TM1637/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:02:33","StartupUTC":"2021-03-16T17:10:03","Sleep":50,"CfgHolder":4617,"BootCount":11,"BCResetTime":"2021-03-16T16:49:04","SaveCount":26,"SaveAddress":"FA000"}}
18:12:36.769 MQT: stat/TM1637/STATUS2 = {"StatusFWR":{"Version":"9.3.1(display)","BuildDateTime":"2021-03-09T16:13:09","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"353/699"}}
18:12:36.789 MQT: stat/TM1637/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Andromeda",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000080","00006000","00000000"]}}
18:12:36.818 MQT: stat/TM1637/STATUS4 = {"StatusMEM":{"ProgramSize":579,"Free":424,"Heap":27,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0FAA858E","04049FA1","000000C3","00000000","00004080","00000020","401F8000","00000000"],"Drivers":"1,2,4,5,8,9,10,13,16","Sensors":"1,2,5,6"}}
18:12:36.848 MQT: stat/TM1637/STATUS5 = {"StatusNET":{"Hostname":"TM1637-5904","IPAddress":"172.20.1.13","Gateway":"172.20.1.1","Subnetmask":"255.255.255.0","DNSServer":"172.20.1.1","Mac":"84:F3:EB:25:77:10","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
[... 26 lines cut ...]
18:13:20.085 MQT: Subscribe to cmnd/tasmotas/#
18:13:20.086 MQT: Subscribe to cmnd/DVES_257710_fb/#
18:13:20.089 MQT: tele/TM1637/INFO1 = {"Module":"Generic","Version":"9.3.1(display)","FallbackTopic":"cmnd/DVES_257710_fb/","GroupTopic":"cmnd/tasmotas/"}
18:13:20.093 MQT: tele/TM1637/INFO2 = {"WebServerMode":"Admin","Hostname":"TM1637-5904","IPAddress":"172.20.1.13"}
18:13:20.103 MQT: tele/TM1637/INFO3 = {"RestartReason":"External System"}
18:13:23.470 QPC: Reset
18:13:24.382 MQT: tele/TM1637/STATE = {"Time":"2021-03-16T18:13:24","Uptime":"0T00:00:09","UptimeSec":9,"Heap":30,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Andromeda","BSSId":"00:11:32:6C:33:E2","Channel":5,"RSSI":100,"Signal":-38,"LinkCount":1,"Downtime":"0T00:00:03"}}
18:13:25.382 APP: Boot Count 12
18:13:25.735 CFG: Saved to flash at F8, Count 28, Bytes 4096

```

### TO REPRODUCE
Download and flash the **9.3.1** **tasmota-display.bin** build or build and flash the latest **development** branch code with the **tasmota-display** env. 

### EXPECTED BEHAVIOUR
Tasmota 9.3.1 Display build firmware should support TM1637 Display as it is documented as supported, in the Release notes.

### SCREENSHOTS
NA

### ADDITIONAL CONTEXT
See discussion in https://github.com/arendst/Tasmota/pull/11031  for additional context

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2021-03-16:
As this display type doesn't need SPI or I2C it should not bother with these interfaces.

The goal of adding the dedicated GPIOs was (and is) to only allow this display type to work if a user has configured the specific TM1637 GPIO's. The TM1637 driver should only work when these GPIO's are configured without any need for SPI or I2C.

arendst, 2021-03-16:
Ah I see what you mean.

I will need to rewrite the check for display support WITHOUT either I2c or SPI present.

ajithvasudevan, 2021-03-16:
> Ah I see what you mean.
> 
> I will need to rewrite the check for display support WITHOUT either I2c or SPI present.

I had indeed made those changes earlier (as part of TM1638 PR) here: https://github.com/arendst/Tasmota/commit/75699fe203202a2832fbc81d73862828a8cbbc2b

arendst, 2021-03-17:
Pls try if latest fix solves selecting the TM1637 driver

ajithvasudevan, 2021-03-17:
> Pls try if latest fix solves selecting the TM1637 driver

@arendst 
This latest fix does allow **TM1637** driver selection in the _display_ build.

However, I see that the ```xdsp_interface.ino``` has the following:

```
#if defined(USE_I2C) || defined(USE_SPI)
#ifdef USE_DISPLAY
```

Doesn't this mean the TM1637 driver still "depends" on the existence of either I2C or SPI ?
It probably means I can't use TM1637 Display if I have something like the following in my ```user_config_override.h``` :
```
#undef  USE_I2C
#undef USE_SPI

#define USE_DISPLAY
#define USE_DISPLAY_TM1637
```
But being able to do this is not critical. It is good to have.

arendst, 2021-03-17:
Ah forgot to change . Will have a look now.

arendst, 2021-03-17:
Fixed again

ajithvasudevan, 2021-03-17:
Tested the fix. 
It works fine with both the display build as well as the user_config_override.h settings I mentioned above.

I think the [**TM1638 PR**](https://github.com/arendst/Tasmota/pull/11031) should work now. Will merge the fix there and test.  

Hope it is fine that the same **TM1637** driver is used for the **TM1638** as well. 

Also, I've originally planned to use the the same driver for the **MAX7912** as well. But now having second thoughts about it. Maybe I should make the **MAX7912** a separate driver?  Please suggest.

Thank you!

arendst, 2021-03-17:
Make it seperate first and if in the end there is too much redundancy you can always merge.

ajithvasudevan, 2021-03-17:
> Make it seperate first and if in the end there is too much redundancy you can always merge.

You mean the MAX driver separate, right?  And let TM1638 still be part of TM1637, right?

arendst, 2021-03-17:
Correct, Right.

ajithvasudevan, 2021-03-17:
I have merged this fix with the [**TM1638 PR**](https://github.com/arendst/Tasmota/pull/11031) and it works fine without any further changes to the PR code. 

Would you please accept the PR after review?

closed 2021-03-18 by ascillato2

sugus6, 2021-03-18:
I tried a 6-Digit TM1637 display today.
Program Version **9.3.1.2(display)**, Build Date & Time **2021-03-18T16:46:09**.
The command DisplayType or DisplayType 1 doesn’t work for me.
`21:18:13.686 CMD: DisplayType 1
21:18:13.692 RSL: stat/tasmota_7BE2B9/RESULT = {"Command":"Unknown"}
`

The following commands works: Display, DisplayText 1234, DisplayClock, ....
`21:18:34.554 CMD: display
21:18:34.559 RSL: stat/tasmota_7BE2B9/RESULT = {"Display":{"Model":15,"Width":0,"Height":0,"Mode":1,"Dimmer":1,"Size":4,"Font":1,"Rotate":0,"Refresh":2,"Cols":[16,8],"Rows":2}}
`

What am I doing wrong?

ajithvasudevan, 2021-03-19:
> I tried a 6-Digit TM1637 display today.
> . . .
> . . .
> What am I doing wrong?

If you built the code from the latest arendst/Tasmota development branch, then please use ```DisplaySize``` instead. See the documentation in the driver code here: https://github.com/arendst/Tasmota/blob/development/tasmota/xdsp_15_tm1637.ino

```DisplayType``` was introduced in a subsequent  [**PR**](https://github.com/arendst/Tasmota/pull/11031), but it has been decided to discontinue it. Instead, the command would be ```DisplayCols``` to change the number of digits in the future.

ajithvasudevan, 2021-03-19:
The future (that I mentioned above) has arrived sooner than expected :)  Please ignore the above if you are taking the _latest_ code from arendst/Tasmota development branch.
You should now use ```DisplayCols``` to set the number of digits. In your case, ```Backlog DisplayCols 6 ; Restart 1``` should do it.

sugus6, 2021-03-19:
Thank you for your quick feedback.
I tried the latest de development build from http://ota.tasmota.com/tasmota/tasmota-display.bin.gz
Tasmota Version 9.3.1.2(display)-2_7_4_9(2021-03-19T14:44:17)

When I execute this command `displaycols` I get…
19:41:44.189 RSL: stat/tasmota_7BE2B9/RESULT = {"DisplayCols1":4}

I change to 6-digit `DisplayCols 6`
19:47:10.286 CMD: DisplayCols 6
19:47:10.293 RSL: stat/tasmota_7BE2B9/RESULT = {"DisplayCols1":6}

My Wemos D1 mini reboot automaticaly but is still a 4 digits display!

00:00:00.049 CFG: Loaded from flash at F8, Count 28
00:00:00.054 QPC: Count 1
00:00:00.062 DSP: TM1637 with 4 digits
00:00:00.065 Project tasmota Tasmota Version 9.3.1.2(display)-2_7_4_9(2021-03-19T14:44:17)

19:48:05.516 CMD: displaycols
19:48:05.521 RSL: stat/tasmota_7BE2B9/RESULT = {"DisplayCols1":4}
Displaycols is not saved!

My display lights up at value higer then `DisplayDimmer 13` 
19:34:57.407 CMD: displaydimmer 13
19:34:57.413 RSL: stat/tasmota_7BE2B9/RESULT = {"DisplayDimmer":14}
The FW default value for displaydimmer is 7. Can this value be adjustet to 20 as default value?

ajithvasudevan, 2021-03-20:
@sugus6 ,

The code is in a bit of flux :)
Can you please try `DisplayWidth 6` ?

I will update the default `DisplayDimmer` value to something higher in the forthcoming [PR](https://github.com/arendst/Tasmota/pull/11387) update to the same driver.

Presently, `DisplayMode` is supported with values 0,1,2, and 3.

All the 7-segment specific Display- commands work only under `DisplayMode 0` 

Also, Power control is implemented. `Power 1` to turn on the display.

sugus6, 2021-03-21:
Hi ajithvasudevan

I did tests with the latest build: `9.3.1.2(display)-2_7_4_9(2021-03-21T16:59:32)`
I was able to set with command `DisplayWidth 6` the correct display size.
Also the Dimmer default value is set to a higher value. Thank you so match for you effort.

When I execute `DisplayText 123456` I get a wrong output on my display: **321654**.
I’ts look that the two blocks of 3digit are swapped!
I have testet my two 6-digit displays 0.36/0,56 inch, and I get with both the same misrepresentation.
I did a test with the ESPEasy firmware and there is the order of the digts correct.
Can you have a look at this issue?

These are my displays:

TM1637 6-digit displays 0.36/0,56 inch:
https://de.aliexpress.com/item/1005001928450315.html?spm=a2g0s.9042311.0.0.430b4c4dCoFbG8

TM1637 4-digit displays 0.36:
https://de.aliexpress.com/item/1005001993464723.html?spm=a2g0s.9042311.0.0.27424c4dvpel2p
Command DisplayText 1234 works as expected

MAX7219 8-digit displays:
https://de.aliexpress.com/item/1005001432551106.html?spm=a2g0o.productlist.0.0.50fa20bdIalrS0
Command DisplayText 12345678 works as expected

ajithvasudevan, 2021-03-22:
@sugus6 ,

> 
> I did tests with the latest build: `9.3.1.2(display)-2_7_4_9(2021-03-21T16:59:32)`

Thanks for testing this driver thoroughly!

> 
> When I execute `DisplayText 123456` I get a wrong output on my display: **321654**.
> I’ts look that the two blocks of 3digit are swapped!

It looks like there are different variants of the 6-digit TM1637 modules, with the digits differently wired. 
I'm working on [support for this additional variant](https://github.com/arendst/Tasmota/pull/11422) of the TM1637 6-digit module.

ajithvasudevan, 2021-03-22:
@sugus6 ,
You can now use the command `DisplayType 1` to switch to the type of 6-digit TM1637 display you have.   
Please test with the latest dev build.

sugus6, 2021-03-22:
**@ajithvasudevan**

Thank you for the quick fix.
With the latest version `9.3.1.2(display)-2_7_4_9(2021-03-22T15:35:22` and command `DisplayType 1` 
my display show now 123456 for `DisplayText 123456`.

Well done.

lalo-uy, 2021-06-05:
Is there any chance to have more than one TM1637? May be sharing the DIO pin or on 2 separated ones.

ajithvasudevan, 2021-06-05:
I am not entirely sure but I think the TM1637 hardware has a limitation on
sharing the DIO pin, as the protocol isn't exactly SPI or I2C.
What you want should be possible with two separate sets of pins, though it
will need some major changes to the TM1637 Tasmota driver and not to
mention changes elsewhere, like adding more GPIO Pins, etc.,
What exactly is your requirement? Is it to just have two sets of 4-digit
displays independently controllable by Tasmota? If this is the case, you
can simply use a single 8-digit MAX 7219 seven segment display module,
separating one set of 4 digits from the module by lengthening its
terminals, if needed. You can then concatenate the two 4-digit texts and
send it to the MAX.

See attachment, for an example of what I mean.

Ajith

ajithvasudevan, 2021-06-05:
[image]

lalo-uy, 2021-06-05:
Yes, I need to show 2 temps. Weird no one build the max module as 2 x 4
digit lines.
I also have a bounch of TM1637 modules arround.
Now I use some rules to alternate 2 values in the same display ever 5 sec.
Thanks anyway.

ajithvasudevan, 2021-06-06:
> ... Weird no one build the max module as 2 x 4 digit lines ...

It does exist.  https://robu.in/product/max7219-digital-tube-display-module-control-module/     This is the one I used to implement a similar requirement as yours (shown in the image in my previous post)

lalo-uy, 2021-06-06:
I want it in 2 lines, like this one
https://images.app.goo.gl/hcbXGceF4QfBPmug7
But with Max chip. The picture is with 74hc595.

ajithvasudevan, 2021-06-07:
> I want it in 2 lines...
That was the point of the example picture I attached previously.
In the example above, I have separated the two 4-digit displays on a single MAX module by de-soldering one of the 4-digit seven-segments from the board and re-connecting it by soldering short lengths of wire between each of the de-soldered display terminals and the board. Now the 4-digit display at the end of flexible wires can be re-oriented and mounted any which way you want, including one below the other, in 2 lines.
Of course this solution requires a bit of soldering skill. You can probably take help from someone who has those skills, if you are not upto it.