arendst/Tasmota #11365: Got an error when compiling Tasmota to support TLS

smilexth, 2021-03-17:
### PROBLEM DESCRIPTION

When i try to add "lib_extra_dirs = lib/lib_ssl" into standard tasmota environment to enable tts, i got an error about " IRremoteESP8266.h: No such file or directory" could you please give me some advise ?

### REQUESTED INFORMATION

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): _____
- [X] Tasmota binary firmware version number used: 9.3.1
  - [] Pre-compiled
  - [X] Self-compiled
- [X] Flashing tools used: platformio via visual studio code
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
**terminal compiler debug message**

[image]

**platformio_tasmota_env.ini**

[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2021-03-17:
Do NOT change `[env:Tasmota]` all libs are already enabled.
With your change only lib `lib_ssl` is active, all others are disabled!

smilexth, 2021-03-17:
> Do NOT change `[env:Tasmota]` all libs are already enabled.
> With your change only lib `lib_ssl` is active, all others are disabled!

Thanks sir

closed 2021-03-17 by smilexth