arendst/Tasmota #11366: Can't connect to Mosquitto TLS

smilexth, 2021-03-17:
### PROBLEM DESCRIPTION

1. I have Mosquitto installed with TLS enabled on Docker container (https://github.com/PythonLinks/letsencrypt-mosquitto)
2. Let's Encrypt looks fine
3. Everything works fine when i connected to 1883 but when try to connect to 8883 it has an error with code : 296 which describted as "SSL3_ALERT_HANDSHAKE_FAILURE"
4. I have tls enabled tasmota self-compiled with suggested parameters as describted at https://tasmota.github.io/docs/TLS/ with auto learning param

### REQUESTED INFORMATION

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): _____
- [X] Tasmota binary firmware version number used: 9.3.1
  - [ ] Pre-compiled
  - [X] Self-compiled

### TO REPRODUCE

1. Everything seem fine when switch back to connecting via port 1883 with same mqtt credentials but error still remaining when switch to port 8883 with same credential provided.
2. I have try to define  #define MQTT_FINGERPRINT1 with auto-learning (It won't fix)
3. I have try to define #define USE_MQTT_TLS_CA_CERT  (It won't fix)

### EXPECTED BEHAVIOUR

Successfully connected via port 8883

### SCREENSHOTS

Mosquitto Startup Logs

[image]

Mosquitto Error Log when try to connecting to 8883

[image]

user_config_override.h

[image]

Jason2866, 2021-03-17:
You are using a LetsEncrypt Mosquitto build.
The way you configure Tasmota is NOT for LetsEncrypt certificates.

smilexth, 2021-03-17:
> You are using a LetsEncrypt Mosquitto build. You have to use `#define USE_MQTT_TLS_CA_CERT`
> Have you tested your mqtt broker setup?
> Take a look https://medium.com/@shayaman321/how-to-setup-a-secured-mqtt-broker-c43d4acb95bf

I've using #define USE_MQTT_TLS_CA_CERT but it get no luck

smilexth, 2021-03-17:
with **#define USE_MQTT_TLS_CA_CERT** got an error from mqtt server side as **"no shared cipher"**

any idea ?

smilexth, 2021-03-17:
Now i tried to test with mosquitto_sub to 8883, All works fine. but still can't connect from tasmota

s-hadinger, 2021-03-17:
Can you give details about your mosqitto tls configuration ? Hostname ? Certificate?

Elix-g, 2021-03-17:
Also interesting: TLS ciphers that are available on server site. You can find out which ciphers your server support with nmap:

`nmap --script ssl-enum-ciphers -p 8883 mqtt.server.hostname`

I'm using mosquitto as central mqtt server with TLS (v1.2) on port 8883 without any issues.

smilexth, 2021-03-18:
> Also interesting: TLS ciphers that are available on server site. You can find out which ciphers your server supports using nmap and/or openssl:
> 
> TLSv1.2:
> `nmap --script ssl-enum-ciphers -p 8883 mqtt.server.hostname` or:
> `openssl s_client -tls1_2 -connect mqtt.server.hostname:8883`
> 
> TLSv1.3:
> `openssl s_client -tls1_3 -connect mqtt.server.hostname:8883`
> 
> I'm using mosquitto as central mqtt server with TLS (v1.2) on port 8883 without any issues.

No client certificate CA names sent
Peer signing digest: SHA256
Peer signature type: ECDSA
Server Temp Key: X25519, 253 bits
---
SSL handshake has read 2703 bytes and written 304 bytes
Verification: OK
---
New, TLSv1.2, Cipher is ECDHE-ECDSA-AES256-GCM-SHA384
Server public key is 256 bit
Secure Renegotiation IS supported
Compression: NONE
Expansion: NONE

smilexth, 2021-03-18:
> Can you give details about your mosqitto tls configuration ? Hostname ? Certificate?

```
#MQTT, NO ENCRYPTION
listener 1883
protocol mqtt

#MQTT, ENCRYPTED
listener 8883
protocol mqtt
capath /etc/ssl/certs
keyfile  /root/.caddy/acme/acme-v02.api.letsencrypt.org/sites/mqtt.xxxx.com/mqtt.xxxx.com.key
certfile /root/.caddy/acme/acme-v02.api.letsencrypt.org/sites/mqtt.xxxx.com/mqtt.xxxx.com.crt

#SMX
tls_version tlsv1.2
require_certificate false

#WEBSOCKECTS, NO ENCRYPTION
listener 8080
protocol websockets

#WEBSOCKECTS, ENCRYPTED
listener 8081
protocol websockets
capath /etc/ssl/certs
keyfile  /root/.caddy/acme/acme-v02.api.letsencrypt.org/sites/mqtt.xxxx.com/mqtt.xxxx.com.key
certfile /root/.caddy/acme/acme-v02.api.letsencrypt.org/sites/mqtt.xxxx.com/mqtt.xxxx.com.crt

#WEBSOCKETS ENCRYPTED, SAME ORIGIN ONLY
#NGINX FORWARDS PORT 443 to MOSQUITTO ON PORT 8080
#NOTE NOT ON PORT 8081!

# AUTHENTICATION PLUGINS
# There are so many plugins , requiring so much
# configuration, that they have not been tested with
# this docker container.  So please let me know if they
# work for you or not.
# Personally I am going to be using and testing the JWT plugins.
```

Elix-g, 2021-03-18:
Your openssl output shows

`ECDHE-ECDSA-AES256-GCM-SHA384`

while the result on my machine is

`ECDHE-RSA-AES256-GCM-SHA384`.

Both ciphers are recommended for use with TLS1.2 and are defined in IETF RFC5289 [](https://tools.ietf.org/html/rfc5289) and use Elliptic Curve so you can try to re-compile Tasmota with USE_MQTT_TLS_FORCE_EC_CIPHER.

smilexth, 2021-03-18:
> ECDHE-ECDSA-AES256-GCM-SHA384

[image]

As your mention but i still get no luck

Elix-g, 2021-03-18:
Can you run the nmap command as shown in my previous comment? nmap will show all ciphers supported by your broker.

smilexth, 2021-03-18:
> Can you run the nmap command as shown in my previous comment? nmap will show all ciphers supported by your broker.

```
Starting Nmap 7.80 ( https://nmap.org ) at 2021-03-18 11:36 UTC
Nmap scan report for mqtt.xxx.com (143.198.196.xxx)
Host is up.
Other addresses for mqtt.xxx.com (not scanned): 128.199.145.xxx
rDNS record for 143.198.196.xxx: mqtt-tls

PORT     STATE    SERVICE
8883/tcp filtered secure-mqtt

Nmap done: 1 IP address (1 host up) scanned in 3.16 seconds

```

Elix-g, 2021-03-18:
That's exactly nmap's behaviour when mosquitto is configured to use TLSv1.3. When I enable TLSv1.3 on my broker, I have the same result with nmap and Tasmota also cannot connect to the broker. So the question is, why your broker offers TLSv1.3 although you limit to TLSv1.2 by setting `tls_version tlsv1.2`. Can you rewrite your mosquitto.conf as follows?

```
listener 8883 mqtt.privnet.xmas
protocol mqtt
allow_anonymous false
tls_version tlsv1.2
cafile   /etc/mosquitto/certs/ca.crt
certfile /root/.caddy/acme/acme-v02.api.letsencrypt.org/sites/mqtt.xxxx.com/mqtt.xxxx.com.crt
keyfile  /root/.caddy/acme/acme-v02.api.letsencrypt.org/sites/mqtt.xxxx.com/mqtt.xxxx.com.key
require_certificate false
```

Pay attention to use cafile instead of capath and do not omit the hostname in the listener option.

smilexth, 2021-03-18:
I just switch to another mosquitto tls server, It's now OK

But when i reflash with same parameter with another devices it return "Bad fingerprint" 

```
20:23:16.282 MQT: TLS connection error: 1
20:23:16.286 MQT: Connect failed to mqtt.xxx.com:8883, rc -2. Retry in 40 sec
```

Elix-g, 2021-03-18:
Try `mqttfingerprint1 0` from the web console.

smilexth, 2021-03-18:
> Try `mqttfingerprint1 0` from the web console.

Excellent, Thank you all guys

closed 2021-03-18 by smilexth

Jason2866, 2021-03-18:
@ascillato2  can you move to discussions?

ascillato2, 2021-03-18:
Done