arendst/Tasmota #11367: [Suggestion] Set setoption41 105 as default when wemo/hue emulation is active

raphamotta, 2021-03-17:
Hello,

I have several issues with Alexa not working sometimes (related with #10430 ) and 2 actions solved my problem after several tests:
 - Allow IGMP snooping in my LAN
 - Set option "setoption41 105" for gratuitous ARP every 5 minutes

Not every router has the IGPM option, and I had this problem on my parents house too. Then, "setoption41 105" fixed the issue.

I suggest to put option "setoption41 105" as default when wemo/hue emulation is active to prevent this problem. Any problem to to it?

Thanks

closed 2021-03-17 by close-issue-app[bot]

raphamotta, 2021-03-17:
Where can I suggest it?

raphamotta, 2021-03-17:
> In discussions. But this is not a solution for a Multicast problem.
> Setoption 41 is a workaround for routers which has not correctly implemented ARP treatment.
> This issue is a general problem and affects **all** Tasmota functions.

Well, if it's a workaround to some routers, why not to keep it by default?

jaymemaurice, 2021-03-17:
Please don't. Don't encourage fixing some broken networking devices by shipping default broken network protocols. 
I personally don't want ARP broadcasts from all of my Tasmota devices waking up all the devices on my network because someone else's junk router/switch can't handle IGMP or keep it's ARP state.

Jason2866, 2021-03-17:
Thats why Setoption41 is off by default. This will not be changed.

suzsuzuki, 2023-01-08:
> Thats why Setoption41 is off by default. This will not be changed.

Well... the default value of setoption41 is 60 for a while. Never say never..

rockbruno, 2025-04-21:
I found this thread after seeing tasmota spamming my networking for no reason and trying to find out why. Am also surprised they made this a default... In any case, thankfully you can disable it with `SetOption41 0`

s-hadinger, 2025-04-22:
Sending one ARP packet every 105 seconds hardly qualifies for "spamming". Sending gratuitous ARP seemed to solve some multicast problems.