arendst/Tasmota #11372: HC-SR04 issue

traverza, 2021-03-17:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Connecting ultrasonic range sensor HC-SR04 or HC-SR04T 2.0 shows data and correct distance. When distance is greater than 300cm no data is shown. Both sensors shows correct measurements greater than 300cm when connected to Arduino with simple sketch. HC-SR04T 2.0 works OK on 3.3V on Arduino and shows correct measurements greater than 300cm, same with 3.3V on D1 Mini with Tasmota but only up to 300cm.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): D1 Mini
- [x] Tasmota binary firmware version number used: 9.3.1
  - [x] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: Tasmotizer, OTA
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

17:11:12.782 CMD: Backlog Template; Module; GPIO 255
17:11:12.836 RSL: stat/tasmota_610797/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
17:11:13.037 RSL: stat/tasmota_610797/RESULT = {"Module":{"18":"Generic"}}
17:11:13.274 RSL: stat/tasmota_610797/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"1856":"SR04 Tri/TX"},"GPIO13":{"0":"None"},"GPIO14":{"1888":"SR04 Ech/RX"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

17:12:25.057 CMD: status 0
17:12:25.063 RSL: stat/tasmota_610797/STATUS = {"Status":{"Module":18,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_610797","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
17:12:25.091 RSL: stat/tasmota_610797/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-minimal.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:03:57","StartupUTC":"2021-03-17T16:08:28","Sleep":50,"CfgHolder":4617,"BootCount":32,"BCResetTime":"2021-03-16T17:28:27","SaveCount":56,"SaveAddress":"F6000"}}
17:12:25.127 RSL: stat/tasmota_610797/STATUS2 = {"StatusFWR":{"Version":"9.3.1(sensors)","BuildDateTime":"2021-03-09T16:13:23","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"357/699"}}
17:12:25.148 RSL: stat/tasmota_610797/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["removed",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
17:12:25.182 RSL: stat/tasmota_610797/STATUS4 = {"StatusMEM":{"ProgramSize":645,"Free":356,"Heap":23,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"16400E","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDA8787","0415A005","B7FFBFCF","01DA9BC4","64367CC7","00084052","20000000","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,51,52,55,56,58,59,62,64,66,67,74"}}
17:12:25.223 RSL: stat/tasmota_610797/STATUS5 = {"StatusNET":{"Hostname":"tasmota_610797-1943","IPAddress":"192.168.1.106","Gateway":"192.168.1.13","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.11","Mac":"DC:4F:22:61:07:97","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
17:12:25.244 RSL: stat/tasmota_610797/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_610797","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
17:12:25.266 RSL: stat/tasmota_610797/STATUS7 = {"StatusTIM":{"UTC":"2021-03-17T16:12:25","Local":"2021-03-17T17:12:25","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"06:58","Sunset":"18:57"}}
17:12:25.287 RSL: stat/tasmota_610797/STATUS10 = {"StatusSNS":{"Time":"2021-03-17T17:12:25","SR04":{"Distance":256.368}}}
17:12:25.295 RSL: stat/tasmota_610797/STATUS11 = {"StatusSTS":{"Time":"2021-03-17T17:12:25","Uptime":"0T00:03:57","UptimeSec":237,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":37,"MqttCount":0,"Wifi":{"AP":1,"SSId":"<removed>,"Channel":6,"RSSI":66,"Signal":-67,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

no console output when measurement disappear

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Setup tasmota-sensors without any settings, connect SR04 sensor according to the documentation, measure distance greater than 300cm

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Measurement of distance greater than 300cm, sensors are capable of measuring up to 450cm and working OK on Arduino

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

jaymemaurice, 2021-03-17:
Can you show the weblog4 showing the measurements up until the last successful measurement?
Is the last successful measurement around 327.67cm? 
If so, perhaps we are storing this measurement somewhere as a signed when we could use unsigned.

Jason2866, 2021-03-17:
As designed https://github.com/arendst/Tasmota/blob/development/tasmota/xsns_22_sr04.ino#L67

traverza, 2021-03-18:
Is there any reason to limit the distance measurement to 300 cm?

arendst, 2021-03-18:
> Is there any reason to limit the distance measurement to 300 cm?

Change it to whatever you like and let us know if it works consistently.

traverza, 2021-03-18:
So I've change the value from 300 to 450 and it is now showing correct distance without any problems. There is some jitter in the data, but that is most likely a sensor problem. So can this be changed in the main version or do I have to build my own every time there is an update? Thank you for your work, I greatly appreciate it!

[image]

ascillato2, 2021-03-18:
The datasheet of the HC-SR04 set the maximum measurement to 400 cm. So, it is a good improvement to use the whole range of the sensor.
Can you provide a PR to increase the limit?  Thanks.

arendst, 2021-03-18:
I''ll change it tomorrow.

traverza, 2021-03-18:
I'm using the JSN-SR04T-2.0 sensor that should have upper limit of 600cm - https://www.jahankitshop.com/getattach.aspx?id=4635&Type=Product

I'll check this range tomorrow and report back, I don't have enough space here.

closed 2021-03-19 by ascillato2