arendst/Tasmota #11379: PMS5003 sensor reboot esp8266 due to watchdog or exception

holblin, 2021-03-18:
### PROBLEM DESCRIPTION
Tasmota reboot due to watchdog timeout or exception (not always the same reason) on my Wemos D1 mini with the latest Tasmota 9.3.1 sensors. Both TX and RX are connected. Nothing else is connected to the Wemos D1 mini.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Wemos D1 Mini
- [X] Tasmota binary firmware version number used: 9.3.1-sensors
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: Upload through UI, previous version with tasmotizer
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
11:10:11.341 CMD: Backlog Template; Module; GPIO 255
11:10:11.365 MQT: stat/air_quality_sensor/RESULT = {"NAME":"Air Quality Sensor","GPIO":[1,1,1696,1,1,1,1,1,1664,1,1,1,1,1],"FLAG":0,"BASE":18}
11:10:11.595 MQT: stat/air_quality_sensor/RESULT = {"Module":{"0":"Air Quality Sensor"}}
11:10:11.844 MQT: stat/air_quality_sensor/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"1696":"PMS5003 Rx"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"1664":"PMS5003 Tx"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
11:10:45.068 CMD: Backlog Rule1; Rule2; Rule3
11:10:45.093 MQT: stat/air_quality_sensor/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
11:10:45.337 MQT: stat/air_quality_sensor/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
11:10:45.567 MQT: stat/air_quality_sensor/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
11:11:09.674 CMD: Status 0
11:11:09.684 MQT: stat/air_quality_sensor/STATUS = {"Status":{"Module":0,"DeviceName":"Air Quality Sensor","FriendlyName":["Air Quality Sensor"],"Topic":"air_quality_sensor","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
11:11:09.720 MQT: stat/air_quality_sensor/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-sensors.bin.gz	","RestartReason":"Exception","Uptime":"0T00:05:28","StartupUTC":"2021-03-18T18:05:41","Sleep":50,"CfgHolder":4617,"BootCount":2,"BCResetTime":"2021-03-18T10:34:30","SaveCount":1933,"SaveAddress":"F6000"}}
11:11:09.754 MQT: stat/air_quality_sensor/STATUS2 = {"StatusFWR":{"Version":"9.3.1(sensors)","BuildDateTime":"2021-03-09T16:13:23","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"487/699"}}
11:11:09.774 MQT: stat/air_quality_sensor/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["holblin_domotic",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00028009","2805C8000100060000005A0A000000000000","00000280","00006000","00000000"]}}
11:11:09.808 MQT: stat/air_quality_sensor/STATUS4 = {"StatusMEM":{"ProgramSize":645,"Free":356,"Heap":20,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDA8787","0415A005","B7FFBFCF","01DA9BC4","64367CC7","00084052","20000000","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,51,52,55,56,58,59,62,64,66,67,74"}}
11:11:09.851 MQT: stat/air_quality_sensor/STATUS5 = {"StatusNET":{"Hostname":"air_quality_sensor-0157","IPAddress":"192.168.1.100","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"40:F5:20:24:60:9D","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
11:11:09.873 MQT: stat/air_quality_sensor/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.240","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_24609D","MqttUser":"mosquitto","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
11:11:09.896 MQT: stat/air_quality_sensor/STATUS7 = {"StatusTIM":{"UTC":"2021-03-18T18:11:09","Local":"2021-03-18T11:11:09","StartDST":"2021-03-14T02:00:00","EndDST":"2021-11-07T02:00:00","Timezone":99,"Sunrise":"07:15","Sunset":"19:18"}}
11:11:09.918 MQT: stat/air_quality_sensor/STATUS10 = {"StatusSNS":{"Time":"2021-03-18T11:11:09","PMS5003":{"CF1":5,"CF2.5":7,"CF10":8,"PM1":5,"PM2.5":7,"PM10":8,"PB0.3":1179,"PB0.5":332,"PB1":48,"PB2.5":4,"PB5":0,"PB10":0}}}
11:11:09.936 MQT: stat/air_quality_sensor/STATUS11 = {"StatusSTS":{"Time":"2021-03-18T11:11:09","Uptime":"0T00:05:28","UptimeSec":328,"Heap":20,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"holblin_domotic","BSSId":"76:AC:B9:92:F4:FE","Channel":6,"RSSI":100,"Signal":-35,"LinkCount":1,"Downtime":"0T00:00:06"}}}
11:11:09.970 MQT: stat/air_quality_sensor/STATUS12 = {"StatusSTK":{"Exception":29,"Reason":"Exception","EPC":["4000e1b2","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["4028fb80","40101740","402762df","4028139c","402762d3","40263bdf","402679cf","40282572","40284086","40269993","40269564","402690e4","40280f78","4026959f","4026b514","4026b4d8","40269554","40269528","402694f3","40100d19","40269a61","4027407f","402529d8","40279d99","40279dde","402515a0","40000f49","40000f49","40000f49","40000e19","40105d21"]}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Wait and see the count reboot increase and reason reboot update

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
I expect to have no reboot.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
The build was done according to https://www.youtube.com/watch?v=5FdTRIq6_mY&ab_channel=SuperHouseTV
I added the TX connection after I've seen a lot of reboots but this doesn't;t solve the issue.

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2021-03-18:
Hi, can you provide the log requested on the latest point of the template? Please, remember to set WEBLOG to 4. Provide also the complete log when your device starts.

holblin, 2021-03-18:
Hi @ascillato ,
I set the WEBLOG to 4 already but I have difficulties catching the issue in the logs. You see, when the issue appears (which is random to my knowledge) I couldn't copy the logs because my Wemos D1 Mini is rebooting and the Web is not responding. Once the Wemos D1 Mini has restarted, I can't see the past logs.

I will try to do my best to catch one of these logs but could you remove the `template missing/incomplete` label in the meantime?

ascillato, 2021-03-18:
Please, try to share the full log when your device starts. I'm interested on the part that say if it is using hardware serial or software serial

holblin, 2021-03-18:
Ok, I post the log that I get after a restart into the initial message.

I assume this is a software serial since Wemos D1 Mini pin D4 (GPIO 2) is not hardware Serial Pin according to the Wemos D1 Mini pinout diagram.

ascillato2, 2021-03-18:
Thanks for sharing your logs. Are you able to test your PMS5003 on TX and RX pins of your Wemos D1 mini?

holblin, 2021-03-18:
No, I will try today/tomorrow to solder a second similar setup with TX and RX to have a comparison point :-)

Jason2866, 2021-03-19:
Can not reproduce. Code for PMS5003 and the PMS7003 is the same
```
00:00:00.052 CFG: Loaded from flash at F6, Count 166
00:00:00.058 SER: Set to 8N1 9600 bit/s
00:00:00.059 QPC: Count 1
00:00:00.065 Project sonoff PMS7003 Version 9.3.1.2(UNIVERSAL)-2_7_4_9(2021-03-19T11:08:55)
00:00:00.067 SNS: Hardware Serial
00:00:04.459 WIF: Connecting to AP1 Jason_Home_WLAN Channel 4 BSSId 88:C3:97:B1:1D:56 in mode 11n as sonoff-1585B3-1459...
00:00:06.518 QPC: Reset
00:00:06.523 WIF: Connected
00:00:06.752 HTP: Web server active on sonoff-1585B3-1459 with IP address 192.168.2.110
13:42:13.260 MQT: Attempting connection...
13:42:13.287 MQT: Connected
13:42:13.290 MQT: tele/sonoff-1585B3/LWT = Online (retained)
13:42:13.292 MQT: cmnd/sonoff-1585B3/POWER = 
13:42:13.297 MQT: tele/sonoff-1585B3/INFO1 = {"Module":"PMS7003","Version":"9.3.1.2(UNIVERSAL)","FallbackTopic":"cmnd/sonoff-1585B3_fb/","GroupTopic":"cmnd/sonoffs/"}
13:42:13.300 MQT: tele/sonoff-1585B3/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-1585B3-1459","IPAddress":"192.168.2.110"}
13:42:13.301 MQT: tele/sonoff-1585B3/INFO3 = {"RestartReason":"External System"}
13:42:18.474 MQT: tele/sonoff-1585B3/STATE = {"Time":"2021-03-19T13:42:18","Uptime":"0T00:00:14","UptimeSec":14,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","BSSId":"88:C3:97:B1:1D:56","Channel":4,"RSSI":100,"Signal":-46,"LinkCount":1,"Downtime":"0T00:00:08"}}
13:42:48.640 CMD: teleperid 300
13:42:48.647 MQT: stat/sonoff-1585B3/RESULT = {"Command":"Unknown"}
13:42:57.141 CMD: teleperiod 300
13:42:57.148 MQT: stat/sonoff-1585B3/RESULT = {"TelePeriod":300}
13:42:57.474 MQT: tele/sonoff-1585B3/STATE = {"Time":"2021-03-19T13:42:57","Uptime":"0T00:00:53","UptimeSec":53,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","BSSId":"88:C3:97:B1:1D:56","Channel":4,"RSSI":100,"Signal":-47,"LinkCount":1,"Downtime":"0T00:00:08"}}
13:42:57.479 MQT: tele/sonoff-1585B3/SENSOR = {"Time":"2021-03-19T13:42:57","PMS5003":{"CF1":48,"CF2.5":67,"CF10":73,"PM1":32,"PM2.5":47,"PM10":60,"PB0.3":8262,"PB0.5":2500,"PB1":428,"PB2.5":37,"PB5":10,"PB10":4}}```

Jason2866, 2021-03-19:
I would make a bet that the issue is caused from the Vreg on your D1 mini board.
Nearly all sold boards are clones and uses a Vreg which only delivers about 150mA.
Change the Vreg to a type which can deliver 500mA or use a NodeMCU board (delivers >500mA)

holblin, 2021-03-19:
I don't have another esp 8266 board on hand for now. I will try to found another one.

About the 5V regulation, the PMS5003 is connected to the 5v of my Wemos D1 Mini. This doesn't have any regulation, the regulation comes directly from an Anker charger and I'm pretty sure it could deliver the 500mA. Maybe I miss something but what is the potential issue there?

Jason2866, 2021-03-19:
The ESP8266 is powered with 3.3V. The 3.3V are generated on the D1 mini board (Vreg).
Many of this boards have a to weak Vreg assembled.
If your 5V power supply could deliver 100A would not solve this issue.

holblin, 2021-03-21:
After 21 boot count, I rewire the module on Hardware RX and TX. I still looking on another esp8266 to test @Jason2866 hypothesis.

holblin, 2021-03-23:
Hi,
After switching to Hardware RX and TX pin I don't have any reboot. I think the issue is now solved :-)

_BTW @Jason2866 , I still use my Wemos D1 mini, so it doesn't seem the Vreg was the issue. To back my claim on that I checked the schematic and while I'm not an expert I'm pretty sure that the 5v pin on the Wemos D1 Mini doesn't have any regulation and it's directly connected to the 5v of the USB connection ;-) And I used the 5v for powering the PMS5003 :-)_

holblin, 2021-03-23:
Thanks for all the help guys, I will continue to monitor the reboot for a while but the situation is not concerning for me anymore :-)

closed 2021-03-23 by holblin

holblin, 2021-08-13:
FYI, at this date, my boot count is still 22 :-)