arendst/Tasmota #11389: CSE7766 power monitoring

ahmaddxb, 2021-03-19:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

I have a plug with CSE7766 chip for power monitoring which works but after a period of time it stop seeing the power data.
A restart doesn't help, I need to turn it off from the main or unplug it from the wall socket and back again.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

17:20:15.078 MQT: stat/sonoff/mini_fridge/STATUS = {"Status":{"Module":0,"DeviceName":"Mini Fridge","FriendlyName":["Mini Fridge"],"Topic":"sonoff/mini_fridge","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
17:20:15.083 MQT: stat/sonoff/mini_fridge/STATUS1 = {"StatusPRM":{"Baudrate":4800,"SerialConfig":"8E1","GroupTopic":"sonoffs","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:04:59","StartupUTC":"2021-03-19T13:15:16","Sleep":50,"CfgHolder":4617,"BootCount":73,"BCResetTime":"2020-07-16T13:06:43","SaveCount":432,"SaveAddress":"F6000"}}
17:20:15.089 MQT: stat/sonoff/mini_fridge/STATUS2 = {"StatusFWR":{"Version":"9.3.1(tasmota)","BuildDateTime":"2021-02-23T12:02:38","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"408/699"}}
17:20:15.094 MQT: stat/sonoff/mini_fridge/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["ASUS",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["002AA009","2805C8000100068000005A0A000000000000","000002C8","00006000","00000000"]}}
17:20:15.104 MQT: stat/sonoff/mini_fridge/STATUS4 = {"StatusMEM":{"ProgramSize":593,"Free":408,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
17:20:15.111 MQT: stat/sonoff/mini_fridge/STATUS5 = {"StatusNET":{"Hostname":"sonoff/mini_fridge-8064","IPAddress":"192.168.1.211","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.104","Mac":"C8:2B:96:45:3F:80","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
17:20:15.118 MQT: stat/sonoff/mini_fridge/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.235","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_453F80","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
17:20:15.126 MQT: stat/sonoff/mini_fridge/STATUS7 = {"StatusTIM":{"UTC":"2021-03-19T13:20:15","Local":"2021-03-19T17:20:15","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+04:00","Sunrise":"09:54","Sunset":"22:00"}}
17:20:15.131 MQT: stat/sonoff/mini_fridge/STATUS9 = {"StatusPTH":{"PowerDelta":[101,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
17:20:15.139 MQT: stat/sonoff/mini_fridge/STATUS10 = {"StatusSNS":{"Time":"2021-03-19T17:20:15","ENERGY":{"TotalStartTime":"2020-07-16T12:59:53","Total":9.149,"Yesterday":0.000,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
17:20:15.144 MQT: stat/sonoff/mini_fridge/STATUS11 = {"StatusSTS":{"Time":"2021-03-19T17:20:15","Uptime":"0T00:04:59","UptimeSec":299,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"ASUS","BSSId":"B4:FB:E4:71:42:B0","Channel":11,"RSSI":80,"Signal":-60,"LinkCount":1,"Downtime":"0T00:00:07"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2021-03-19:
Without any logs or more infos we can not help.
Please enable `weblog 3` and submit the log files when the issue occurs

ahmaddxb, 2021-03-19:
I've enabled weblog 4 now the plug is still not reporting power data but there is nothing in the log to see

Jason2866, 2021-03-19:
Without any info we do not know where to search. Sorry

ahmaddxb, 2021-03-19:
17:37:35.453 DMP: 55 5A 02 FB 48 00 03 2B 00 40 4C 00 1F BB 52 65 C0 00 32 52 71 02 3D 84
17:37:35.559 DMP: 55 5A 02 FB 48 00 03 2B 00 40 4C 00 1F BB 52 65 C0 00 32 5D 71 02 45 97
17:37:35.664 DMP: 55 5A 02 FB 48 00 03 2B 00 40 4C 00 1F BB 52 65 C0 00 32 5D 71 02 4D 9F
17:37:35.770 DMP: 55 5A 02 FB 48 00 03 2B 00 40 4C 00 1F BB 52 65 C0 00 32 5D 71 02 55 A7
17:37:35.876 DMP: 55 5A 02 FB 48 00 03 2B 00 40 4C 00 1F BB 52 65 C0 00 32 50 71 02 5D A2
17:37:35.981 DMP: 55 5A 02 FB 48 00 03 2B 00 40 4C 00 1F BB 52 65 C0 00 32 50 71 02 65 AA
17:37:36.086 DMP: 55 5A 02 FB 48 00 03 2B 00 40 4C 00 1F BB 52 65 C0 00 32 50 71 02 6D B2
17:37:36.193 DMP: 55 5A 02 FB 48 00 03 2B 00 40 4C 00 1F BB 52 65 C0 00 32 55 71 02 76 C0
17:37:36.298 DMP: 55 5A 02 FB 48 00 03 2B 00 40 4C 00 1F CE 52 65 C0 00 32 55 71 02 7E DB
17:37:36.403 DMP: 55 5A 02 FB 48 00 03 2B 00 40 4C 00 1F CE 52 65 C0 00 32 55 71 02 86 E3
17:37:36.454 APP: Boot Count 76
17:37:36.509 DMP: 55 5A 02 FB 48 00 03 2B 00 40 4C 00 1F CE 52 65 C0 00 32 54 71 02 8E EA
17:37:36.615 DMP: 55 5A 02 FB 48 00 03 2B 00 40 4C 00 1F CE 52 65 C0 00 32 54 71 02 96 F2
17:37:36.720 DMP: 55 5A 02 FB 48 00 03 2B 00 40 4C 00 1F CE 52 65 C0 00 32 54 71 02 9E FA
17:37:36.826 DMP: 55 5A 02 FB 48 00 03 2B 00 40 4C 00 1F CE 52 65 C0 00 32 60 71 02 A6 0E
17:37:36.932 DMP: 55 5A 02 FB 48 00 03 2B 00 40 4C 00 1F CE 52 65 C0 00 32 60 71 02 AF 17
17:37:37.038 DMP: 55 5A 02 FB 48 00 03 2B 00 40 4C 00 1F CE 52 65 C0 00 32 60 71 02 B7 1F
17:37:37.144 DMP: 55 5A 02 FB 48 00 03 2B 00 40 4C 00 1F CE 52 65 C0 00 32 64 71 02 BF 2B
17:37:37.249 DMP: 55 5A 02 FB 48 00 03 2B 00 40 4C 00 1F CE 52 65 C0 00 32 64 71 02 C7 33
17:37:37.402 CFG: Saved to flash at F7, Count 439, Bytes 4096
17:37:37.408 DMP: 55 5A 02 FB 48 00 03 2B 00 40 4C 00 1F CE 52 65 C0 00 32 64 71 02 CF 3B
17:37:37.460 DMP: 55 5A 02 FB 48 00 03 2B 00 40 4C 00 1F CE 52 65 C0 00 32 7D 71 02 D7 5C
17:37:37.566 DMP: 55 5A 02 FB 48 00 03 2B 00 40 4C 00 1F CE 52 65 C0 00 32 7D 71 02 DF 64
17:37:37.671 DMP: 55 5A 02 FB 48 00 03 2B 00 40 4C 00 1F CE 52 65 C0 00 32 7D 71 02 E7 6C
17:37:37.777 DMP: 55 5A 02 FB 48 00 03 2B 00 40 4C 00 1F CE 52 65 C0 00 32 9B 71 02 EF 92
17:37:37.882 DMP: 55 5A 02 FB 48 00 03 2B 00 40 4C 00 1F CE 52 65 C0 00 32 9B 71 02 F7 9A
17:37:37.988 DMP: 55 5A 02 FB 48 00 03 2B 00 40 4C 00 1F CE 52 65 C0 00 32 9B 71 02 FF A2
17:37:38.092 DMP: 55 5A 02 FB 48 00 03 2B 00 40 4C 00 20 03 52 65 C0 00 32 AA 71 03 08 F1
17:37:38.198 DMP: 55 5A 02 FB 48 00 03 2B 00 40 4C 00 20 03 52 65 C0 00 32 AA 71 03 10 F9
17:37:38.304 DMP: 55 5A 02 FB 48 00 03 2B 00 40 4C 00 20 03 52 65 C0 00 32 AA 71 03 18 01
17:37:38.409 DMP: 55 5A 02 FB 48 00 03 2B 00 40 4C 00 20 03 52 65 C0 00 32 AB 71 03 20 0A
17:37:38.464 MQT: tele/sonoff/mini_fridge/MARGINS = {"MARGINS":{"PowerDelta":-3}}
17:37:38.472 MQT: tele/sonoff/mini_fridge/SENSOR = {"Time":"2021-03-19T17:37:38","ENERGY":{"TotalStartTime":"2020-07-16T12:59:53","Total":9.150,"Yesterday":0.000,"Today":0.001,"Power":416,"ApparentPower":475,"ReactivePower":230,"Factor":0.87,"Voltage":238,"Current":1.992}}
17:37:38.515 DMP: 55 5A 02 FB 48 00 03 2B 00 40 4C 00 20 03 52 65 C0 00 32 AB 71 03 28 12
17:37:38.620 DMP: 55 5A 02 FB 48 00 03 2B 00 40 4C 00 20 03 52 65 C0 00 32 AB 71 03 30 1A
17:37:38.726 DMP: 55 5A 02 FB 48 00 03 2B 00 40 4C 00 20 03 52 65 C0 00 32 BC 71 03 38 33
17:37:38.832 DMP: 55 5A 02 FB 48 00 03 2B 00 40 4C 00 20 03 52 65 C0 00 32 BC 71 03 40 3B
17:37:38.937 DMP: 55 5A 02 FB 48 00 03 2B 00 40 4C 00 20 03 52 65 C0 00 32 BC 71 03 48 43
17:37:39.043 DMP: 55 5A 02 FB 48 00 03 2B 00 40 4C 00 20 03 52 65 C0 00 32 B3 71 03 50 42
17:37:39.147 HTP: Console
17:37:39.194 DMP: 55 5A 02 FB 48 00 03 2B 00 40 4C 00 20 03 52 65 C0 00 32 B3 71 03 58 4A
17:37:39.254 DMP: 55 5A 02 FB 48 00 03 2B 00 40 4C 00 20 03 52 65 C0 00 32 B3 71 03 60 52
17:37:39.429 DMP: 55 5A 02 FB 48 00 03 2B 00 40 4C 00 20 03 52 65 C0 00 32 B0 71 03 68 57
17:37:39.440 MQT: tele/sonoff/mini_fridge/STATE = {"Time":"2021-03-19T17:37:39","Uptime":"0T00:00:13","UptimeSec":13,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"ASUS","BSSId":"B4:FB:E4:71:42:B0","Channel":11,"RSSI":80,"Signal":-60,"LinkCount":1,"Downtime":"0T00:00:07"}}
17:37:39.452 MQT: tele/sonoff/mini_fridge/SENSOR = {"Time":"2021-03-19T17:37:39","ENERGY":{"TotalStartTime":"2020-07-16T12:59:53","Total":9.150,"Yesterday":0.000,"Today":0.001,"Period":1,"Power":415,"ApparentPower":475,"ReactivePower":231,"Factor":0.87,"Voltage":238,"Current":1.992}}
17:37:39.465 DMP: 55 5A 02 FB 48 00 03 2B 00 40 4C 00 20 03 52 65 C0 00 32 B0 71 03 71 60
17:37:39.571 DMP: 55 5A 02 FB 48 00 03 2B 00 40 4C 00 20 03 52 65 C0 00 32 B0 71 03 79 68
17:37:39.677 DMP: 55 5A 02 FB 48 00 03 2B 00 40 4C 00 20 03 52 65 C0 00 32 A5 71 03 81 65
17:37:39.782 DMP: 55 5A 02 FB 48 00 03 2B 00 40 4C 00 20 03 52 65 C0 00 32 A5 71 03 89 6D
17:37:39.888 DMP: 55 5A 02 FB 48 00 03 2B 00 40 4C 00 20 03 52 65 C0 00 32 A5 71 03 91 75
17:37:39.993 DMP: 55 5A 02 FB 48 00 03 2B 00 40 4C 00 20 18 52 65 C0 00 32 B0 71 03 99 9D
17:37:40.100 DMP: 55 5A 02 FB 48 00 03 2B 00 40 4C 00 20 18 52 65 C0 00 32 B0 71 03 A1 A5
17:37:40.205 DMP: 55 5A 02 FB 48 00 03 2B 00 40 4C 00 20 18 52 65 C0 00 32 B0 71 03 A9 AD
17:37:40.310 DMP: 55 5A 02 FB 48 00 03 2B 00 40 4C 00 20 18 52 65 C0 00 32 BE 71 03 B1 C3
17:37:40.416 DMP: 55 5A 02 FB 48 00 03 2B 00 40 4C 00 20 18 52 65 C0 00 32 BE 71 03 B9 CB
[... cut ...]
17:37:43.689 DMP: 55 5A 02 FB 48 00 03 2D 00 40 4C 00 20 4E 52 65 C0 00 33 02 71 04 B2 42
17:37:43.794 DMP: 55 5A 02 FB 48 00 03 2D 00 40 4C 00 20 4E 52 65 C0 00 32 FF 71 04 BA 46
17:37:43.899 DMP: 55 5A 02 FB 48 00 03 2D 00 40 4C 00 20 4E 52 65 C0 00 32 FF 71 04 C2 4E
17:37:44.005 DMP: 55 5A 02 FB 48 00 03 2D 00 40 4C 00 20 4E 52 65 C0 00 32 FF 71 04 CB 57
17:37:44.111 DMP: 55 5A 02 FB 48 00 03 2D 00 40 4C 00 20 4E 52 65 C0 00 33 06 71 04 D3 67
17:37:44.216 DMP: 55 5A 02 FB 48 00 03 2D 00 40 4C 00 20 4E 52 65 C0 00 33 06 71 04 DB 6F

ahmaddxb, 2021-03-19:
that how it look when I unplug it and plug it back in and power data is working

Jason2866, 2021-03-19:
Please post output of `Status 9`

ahmaddxb, 2021-03-19:
18:42:59.264 CMD: Status 9
18:42:59.266 SRC: WebConsole from 192.168.1.200
18:42:59.270 CMD: Group 0, Index 1, Command "STATUS", Data "9"
18:42:59.276 MQT: stat/sonoff/mini_fridge/STATUS9 = {"StatusPTH":{"PowerDelta":[101,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}

Jason2866, 2021-03-19:
Try with a higher `powerdelta` value. Your choosen `101` is only a change of 1 Watt!
Most devices are not capable to such a precise measurement and generates a lot of traffic
and maybe crashes the device. I would recommend to change to `110` for tests

ahmaddxb, 2021-03-19:
ok will try

ahmaddxb, 2021-03-19:
I have other plugs HLW chip and sonoff pow all handling powerdelta 101 without issues.

ascillato2, 2021-03-22:
Hi, any news on this?

ahmaddxb, 2021-03-22:
Well its still showing power data since I made the change

ahmaddxb, 2021-03-22:
Also have another plug same model with powerdelta 101 which never stops showing data for as long as I've had it, maybe in a few more days I can close this issue

ascillato2, 2021-03-22:
Glad to know that now it is working for you.

Please, tell us in few days if it continue working. If not, please ask to reopen this issue. Thanks.

closed 2021-03-22 by ascillato2