arendst/Tasmota #11400: FW ISSUE WITH SONOFF 4CH TASMOTA STARTING FROM 9.0 TO 9.3.1 SETOPTION80=1

mircozappon, 2021-03-20:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
No status retrieved (instead errore returned) from json for more than one shutterrelay. I found this issue in TasmoAdmin and it is related to setoption80 when active (for blinds and rollers).
The issue appear only when 2 shutterrelay are defined shutterrelay1=1 shutterrelay2=3. If setoption80 = 0 no errors present in TasmoAdmin. I use 7 sonoff 4ch for my covers with shuttermode 1, Interlock 1,2, interlock on, 3 are right because I define only shutterarray1 the 4 having also shutterrelay2=3 have the issue.

I want to clarify that in homeassistanst and mqtt all is perfect and right, no issues. I can connect to device from TasmoAdmin, but it always show offline status with red icon.

The issue seems to be a json problem, like reloxx13 found in TasmoAdmin log. Here attached the point where error is, and it is present since sonoff version after 8.3.1:
[image]

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): SONOFF 4CH
- [x ] Tasmota binary firmware version number used: 9.3.1
  - [x] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: espeasy
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
  - [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
Backlog Rule1; Rule2; Rule3
20:52:07.949 MQT: stat/dbs/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":280,"Free":231,"Rules":"ON Shutter1#Position DO backlog Publish status/%topic%/level {\"%value%\"}; Publish cmnd/doubleroom_switch_01/mem1 %value% ENDON ON Shutter2#Position DO Backlog Publish status/%topic%/level {\"%value%\"}; publish cmnd/doubleroom_switch_01/mem3 %value%; mem3 %mem2%; mem2 %value% ENDON"}
20:52:08.186 MQT: stat/dbs/RESULT = {"Rule2":"ON","Once":"OFF","StopOnError":"OFF","Length":303,"Free":208,"Rules":"ON Power1#state=0 do backlog publish cmnd/doubleroom_switch_01/mem2 0 ENDON ON Power2#state=0 do backlog publish cmnd/doubleroom_switch_01/mem2 0 ENDON ON Power3#state=0 do backlog publish cmnd/doubleroom_switch_01/mem4 0 ENDON ON Power4#state=0 do backlog publish cmnd/doubleroom_switch_01/mem4 0 ENDON"}
20:52:08.434 MQT: stat/dbs/RESULT = {"Rule3":"ON","Once":"OFF","StopOnError":"OFF","Length":109,"Free":402,"Rules":"ON mem1#State do if (mem1==1) POWER2 off ENDIF ENDON ON POWER2#State=1 do if (mem1==1) POWER2 off ENDIF ENDON"}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
status0
20:51:07.532 MQT: stat/dbs/STATUS = {"Status":{"Module":7,"DeviceName":"dbs","FriendlyName":["dbs_door_shutter_up","dbs_door_shutter_down","dbs_window_shutter_up","dbs_window_shutter_down"],"Topic":"dbs","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":300,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1,"InfoRetain":0,"StateRetain":0}}
20:51:10.040 MQT: stat/dbs/STATUS = {"Status":{"Module":7,"DeviceName":"dbs","FriendlyName":["dbs_door_shutter_up","dbs_door_shutter_down","dbs_window_shutter_up","dbs_window_shutter_down"],"Topic":"dbs","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":300,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1,"InfoRetain":0,"StateRetain":0}}
20:51:10.046 MQT: stat/dbs/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T01:58:47","StartupUTC":"2021-03-20T17:52:23","Sleep":50,"CfgHolder":4617,"BootCount":19,"BCResetTime":"2021-03-03T16:53:38","SaveCount":8248,"SaveAddress":"FA000"}}
20:51:10.053 MQT: stat/dbs/STATUS2 = {"StatusFWR":{"Version":"9.3.1(tasmota)","BuildDateTime":"2021-02-26T13:42:23","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"453/699"}}
20:51:10.057 MQT: stat/dbs/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["CHAOS",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000C029","2805C8000100060000005A0A000000000000","40000080","00006000","00000000"]}}
20:51:10.066 MQT: stat/dbs/STATUS4 = {"StatusMEM":{"ProgramSize":605,"Free":396,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
20:51:10.073 MQT: stat/dbs/STATUS5 = {"StatusNET":{"Hostname":"dbs-2876","IPAddress":"192.168.100.200","Gateway":"192.168.100.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.100.1","Mac":"DC:4F:22:C7:0B:3C","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
20:51:10.078 MQT: stat/dbs/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.100.150","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_C70B3C","MqttUser":"root","MqttCount":3,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
20:51:10.085 MQT: stat/dbs/STATUS7 = {"StatusTIM":{"UTC":"2021-03-20T19:51:10","Local":"2021-03-20T20:51:10","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"06:52","Sunset":"19:02"}}
20:51:10.092 MQT: stat/dbs/STATUS10 = {"StatusSNS":{"Time":"2021-03-20T20:51:10","Shutter1":{"Position":0,"Direction":0,"Target":0},"Shutter2":{"Position":0,"Direction":0,"Target":0}}}
20:51:10.100 MQT: stat/dbs/STATUS11 = {"StatusSTS":{"Time":"2021-03-20T20:51:10","Uptime":"0T01:58:47","UptimeSec":7127,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":22,"MqttCount":3,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"CHAOS","BSSId":"8C:DE:F9:21:D4:4A","Channel":1,"RSSI":88,"Signal":-56,"LinkCount":1,"Downtime":"0T00:00:03"}}}
20:51:10.105 MQT: stat/dbs/STATUS13 = {"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":324,"Close":314,"50perc":68,"Delay":0,"Opt":"0000","Calib":[2340,414,5094,900,1000],"Mode":"0"}},{"SHT1":{"Relay1":3,"Relay2":4,"Open":222,"Close":220,"50perc":66,"Delay":0,"Opt":"0000","Calib":[1430,301,3161,900,1000],"Mode":"0"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
The behavior appear using more than one shutterrelay.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]
[image]
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2021-03-22:
Fixed in development release (currently 9.3.1.2)

closed 2021-03-22 by ascillato2

mircozappon, 2021-03-22:
All right, I tried new version succesfully. Thanks a lot.