arendst/Tasmota #11406: Sonoff mini hw is switching quickly after a single use of hw switch

vahaldor, 2021-03-21:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
When using the hw switch the sonoff mini begin switching quickly on and off or does nothing or wait second and begin the fast switching.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff mini
- [x] Tasmota binary firmware version number used: 9.3.1
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
10:41:00.156 MQT: stat/tasmota/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
10:41:00.376 MQT: stat/tasmota/RESULT = {"Module":{"0":"Generic"}}
10:41:00.628 MQT: stat/tasmota/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"160":"Switch1"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"288":"Led1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
10:41:28.596 MQT: stat/tasmota/STATUS = {"Status":{"Module":0,"DeviceName":"Bathroom fan","FriendlyName":["Bathroom fan"],"Topic":"tasmota","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
10:41:28.604 MQT: stat/tasmota/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:26:55","StartupUTC":"2021-03-21T09:14:33","Sleep":50,"CfgHolder":4617,"BootCount":22,"BCResetTime":"2021-03-14T14:23:41","SaveCount":283,"SaveAddress":"F6000"}}
10:41:28.613 MQT: stat/tasmota/STATUS2 = {"StatusFWR":{"Version":"9.3.1(tasmota)","BuildDateTime":"2021-03-09T16:12:28","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"467/699"}}
10:41:28.619 MQT: stat/tasmota/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["BIOSPACE",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
10:41:28.628 MQT: stat/tasmota/STATUS4 = {"StatusMEM":{"ProgramSize":593,"Free":408,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
10:41:28.635 MQT: stat/tasmota/STATUS5 = {"StatusNET":{"Hostname":"tasmota-7328","IPAddress":"192.168.2.204","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"D8:F1:5B:8B:DC:A0","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
10:41:28.641 MQT: stat/tasmota/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.138","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_8BDCA0","MqttUser":"homeassistant","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
10:41:28.651 MQT: stat/tasmota/STATUS7 = {"StatusTIM":{"UTC":"2021-03-21T09:41:28","Local":"2021-03-21T10:41:28","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"06:50","Sunset":"19:03"}}
10:41:28.658 MQT: stat/tasmota/STATUS10 = {"StatusSNS":{"Time":"2021-03-21T10:41:28","Switch1":"ON"}}
10:41:28.666 MQT: stat/tasmota/STATUS11 = {"StatusSTS":{"Time":"2021-03-21T10:41:28","Uptime":"0T00:26:55","UptimeSec":1615,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"BIOSPACE","BSSId":"C0:C9:E3:8C:64:A7","Channel":4,"RSSI":100,"Signal":-44,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
10:41:56.160 CMD: weblog 4
10:41:56.167 MQT: stat/tasmota/RESULT = {"WebLog":4}
10:41:56.362 CFG: Saved to flash at F5, Count 284, Bytes 4096
10:42:00.476 WIF: Checking connection...
10:42:20.470 WIF: Checking connection...
10:42:40.502 WIF: Checking connection...
10:43:00.480 WIF: Checking connection...
10:43:20.471 WIF: Checking connection...
10:43:40.507 WIF: Checking connection...
10:44:00.497 WIF: Checking connection...
10:44:20.487 WIF: Checking connection...
10:44:40.472 WIF: Checking connection...
10:44:46.386 MQT: tele/tasmota/STATE = {"Time":"2021-03-21T10:44:46","Uptime":"0T00:30:13","UptimeSec":1813,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"BIOSPACE","BSSId":"C0:C9:E3:8C:64:A7","Channel":4,"RSSI":100,"Signal":-44,"LinkCount":1,"Downtime":"0T00:00:05"}}
10:44:46.392 MQT: tele/tasmota/SENSOR = {"Time":"2021-03-21T10:44:46","Switch1":"ON"}
10:45:00.509 WIF: Checking connection...
10:45:20.497 WIF: Checking connection...
10:45:40.488 WIF: Checking connection...
10:45:55.527 SRC: Switch
10:45:55.532 MQT: stat/tasmota/RESULT = {"POWER":"ON"}
10:45:55.536 MQT: stat/tasmota/POWER = ON
10:45:55.627 SRC: Switch
10:45:55.632 MQT: stat/tasmota/RESULT = {"POWER":"OFF"}
10:45:55.637 MQT: stat/tasmota/POWER = OFF
10:45:55.677 SRC: Switch
10:45:55.682 MQT: stat/tasmota/RESULT = {"POWER":"ON"}
10:45:55.687 MQT: stat/tasmota/POWER = ON
10:45:55.776 SRC: Switch
10:45:55.782 MQT: stat/tasmota/RESULT = {"POWER":"OFF"}
10:45:55.786 MQT: stat/tasmota/POWER = OFF
10:45:55.877 SRC: Switch
10:45:55.882 MQT: stat/tasmota/RESULT = {"POWER":"ON"}
10:45:55.887 MQT: stat/tasmota/POWER = ON
10:45:56.370 CFG: Saved to flash at F4, Count 285, Bytes 4096
10:46:00.490 WIF: Checking connection...

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Using the hw switch.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Turn on or off.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
Sonoff mini controls a fan that was previously controlled by a on/off switch, not a continuous switch. 

**(Please, remember to close the issue when the problem has been addressed)**

ThePirat91, 2021-03-21:
+1 here

schrej-zz, 2021-03-21:
Hi,
Sonoff Mini switch contact are running on just 3.3V.
If your line is very long, it might result in unclear on/off status. Are you lines to your rocker switches  very long (several meters)?

vahaldor, 2021-03-21:
No, they are less then 10cm.

vahaldor, 2021-03-22:
Solved, this was a problem of the mechanical rocker. When using wires against each other, it behaves normally. Sorry for rising it as a sw bug.

closed 2021-03-22 by vahaldor