arendst/Tasmota #11409: Cannot "read" the CT value of Hue light

LucReynders, 2021-03-21:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

When I issue the Zigbee command "read":0 in order to get the status of the hue light I can only get the values of the clusters 6 (power) and 8 (dimmer).  The cluster for the CT (color temp) is stored with cluster 0x0300 but when I issue "read":"0x0300" there is no response.
How can I read the CT status from the Hue light ?

Below : status 0
82 CMD: status 0
11:35:51.987 MQT: stat/ZBBridge/STATUS = {"Status":{"Module":75,"DeviceName":"ZBBridge","FriendlyName":["ZBBridge"],"Topic":"ZBBridge","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
11:35:51.991 MQT: stat/ZBBridge/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Power On","Uptime":"1T20:29:48","StartupUTC":"2021-03-19T14:06:03","Sleep":50,"CfgHolder":4617,"BootCount":22,"BCResetTime":"2021-03-12T18:49:13","SaveCount":41,"SaveAddress":"FC000"}}
11:35:51.994 MQT: stat/ZBBridge/STATUS2 = {"StatusFWR":{"Version":"9.3.1.1(zbbridge)","BuildDateTime":"2021-03-12T09:08:00","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":160,"Hardware":"ESP8266EX","CR":"398/699"}}
11:35:51.996 MQT: stat/ZBBridge/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["telenet-D7ABC",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00108009","2805C8000100060000005A00000000000000","000000C0","40006082","00000000"]}}
11:35:52.000 MQT: stat/ZBBridge/STATUS4 = {"StatusMEM":{"ProgramSize":574,"Free":1216,"Heap":25,"ProgramFlashSize":2048,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F1007C4","04400001","00000002","00000000","00000000","00020000","00000000","00000000"],"Drivers":"1,2,7,9,10,20,23,41","Sensors":""}}
11:35:52.003 MQT: stat/ZBBridge/STATUS5 = {"StatusNET":{"Hostname":"ZBBridge-5222","IPAddress":"192.168.0.45","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"195.130.130.3","Mac":"84:CC:A8:9B:94:66","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
11:35:52.007 MQT: stat/ZBBridge/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.10","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_9B9466","MqttUser":"DVES_USER","MqttCount":9,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
11:35:52.013 MQT: stat/ZBBridge/STATUS7 = {"StatusTIM":{"UTC":"2021-03-21T10:35:52","Local":"2021-03-21T11:35:52","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"06:50","Sunset":"19:03"}}
11:35:52.018 MQT: stat/ZBBridge/STATUS10 = {"StatusSNS":{"Time":"2021-03-21T11:35:52"}}
11:35:52.022 MQT: stat/ZBBridge/STATUS11 = {"StatusSTS":{"Time":"2021-03-21T11:35:52","Uptime":"1T20:29:49","UptimeSec":160189,"Vcc":3.504,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":9,"Wifi":{"AP":1,"SSId":"","BSSId":"34:2C:C4:F8:DC:23","Channel":11,"RSSI":100,"Signal":-38,"LinkCount":9,"Downtime":"0T00:00:32"}}}
11:36:12.104 MQT: tele/ZBBridge/STATE = {"Time":"2021-03-21T11:36:12","Uptime":"1T20:30:09","UptimeSec":160209,"Vcc":3.504,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":9,"Wifi":{"AP":1,"SSId":"","BSSId":"34:2C:C4:F8:DC:23","Channel":11,"RSSI":100,"Signal":-39,"LinkCount":9,"Downtime":"0T00:00:32"}}

s-hadinger, 2021-03-21:
The simplest way is `ZbSend {"Device":"<device>","Read":{"CT":true"}}`

Example:
```
ZbSend {"Device":"IKEA_Bulb","Read":{"CT":true"}}

13:42:22.650 MQT: stat/tasmota/ZBBridge/RESULT = {"ZbSend":"Done"}
13:42:23.109 MQT: tele/tasmota/ZBBridge/IKEA_Bulb_1/SENSOR = {"IKEA_Bulb":{"Device":"0xCA8B","Name":"IKEA_Bulb","CT":350,"Endpoint":1,"LinkQuality":105}}
```

LucReynders, 2021-03-21:
Great.  
I checked this also for "Power" and "Dimmer" values of the hue light (Ambiance 30x120) ... 
Thanks a lot.
PS :  should this not be part of the doc in tasmota ?  this is a nice feature for getting the status of attributes without knowing endpoints and clusters.

s-hadinger, 2021-03-21:
Doc is open source, feel free to submit improvements. Thx

closed 2021-03-21 by ascillato2