arendst/Tasmota #11420: ZBBridge image crashing on D1 mini

joseal, 2021-03-22:
### PROBLEM DESCRIPTION
D1 mini crashing when flashed with latest ZBBridge image.
Upon boot you can connect to the wifi setup, but you cannot even save your configuration, as the device keeps crashing.
Same device with the default image (tasmota.bin) works just fine.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): D1 mini
- [x] Tasmota binary firmware version number used: 9.3.1
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: Tasmotizer
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3584, room 16
tail 0
chksum 0xb0
csum 0xb0
v3969889e
~ld

00:00:00.024 CFG: Loaded from flash at FB, Count 4

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (9):
epc1=0x4023ab37 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00004bca depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffa20 end: 3fffffc0 offset: 0190
3ffffbb0:  00000005 fffffffd 00000002 4023a9b8
3ffffbc0:  3ffffbd0 3ffffc10 00000008 3fff362c
3ffffbd0:  3fffff10 3fffff00 00000008 3fff0020
3ffffbe0:  3fffff10 3fffff00 00000010 3fff362c
3ffffbf0:  00003000 3fff3656 3fffff0c 3ffffc30
3ffffc00:  000002bc 0000005f fffffffe 4020e85c
3ffffc10:  3a47495a 73655220 69747465 3fff21e0
3ffffc20:  00000005 fffffffd 00000002 4020e855
3ffffc30:  3ffffd60 0000004c 3ffffd60 402448fa
3ffffc40:  3ffffcc0 3ffffea0 3ffffc70 00000000
3ffffc50:  3ffffcd0 3fffff10 3ffffc80 40244830
3ffffc60:  3ffffd60 3ffffea0 3ffffcc0 4024535c
3ffffc70:  0000004c 00000017 00009000 00000001
3ffffc80:  00000000 00000001 3ffffd60 402448fa
3ffffc90:  3ffffd03 40286c20 00000000 3ffe835c
3ffffca0:  0000000c 00000001 40245231 40286c20
3ffffcb0:  00000000 40286c21 3ffffd60 40244ae4
3ffffcc0:  00000000 0000004c 00000000 00000000
3ffffcd0:  0000004c 00000153 00302073 40244830
3ffffce0:  3ffffde0 402860d5 3ffffd40 4024535c
3ffffcf0:  00000008 3ffffd30 0000000c 00000001
3ffffd00:  0032335c 00000001 3ffffde0 402448fa
3ffffd10:  3ffffd83 402860ea 00000000 40244150
3ffffd20:  3ffffe30 3ffffe20 00000028 3ffe835c
3ffffd30:  0000004c 00000001 00000000 40244d2f
3ffffd40:  00000000 00000008 00000000 00000007
3ffffd50:  3ffffea0 3ffe835c 00000fa0 40244099
3ffffd60:  3ffefed1 4bc6a7f0 00000e4c ffff0208
3ffffd70:  3ffefd7e 00000f9f 4bc6a7f0 32244150
3ffffd80:  00393230 3ffffd90 0000000c 1810624d
3ffffd90:  00000000 3ffef649 00000000 3ffefd7e
3ffffda0:  3ffffeb0 3ffffea0 00000024 3ffe835c
3ffffdb0:  00000008 00000009 00000001 4020e7fd
3ffffdc0:  00000ac8 3ffffe10 0000000c 3ffffee0
3ffffdd0:  00000af0 3ffef3c0 3ffffe60 402440dc
3ffffde0:  3ffffe30 00000001 00000000 401008b8
3ffffdf0:  00000000 00000009 00000047 4020a6d0
3ffffe00:  40104b87 00040000 00000000 00040000
3ffffe10:  00000000 40104b84 00040000 4021a56d
3ffffe20:  3ffed2f0 40103aeb 00000000 40210edb
3ffffe30:  00000007 3fff0dd0 00000001 3ffffe50
3ffffe40:  00000001 2c9f0300 4000050c 3fffc278
3ffffe50:  4010393c 3fffc200 00000022 3fff3654
3ffffe60:  4023eef7 00000030 0000001e ffffffff
3ffffe70:  4023eef2 3fff22ec f77338d2 4025eded
3ffffe80:  00000000 00000000 00000000 fffffffe
3ffffe90:  00000000 4bc6a7f0 000000c3 ffffffff
3ffffea0:  00000000 00000000 4bc6a7f0 00000000
3ffffeb0:  6d736154 2061746f 4010081e 56041893
3ffffec0:  00000000 00000000 3ffeecd0 40235454
3ffffed0:  3ffeecd0 00000000 3ffef0d0 4023a0d6
3ffffee0:  00000000 4bc6a7f0 000000f5 3fffff10
3ffffef0:  3fffff00 00000008 4bc6a7f0 00000000
3fffff00:  401063e9 fffffffd 3fff361c 00004bc6
3fffff10:  00000000 00000020 3ffee8a0 4023ae29
3fffff20:  00000005 fffffffd 3ffee950 40219557
3fffff30:  00000000 4bc6a7f0 000000f5 3fff21e0
3fffff40:  00000005 00000018 3ffee954 40230662
3fffff50:  3fffdad0 3ffef3c0 00000005 402197e1
3fffff60:  3fffdad0 00000000 000000f5 4022e936
3fffff70:  3fffdad0 00000000 000000f5 4022e9dc
3fffff80:  00000000 00000000 00000001 401004a4
3fffff90:  3fffdad0 00000000 3fff21a0 3fff21e0
3fffffa0:  3fffdad0 00000000 3fff21a0 4023e088
3fffffb0:  feefeffe feefeffe 3ffe857c 401014bd
<<<stack<<<

```

### TO REPRODUCE
Flash D1 mini with latest ZBBridge image.

### EXPECTED BEHAVIOUR
Be able to save the Wifi config and get to main menu.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2021-03-22:
Please try the latest development version. It should be solved.

arendst, 2021-03-22:
zbbridge firmware on a sonoff mini??????

joseal, 2021-03-22:
> Please try the latest development version. It should be solved.

Hi @s-hadinger, is the Dev 9.3.1.2 or do I need to compile latest sources?

joseal, 2021-03-22:
> zbbridge firmware on a sonoff mini??????

Hi Theo, it's not sonoff mini, it the D1 mini (esp8266 + serial).
I've just tried to flash it with the minimal build, do the wifi setup and then upgrade to zbbridge, and this way it works just fine.
I'm using the zbbridge image, because I want to use the TCPSerial to connect a CC2530 + CC2591 board (that only speaks serial) with the zigbee2mqtt on my raspberry. So, the ESP8266 (provided by the D1 mini board) running doing the ser2net, where the zigbee2mqtt is going to connect.

Let me take this opportunity to say thanks for the great job with this amazing project!

Jason2866, 2021-03-22:
Flashing the zbbridge build to any other device will end in unexpected results (as Theo already wrote). You can **NOT** use this build to connect a cc25xx!!
You have to compile yourself. See https://tasmota.github.io/docs/Compile-your-build/ or use a unofficial build. For example https://github.com/Jason2866/Tasmota-specials/tree/firmware

closed 2021-03-22 by ascillato2