arendst/Tasmota #11426: Sunset/Sunraise Times are swaped in Timer Configuration

tubercel, 2021-03-22:
### PROBLEM DESCRIPTION
Sunset/Sunraise Times are swaped in Timer Configuration

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Gosund SP111
- [ ] Tasmota binary firmware version number used: 9.3.1
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
Setup Timezone Longitude Latitude STD/DST for Timecalculation
go To Timer menue.
Times for Sunset / Sunraise seems to be swapped

### EXPECTED BEHAVIOUR
Sunset / Sunraise schould show correct times

### SCREENSHOTS

### ADDITIONAL CONTEXT

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2021-03-22:
Cannot reproduce

[image]

tubercel, 2021-03-22:
[image]

I investigated a bit more:

After a reboot systemtime stays at epoch 0.
If this happens Sunraise/set for 1.1.1970 ist calculated as obove. (seems wrong also)

If Systemtime ist correct everything is fine.

I take a closer look why this happens.
I think this PR can be closed.

ascillato2, 2021-03-22:
Closing this issue as it is **duplicated**. Sorry.

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-03-22 by ascillato2