arendst/Tasmota #11427: [ESP32] Please enable CONFIG_FREERTOS_UNICORE option in menuconfig

TomoKRK, 2021-03-22:
DO NOT DELETE ANY TEXT from this template! Otherwise the issue will be auto-closed.
-->

### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
This is actually the same issue as #522 however original has been locked and still the tasmota32.bin does not boot on ESP-WROOM-32D and on the console the log says

**E (668) cpu_start: Running on single core chip, but application is built with dual core support.
E (668) cpu_start: Please enable CONFIG_FREERTOS_UNICORE option in menuconfig.
abort() was called at PC 0x40087369 on core 0**

Is there an option to provide the build version of tasmota32 with the libraries available at 
[https://github.com/lbernstone/arduino-esp32-solo](https://github.com/lbernstone/arduino-esp32-solo)?

That way the closed issue and this one would be solved per my understanding.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): Yeelight LED Strip Controller/ESP-WROOM-32D
- [ ] Tasmota binary firmware version number used: 9.3.1.2
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [ x] Flashing tools used: esptool.py_____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Tasmota32 runs in SINGLECORE mode

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2021-03-22:
Take a look in `Platformio_override_sample.ini` there it is

Jason2866, 2021-03-22:
You can download a unoffical ESP32-solo1 Tasmota32 from here https://github.com/Jason2866/Tasmota-specials/tree/firmware

TomoKRK, 2021-03-22:
@Jason2866 - thank you for your help and support. I don't have platformIO installed and have no knowledge about it :(
However I will test this firmware version and report back accordingly.

Jason2866, 2021-03-22:
VSC / Platformio is easy to install. No "programmer" knowledge needed for compiling Tasmota.
See https://tasmota.github.io/docs/Compile-your-build/

TomoKRK, 2021-03-22:
@Jason2866 - ok, I will install it and give a try. In the meantime, just wanted to let you know that this build has booted up correctly and I was able to perform wifi setup etc. Now it will be a matter of finding the template to control this YeeLight strip

Jason2866, 2021-03-22:
You can ask the Tasmota community for help in Discord.

See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2021-03-22 by ascillato2