arendst/Tasmota #11434: [Zigbee] Sonoff ZBMini not pairs

snakuzzo, 2021-03-23:
### PROBLEM DESCRIPTION
If I try to pair a Sonoff ZBMini nothing happens. Other devices join ZB Network correctly.
Sonoff ZBMini works fine on other coordinator using Zigbee2mqtt

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP32 + CC2538/92
- [x] Tasmota binary firmware version number used: 9.3.1.2
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: VSCode + PlatformIO
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
01:15:07.923 CMD: Backlog Template; Module; GPIO 255
01:15:07.927 SRC: WebConsole from 192.168.1.2
01:15:07.928 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
01:15:07.971 SRC: Backlog
01:15:07.972 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
01:15:07.984 MQT: stat/zigbee2tasmota/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
01:15:08.192 SRC: Backlog
01:15:08.194 CMD: Group 0, Index 1, Command "MODULE", Data ""
01:15:08.204 MQT: stat/zigbee2tasmota/RESULT = {"Module":{"0":"ESP32-DevKit"}}
01:15:08.442 SRC: Backlog
01:15:08.444 CMD: Group 0, Index 1, Command "GPIO", Data "255"
01:15:08.461 MQT: stat/zigbee2tasmota/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"3584":"Zigbee Rx"},"GPIO23":{"3552":"Zigbee Tx"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO28":{"0":"None"}}
01:15:08.470 MQT: stat/zigbee2tasmota/RESULT = {"GPIO29":{"0":"None"},"GPIO30":{"0":"None"},"GPIO31":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua

```
- [x] Provide the output of this command: `Status 0`:
```lua
01:13:31.956 CMD: Status 0
01:13:31.958 SRC: WebConsole from 192.168.1.2
01:13:31.960 CMD: Group 0, Index 1, Command "STATUS", Data "0"
01:13:31.969 MQT: stat/zigbee2tasmota/STATUS = {"Status":{"Module":0,"DeviceName":"Zigbee2Tasmota","FriendlyName":["Tasmota"],"Topic":"zigbee2tasmota","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
01:13:32.005 MQT: stat/zigbee2tasmota/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:09:52","StartupUTC":"2021-03-23T00:03:39","Sleep":50,"CfgHolder":4617,"BootCount":23,"BCResetTime":"2021-02-23T21:25:37","SaveCount":47}}
01:13:32.040 MQT: stat/zigbee2tasmota/STATUS2 = {"StatusFWR":{"Version":"9.3.1.2(tasmota)","BuildDateTime":"2021-03-22T23:56:38","Core":"1_0_5_RC7","SDK":"v3.3.4-428-gd8082b7f3-dirty","CpuFrequency":80,"Hardware":"ESP32","CR":"451/699"}}
01:13:32.059 MQT: stat/zigbee2tasmota/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SnakeRM_24",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
01:13:32.092 MQT: stat/zigbee2tasmota/STATUS4 = {"StatusMEM":{"ProgramSize":1360,"Free":1984,"Heap":220,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":16384,"FlashSize":16384,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","9FDAC787","00552001","B7FFBFCF","01DA9BC4","60360DC7","00084052","20200000","00000000"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,17,20,21,23,24,26,27,29,34,50","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,51,52,55,56,58,59,64,66,67,74,85"}}
01:13:32.137 MQT: stat/zigbee2tasmota/STATUS5 = {"StatusNET":{"Hostname":"zigbee2tasmota-esp32","IPAddress":"192.168.1.19","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"4C:11:AE:B6:F5:A0","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
01:13:32.157 MQT: stat/zigbee2tasmota/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.10","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_B6F5A0","MqttUser":"snakuzzo","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
01:13:32.181 MQT: stat/zigbee2tasmota/STATUS7 = {"StatusTIM":{"UTC":"2021-03-23T00:13:32","Local":"2021-03-23T01:13:32","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"06:46","Sunset":"19:06"}}
01:13:32.202 MQT: stat/zigbee2tasmota/STATUS10 = {"StatusSNS":{"Time":"2021-03-23T01:13:32"}}
01:13:32.212 MQT: stat/zigbee2tasmota/STATUS11 = {"StatusSTS":{"Time":"2021-03-23T01:13:32","Uptime":"0T00:09:53","UptimeSec":593,"Heap":219,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"SnakeRM_24","BSSId":"DC:39:6F:6E:AD:35","Channel":6,"RSSI":100,"Signal":-44,"LinkCount":1,"Downtime":"0T00:00:04"}}}
```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
01:10:17.718 SRC: WebConsole from 192.168.1.2
01:10:17.719 CMD: Group 0, Index 1, Command "ZBPERMITJOIN", Data "1"
01:10:17.723 ZIG: ZbZNPSent 25360FFCFF3C00
01:10:17.731 MQT: tele/zigbee2tasmota/RESULT = {"ZbState":{"Status":21,"Message":"Pairing mode enabled"}}
01:10:17.737 MQT: stat/zigbee2tasmota/RESULT = {"ZbPermitJoin":"Done"}
01:10:17.790 ZIG: {"ZbZNPReceived":"653600"}
01:10:17.801 ZIG: {"ZbZNPReceived":"45B6000000"}
```

### TO REPRODUCE
Try to join a Sonoff ZBMini

### EXPECTED BEHAVIOUR
I expect it joins ZB Network

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2021-03-23:
Tasmota with ESP32 + CC2538/92 is a not tested (supported) variant.
We dont know if this is the reason, but we cant reproduce since we dont have this zigbee coordinator device

snakuzzo, 2021-03-23:
I'm alreay using latest Tasmota dev source code + latest PlatformIO plugin on VSCode.

Jason2866, 2021-03-23:
Your log says different 
`01:13:32.040 MQT: stat/zigbee2tasmota/STATUS2 = {"StatusFWR":{"Version":"9.3.1.2(tasmota)","BuildDateTime":"2021-03-22T23:56:38","Core":"1_0_5_RC7","SDK":"v3.3.4-428-gd8082b7f3-dirty","CpuFrequency":80,"Hardware":"ESP32","CR":"451/699"}}
`

snakuzzo, 2021-03-23:
I'm using latest version of PlatformIO addon....I will check another time later.

snakuzzo, 2021-03-24:
@Jason2866 this is what VSCode log when I run Build...

```
Processing tasmota32 (framework: arduino; platform: espressif32 @ 3.1.0; board: esp32dev)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (3.1.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 80MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 3.10005.0+rc7 (1.0.5) 
 - tool-esptoolpy 1.30000.201119 (3.0.0) 
 - tool-mklittlefs 1.203.200522 (2.3) 
 - toolchain-xtensa32 2.50200.80 (5.2.0)
 ```

Jason2866, 2021-03-25:
Delete hidden folder `.platformio`, by doing this platformio will rebuild with actual version(s).

snakuzzo, 2021-03-25:
Deleted `.pio` directory...

Arduino version still the same

```
Processing tasmota32 (framework: arduino; platform: espressif32 @ 3.1.0; board: esp32dev)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (3.1.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 80MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 3.10005.0+rc7 (1.0.5) 
 - tool-esptoolpy 1.30000.201119 (3.0.0) 
 - tool-mklittlefs 1.203.210203 (2.3) 
 - toolchain-xtensa32 2.50200.97 (5.2.0)
 ```

Jason2866, 2021-03-25:
Not `.pio` you have to delete folder `.platformio` 
Search in your user account folder

snakuzzo, 2021-03-25:
Done!...

Same result...

```
Processing tasmota32 (framework: arduino; platform: espressif32 @ 3.1.0; board: esp32dev)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (3.1.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 80MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 3.10005.210308 (1.0.5) 
 - tool-esptoolpy 1.30000.201119 (3.0.0) 
 - tool-mklittlefs 1.203.210203 (2.3) 
 ```

ascillato2, 2021-04-01:
Hi, Any news on this?

snakuzzo, 2021-04-01:
Hi @ascillato, I ordered a CC2531 to use as Zigbee sniffer with whsniff...becausa ESP32 was correctly flashed and when I try to pair a zbmini ... nothing happens...nothing was logged on Tasmota

CeesCKS, 2021-04-08:
I am experiencing the same problem. Zigbee ZBMINI does not join and nothing was logged in de console window

closed 2021-05-08 by github-actions[bot]