arendst/Tasmota #11451: Add shutter support to default ESP32 standard binaries

stefanbode, 2021-03-24:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
The default binary os ESP32 seems not to include the shutter option. After setting setoption80 1 the shuttercommands are still not available.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): _____
- [ x] Tasmota binary firmware version number used: latest ESP32
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
setoption80 1
restart 1
shutterclose
--> Error unknown command

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

stefanbode, 2021-03-24:
I also tried to use a platform_setting_override.ini and set #define USE_SHUTTER.

Still the shutter is not available in the self compiled binary. Something strange I do not understand....

stefanbode, 2021-03-24:
Looks like I do not understand the mechanic. Added USE_SHUTTER to platform override and still the commands are not recognized.....

Jason2866, 2021-03-24:
Shutter support is enabled in `my_user_config.h` already. There is no `#undef` in Tasmota32
https://github.com/arendst/Tasmota/blob/development/tasmota/tasmota_configurations_ESP32.h#L151
so shutter support should be in the build?!

stefanbode, 2021-03-24:
hi @Jason2866 , yes agree to your findings. Anyhow the rebuild file shows following after setoption80 1
```
13:49:23.751 SRC: WebConsole from 192.168.178.46
13:49:23.754 CMD: Group 0, Index 1, Command "SHUTTEROPEN", Data ""
13:49:23.759 RSL: stat/tasmota_F50A98/RESULT = {"Command":"Error"}
13:49:24.366 WIF: Checking connection...
```
the general definition: shutterrelay does work. status 13 also looks good.
```
13:50:29.176 CMD: status 13
13:50:29.178 SRC: WebConsole from 192.168.178.46
13:50:29.180 CMD: Group 0, Index 1, Command "STATUS", Data "13"
13:50:29.184 RSL: stat/tasmota_F50A98/STATUS13 = {"StatusSHT":{"SHT0":{"Relay1":11,"Relay2":12,"Open":100,"Close":100,"50perc":50,"Delay":0,"Opt":"0000","Calib":[300,500,700,900,1000],"Mode":"0"},"SHT1":{"Relay1":13,"Relay2":14,"Open":0,"Close":0,"50perc":0,"Delay":0,"Opt":"0000","Calib":[0,0,0,0,0],"Mode":"0"}}}
```
Any Idea?

stefanbode, 2021-03-24:
!!!! I got it. Problem exists with RELAY 11,12,13,14 I use.

Does work with RELAY 1. Lets debug where it came from

stefanbode, 2021-03-24:
Got it. BUG. Submit a change.....

PR #11456 (2021-03-24): Add shutter support for more than 8 relays
## Description:

Add shutter support for more than 8 relays

Additionally change stepper accelerator

**Related issue (if applicable):** fixes #11451

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works on Tasmota core ESP8266 V.2.7.4.9
  - [x] The code change is tested and works with core ESP32 V.1.0.5
  - [x] I accept the [CLA](https://github. [... cut]
commit 9b2dec43 (2021-03-24): Fix #11451

Additionally change stepper accelerator
closed 2021-03-24 by arendst