arendst/Tasmota #11478: Low pulse(3ms) on Relay 200mS after Reset (or Deep-sleep-wake-up)

GadingeL, 2021-03-26:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
On a GPIO defined as Relay a pulse is generated about 200 ms after RST goes high.
The length of the pulse is about 3mS. This happens only in case the state of the relay was on before reset. It also happens at deepsleep wake up. 
[image]

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): NodeMCU , ESP8266 Mod12F, AZ-Delivery
- [X] Tasmota binary firmware version number used: 9.1.3
  - [X] Pre-compiled
  - [X] Self-compiled
- [X] Flashing tools used: OTA/FlashESP8266  
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
23:58:46.797 RSL: tasmota_58CA73/stat/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
23:58:47.004 RSL: tasmota_58CA73/stat/RESULT = {"Module":{"18":"Generic"}}
23:58:47.259 RSL: tasmota_58CA73/stat/RESULT = {"GPIO0":{"4096":"DeepSleep"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"224":"Relay1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
no rules used

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
00:01:08.097 RSL: tasmota_58CA73/stat/STATUS = {"Status":{"Module":18,"DeviceName":"TasmotaLGA","FriendlyName":["TasmotaLGA"],"Topic":"tasmota_58CA73","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
00:01:08.104 RSL: tasmota_58CA73/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"External System","Uptime":"0T00:14:27","StartupUTC":"2021-03-26T22:46:41","Sleep":50,"CfgHolder":4617,"BootCount":123,"BCResetTime":"2021-03-23T22:54:50","SaveCount":227,"SaveAddress":"F9000"}}
00:01:08.110 RSL: tasmota_58CA73/stat/STATUS2 = {"StatusFWR":{"Version":"9.3.1(tasmota)","BuildDateTime":"2021-03-09T16:12:28","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"371/699"}}
00:01:08.114 RSL: tasmota_58CA73/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["GadingeL",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000000060000005A64000000000000","00000088","00006000","00000000"]}}
00:01:08.123 RSL: tasmota_58CA73/stat/STATUS4 = {"StatusMEM":{"ProgramSize":593,"Free":408,"Heap":27,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
00:01:08.129 RSL: tasmota_58CA73/stat/STATUS5 = {"StatusNET":{"Hostname":"tasmota_58CA73-2675","IPAddress":"192.168.17.150","Gateway":"192.168.17.254","Subnetmask":"255.255.255.0","DNSServer":"195.202.128.23","Mac":"EC:FA:BC:58:CA:73","Webserver":2,"WifiConfig":5,"WifiPower":17.0}}
00:01:08.133 RSL: tasmota_58CA73/stat/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_58CA73","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
00:01:08.138 RSL: tasmota_58CA73/stat/STATUS7 = {"StatusTIM":{"UTC":"2021-03-26T23:01:08","Local":"2021-03-27T00:01:08","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"05:44","Sunset":"18:18"}}
00:01:08.144 RSL: tasmota_58CA73/stat/STATUS10 = {"StatusSNS":{"Time":"2021-03-27T00:01:08"}}
00:01:08.149 RSL: tasmota_58CA73/stat/STATUS11 = {"StatusSTS":{"Time":"2021-03-27T00:01:08","Uptime":"0T00:14:27","UptimeSec":867,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"GadingeL","BSSId":"68:FF:7B:DF:77:62","Channel":5,"RSSI":18,"Signal":-91,"LinkCount":1,"Downtime":"0T00:00:07"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
- flash image
- define D5 as Relais 1
- switch Relay to ON
- attach scope Channel 1 to D5, Channel2 to RST
- set trigger to RST pos slope
- press & release Reset Button

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
- D5 should stay HIGH

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
- found a fix, but not sure if its the correct one , (this is my first entry here )
- added 2 lines in support_tasmota.ino
[image]

The command pinmode(.. ) sets the line to LOW, 3 ms later the normal POWERON status is set.
 With my change I set the output register to the status read from the pin.

   

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2021-03-27:
Nice finding.

I'll dive in.

Jason2866, 2021-03-27:
@arendst  we had no feedback of relais blinking (when restarting or upgrading) since the day you did your change to stop this. The peak is to short for relais.

arendst, 2021-03-27:
I thought so too.

Anyway, by moving the poweronstate function up I managed to shorten the pulse to 2mSecs.

Jason2866, 2021-03-27:
Maybe it is not possible to avoid the low pulse. https://randomnerdtutorials.com/esp8266-pinout-reference-gpios/
It seems only D1 and D2 are free from this behaviour

arendst, 2021-03-27:
I think it's best to leave it as designed.

What happens with OPs solution is that it reads the input value based on how the relay is connected to the port. It doesn't read the binary state of the input.

ascillato2, 2021-03-27:
Closing this issue as it has been fixed.

Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-03-27 by ascillato2