arendst/Tasmota #11486: MLX90614 sensor reading incorrect

balajisraghavan, 2021-03-27:
### PROBLEM DESCRIPTION
MLX90614 IR contactless temperature sensor reports incorrect temperatures with Tasmota. Espruino is able to report correct temperatures.
Tasmota web UI shows 31.4c, 29.1c,31.9c,33.0c etc
Espruino shows steady readings - 31.4c, 31.6c, 31.5c, 31.2c

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Node mcu esp8266
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: esptool
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Connect the GY-906 board to the ESP8266 NodeMCu board via standard I2C pins (d1 and d2). GY-906 already has power supply decoupling capacitor on board.

### EXPECTED BEHAVIOUR
Tasmota sensor readings are correct and stable

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
espruino does a crc check on the data, Tasmota doesn't. Maybe Tasmota is not excluding the CRC bit(s) and this is causing variance in the LSB? 

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2021-03-27:
@gemu2015  can you take a look?

gemu2015, 2021-03-27:
i actually don't use this sensor currently.  i remounted it and can confirm very noisy values.
the checksum is indeed ignored but this is only relevant with long wiring.
the driver is extremely simple and is the same implementation as e.g. adafruit.
it would be easy to flatten the values with a moving average, but we should evaluate what is the reason for that noise. we can expect about 0.2 degrees from the data sheet but it is far more.
i am currently short of time but will try to check for possible error source probably next week.

gemu2015, 2021-03-27:
i simply restored the previous version of getting the 3 bytes from i2c and it works again.
very strange. attached is the modified driver. please verify.
have to find out why the tasmota call  I2cRead24() fails in this case.

[xsns_46_MLX90614.ino.zip](https://github.com/arendst/Tasmota/files/6215875/xsns_46_MLX90614.ino.zip)

balajisraghavan, 2021-03-29:
@gemu2015 thanks for sharing a fix. Please add the fix to the dev branch as well when you get the chance

PR #11514 (2021-03-30): Fixes
## Description:

**Related issue (if applicable):** fixes #11486
**Related issue (if applicable):** fixes #7004

fixes MLX90614 erratic readings
SML better modbus serial config
fix scripter executing restart vector on boot

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works on Tasmota core ESP8266 V.2.7.4.9
  - [x] The code change is tested and works with core ESP32 V.1.0.6
  - [x [... cut]
closed 2021-03-30 by arendst