arendst/Tasmota #11488: WIFI connection broken after few time

boaschti, 2021-03-27:
### PROBLEM DESCRIPTION
After few time (5h-30h) the Tasmota/Wemos isn't reachable via WIFI and MQTT.
A restart fixes WIFI connection.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Wemos D1 mini, Fingerprint, Switches, Relays
- [ ] Tasmota binary firmware version number used: latest main, dev branch 1,5weeks ago
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: atom/platformio
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

14:13:01.852 SRC: Backlog
14:13:01.852 CMD: Gruppe 0, Index 1, Befehl "TEMPLATE", Daten ""
14:13:01.855 RSL: stat/tuerpanel/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
14:13:12.496 WIF: Prüfe Verbindung...
14:13:24.411 CMD: Module
14:13:24.414 SRC: Serial
14:13:24.415 CMD: Gruppe 0, Index 1, Befehl "MODULE", Daten ""
14:13:24.417 RSL: stat/tuerpanel/RESULT = {"Module":{"18":"Generic"}}
14:13:25.416 RSL: tele/tuerpanel/STATE = {"Time":"2021-03-27T14:13:25","Uptime":"0T03:47:52","UptimeSec":13672,"Heap":19,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":70,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"SmartHomeZeig","BSSId":"00:E0:4C:13:96:F9","Channel":6,"RSSI":64,"Signal":-68,"LinkCount":1,"Downtime":"0T00:00:04"}}
14:13:25.442 RSL: tele/tuerpanel/SENSOR = {"Time":"2021-03-27T14:13:25","Switch1":"ON","Switch2":"OFF","Switch3":"OFF","Switch4":"OFF"}
14:13:31.347 CMD: GPIO 255
14:13:31.350 SRC: Serial
14:13:31.351 CMD: Gruppe 0, Index 1, Befehl "GPIO", Daten "255"
14:13:31.358 RSL: stat/tuerpanel/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"224":"Relay1"},"GPIO3":{"0":"None"},"GPIO4":{"5504":"AS608 Tx"},"GPIO5":{"5536":"AS608 Rx"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"194":"Switch_n3"},"GPIO13":{"161":"Switch2"},"GPIO14":{"160":"Switch1"},"GPIO15":{"225":"Relay2"},"GPIO16":{"163":"Switch4"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

14:14:59.872 CMD: Gruppe 0, Index 1, Befehl "STATUS", Daten "0"
14:14:59.876 RSL: stat/tuerpanel/STATUS = {"Status":{"Module":18,"DeviceName":"Tasmota","FriendlyName":["Tasmota","Tasmota2"],"Topic":"tuerpanel","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[4,1,15,15,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
14:14:59.909 RSL: stat/tuerpanel/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota-DE.bin.gz","RestartReason":"Software/System restart","Uptime":"0T03:49:26","StartupUTC":"2021-03-27T09:25:33","Sleep":50,"CfgHolder":4617,"BootCount":90,"BCResetTime":"2020-10-17T20:06:41","SaveCount":604,"SaveAddress":"F5000"}}
14:14:59.944 RSL: stat/tuerpanel/STATUS2 = {"StatusFWR":{"Version":"9.3.1(tasmota)","BuildDateTime":"2021.03.27 10:21:13","Boot":31,"Core":"2_7_4_3","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"374/699"}}
14:14:59.962 RSL: stat/tuerpanel/STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SmartHomeZeig",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
14:14:59.994 RSL: stat/tuerpanel/STATUS4 = {"StatusMEM":{"ProgramSize":598,"Free":404,"Heap":19,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":3,"Features":["00000407","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001800","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6,79"}}
14:15:00.001 RSL: stat/tuerpanel/STATUS5 = {"StatusNET":{"Hostname":"tuerpanel","IPAddress":"192.168.178.117","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.38","Mac":"50:02:91:FE:55:40","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
14:15:00.021 RSL: stat/tuerpanel/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.38","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_FE5540","MqttUser":"mqttClients","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
14:15:00.044 RSL: stat/tuerpanel/STATUS7 = {"StatusTIM":{"UTC":"2021-03-27T13:15:00","Local":"2021-03-27T14:15:00","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"06:37","Sunset":"19:13"}}
14:15:00.065 RSL: stat/tuerpanel/STATUS10 = {"StatusSNS":{"Time":"2021-03-27T14:15:00","Switch1":"ON","Switch2":"OFF","Switch3":"OFF","Switch4":"OFF"}}
14:15:00.078 RSL: stat/tuerpanel/STATUS11 = {"StatusSTS":{"Time":"2021-03-27T14:15:00","Uptime":"0T03:49:27","UptimeSec":13767,"Heap":19,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":66,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"SmartHomeZeig","BSSId":"00:E0:4C:13:96:F9","Channel":6,"RSSI":68,"Signal":-66,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

13:55:41.164 CMD: status
13:55:41.171 RSL: stat/tuerpanel/STATUS = {"Status":{"Module":18,"DeviceName":"Tasmota","FriendlyName":["Tasmota","Tasmota2"],"Topic":"tuerpanel","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[4,1,15,15,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
13:57:42.794 MQT: Verbindungsversuch...
13:57:43.007 MQT: Verbindung fehlgeschlagen aufgrund von 192.168.178.38:1883, rc -2. Wiederversuch in 120 s
13:59:00.467 CMD: SerialLog 4
13:59:00.472 RSL: stat/tuerpanel/RESULT = {"SerialLog":{"4":{"Active":"4"}}}
13:59:01.092 CFG: in Flash gespeichert am F5, Anzahl 604, Bytes 4096
[... 34 lines cut ...]
14:01:30.782 WIF: Network 2, AP-, SSId Fuer Gaeste, Channel 5, BSSId CE:CE:1E:C1:6F:DE, RSSI -66, Encryption 1
14:01:30.792 WIF: Network 3, AP1, SSId SmartHomeZeig, Channel 6, BSSId 00:E0:4C:13:96:F9, RSSI -61, Encryption 1
14:01:30.802 WIF: Network 4, AP1, SSId SmartHomeZeig, Channel 11, BSSId 00:E0:4C:13:E5:9B, RSSI -86, Encryption 1
14:01:38.728 WIF: Prüfe Verbindung...
14:01:58.737 WIF: Prüfe Verbindung...
14:02:25.245 NTP: No reply
14:02:40.500 WIF: Prüfe Verbindung...

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

If this problem is no Hardware bug then take a wemos, compile tasmota-DE, uncomment #define USE_AS608 and #define USE_AS608_MESSAGES, config AS608 at any pin. (But I dont think that this triggers a WIFI Problem...)

### EXPECTED BEHAVIOUR
If Wifi connectin was lost, a reinit have to fix it. Other TASMOTAS (older various Versions) in the same Network has no Problems.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2021-03-27:
Is it a Wemos-mini? The issue you have is the typical behaviour of a power issue caused from the two weak onboard
3.3 vreg. Nearly all (clone) boards do have only a 3.3V which only delivers 150mA which is not enough for Tasmota and a
sensor connected. Please try a NodeMCU (which has a decent Vreg on board).

boaschti, 2021-03-27:
Yes Wemos D1 mini Clone. I'll try. Much thanks.

boaschti, 2021-03-27:
[image]

Oh, what a poor voltage regulation. This is measured without any sensor on 3V3 pin. I'll give report if it works now.

closed 2021-03-27 by ascillato2