arendst/Tasmota #11506: USE_GRAPH directive breaks the build

balajisraghavan, 2021-03-29:
### PROBLEM DESCRIPTION
Adding 
`#define USE_GRAPH
#define NUM_GRAPHS 4
`
breaks the current dev branch build
Output:
/workspace/Tasmota/tasmota/xdrv_13_display.ino: In function 'void DisplayInitDriver()':
/workspace/Tasmota/tasmota/xdrv_13_display.ino:1599:58: error: 'graph' was not declared in this scope
   for (uint8_t count = 0; count < NUM_GRAPHS; count++) { graph[count] = 0; }

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Nodemcu v3
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [x] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Add the below to user config override and build
#ifndef USE_DISPLAY
#define USE_DISPLAY
#define USE_DISPLAY_SSD1306
#define USE_DISPLAY_SH1106
#define USE_GRAPH
#define NUM_GRAPHS 4
#endif

### EXPECTED BEHAVIOUR
Build completes successfully

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2021-03-29:
Just tried. Latest development version does compile successfully with 
```
#define USE_DISPLAY
#define USE_DISPLAY_SSD1306
#define USE_DISPLAY_SH1106
#define USE_GRAPH
#define NUM_GRAPHS 4
``` settings in ` user_config_override.h`

balajisraghavan, 2021-03-29:
Interesting. Looks like adding the wifi and MQTT config with USE_GRAPH causes the build to break. Using the below settings causes the build to break for me.

#undef  STA_SSID1
#define STA_SSID1         ""             // [Ssid1] Wifi SSID

//#undef  STA_PASS1
#define STA_PASS1         ""     // [Password1] Wifi password

// -- Setup your own MQTT settings  ---------------
#undef  MQTT_HOST
#define MQTT_HOST         "" // [MqttHost]

#undef  MQTT_PORT
#define MQTT_PORT         1883                   // [MqttPort] MQTT port (10123 on CloudMQTT)

#undef  MQTT_USER
#define MQTT_USER         ""         // [MqttUser] Optional user

#undef  MQTT_PASS
#define MQTT_PASS         ""         // [MqttPassword] Optional password

#ifndef USE_DISPLAY
#define USE_DISPLAY
#define USE_DISPLAY_SSD1306
#define USE_DISPLAY_SH1106
//#define SHOW_SPLASH
//#define USE_AWATCH	
#define USE_GRAPH
#define NUM_GRAPHS 4
#endif

Jason2866, 2021-03-29:
Not possible since this values are already defined in `my_user_config.h`
and you just redefine it. Your setup is screwed somewhere.
There is no bug.