arendst/Tasmota #11507: HC-SR04 sersor not working wit ESP32

rosariopomillo, 2021-03-29:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
HC-SR04 sersor not working wit ESP32, i.e no 'SR04 Distance' shown

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Olimex ESP32-PoE-ISO; ESP32-wroom-32s; (Wemos D1mini for reference)
- [x] Tasmota binary firmware version number used: 9.3.1
  - [x] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: ESP-Flasher-Windows-x64
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
15:21:48.110 CMD: Backlog Template; Module; GPIO 255
15:21:48.159 MQT: stat/meteo32/RESULT = {"NAME":"Olimex ESP32-PoE","GPIO":[1,1,1,1,1,1,0,0,5536,1,1,1,1,0,5600,0,0,0,0,5568,0,0,0,0,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
15:21:48.376 MQT: stat/meteo32/RESULT = {"Module":{"0":"Olimex ESP32-PoE"}}
15:21:48.636 MQT: stat/meteo32/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"5536":"ETH POWER"},"GPIO13":{"640":"I2C SDA"},"GPIO14":{"1856":"SR04 Tri/TX"},"GPIO15":{"1888":"SR04 Ech/RX"},"GPIO16":{"608":"I2C SCL"},"GPIO17":{"0":"None"},"GPIO18":{"5600":"ETH MDIO"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"5568":"ETH MDC"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"}}
15:21:48.688 MQT: stat/meteo32/RESULT = {"GPIO27":{"0":"None"},"GPIO28":{"0":"None"},"GPIO29":{"0":"None"},"GPIO30":{"0":"None"},"GPIO31":{"0":"None"},"GPIO32":{"4768":"ADC Light1"},"GPIO33":{"4704":"ADC Input1"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
15:23:10.017 CMD: status 0
15:23:10.033 MQT: stat/meteo32/STATUS = {"Status":{"Module":0,"DeviceName":"Meteo32","FriendlyName":["Meteo32"],"Topic":"meteo32","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
15:23:10.065 MQT: stat/meteo32/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:01:40","StartupUTC":"2021-03-29T13:21:30","Sleep":50,"CfgHolder":4617,"BootCount":23,"BCResetTime":"2021-03-29T12:02:54","SaveCount":49}}
15:23:10.099 MQT: stat/meteo32/STATUS2 = {"StatusFWR":{"Version":"9.3.1(tasmota)","BuildDateTime":"2021-03-29T11:32:04","Core":"1_0_5_RC7","SDK":"v3.3.4-428-gd8082b7f3-dirty","CpuFrequency":80,"Hardware":"ESP32","CR":"507/699"}}
15:23:10.119 MQT: stat/meteo32/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["mG2tP6PKyTPA","FASTWEB-1-460C67"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000088","00006000","00000000"]}}
15:23:10.154 MQT: stat/meteo32/STATUS4 = {"StatusMEM":{"ProgramSize":1125,"Free":1984,"Heap":197,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","0FD00786","00100001","420014C6","0C0000C0","40080801","40004000","00200000","00000000"],"Drivers":"1,2,3,4,7,8,9,10,12,20,21,24,29,50,82","Sensors":"2,3,5,6,9,10,22,27,45,46,57"}}
15:23:10.189 MQT: stat/meteo32/STATUS5 = {"StatusNET":{"Hostname":"meteo32-1272_eth","IPAddress":"192.168.1.104","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.254","Mac":"FC:F5:C4:3E:E4:FB","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
15:23:10.210 MQT: stat/meteo32/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.1","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_3EE4F8","MqttUser":"UAA170I","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
15:23:10.234 MQT: stat/meteo32/STATUS7 = {"StatusTIM":{"UTC":"2021-03-29T13:23:10","Local":"2021-03-29T15:23:10","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":99,"Sunrise":"07:14","Sunset":"19:52"}}
15:23:10.289 MQT: stat/meteo32/STATUS10 = {"StatusSNS":{"Time":"2021-03-29T15:23:10","ANALOG":{"A1":0,"Illuminance2":6},"BH1750":{"Illuminance":7},"MLX90614":{"OBJTMP":28.3,"AMBTMP":25.3}}}
15:23:10.306 MQT: stat/meteo32/STATUS11 = {"StatusSTS":{"Time":"2021-03-29T15:23:10","Uptime":"0T00:01:40","UptimeSec":100,"Heap":197,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":47,"MqttCount":1,"Wifi":{"AP":1,"SSId":"mG2tP6PKyTPA","BSSId":"F0:B0:14:93:7F:E5","Channel":6,"RSSI":38,"Signal":-81,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
1) any ESP32 self-compiled bin have been compared with the corresponding ESP8266 self-compiled version tested on a wemos-D1mini; the ESP8266 version performed as expected all the times, the ESP32 version NEVER performed as expected.
2) SR04 sensor is arranged on a breadbord as for https://tasmota.github.io/docs/HC-SR04/ Swapping from one ESP to another requires to re-arrange power supply wires only.
3) I know support for ESP32 is a beta; I don't expect all is ok: this is only to highlight the problem in my opinion exists  

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2021-03-29:
I saw the same lack of detection trying to connect HC-SR04 to an ESP32 dev board with tasmota32.bin.

sfromis, 2021-03-30:
My ESP32-connected HC-SR04 test came online :-) Works about as "bad" as usually with ESP8266....

rosariopomillo, 2021-03-30:
Dowloaded and compiled last development and confirm SR04 is online and distance is stable; but someting seems wrong in i2c:
[image]
one non-existent TLS2561 appeared and my VL53L0X disappeared:
[image]
Highlight: these two items share the same i2c address (0x29). Coul'd help?
Then I reverted to previous 9.3.1 release and:
[image]
SR04 disappears (as expected)
VL53L0X re-appears (as expected).
Because I can't say if I2C anomaly and SR04 bug fix are related, I'm asking what's best to do:
a) close this issue and open a new one?
b) leave this issue open and supply more data?
c) what else?
Thanks in advance
R.

sfromis, 2021-03-30:
It does not belong in this issue, and does not need a fix. For I2C operation, you may have to [disable](https://tasmota.github.io/docs/I2CDevices/) drivers getting involved when it is not "their" device on the address.

rosariopomillo, 2021-03-30:
Thanks a lot for direction, now it works as expected. I will close the issue;
Just one more clarification: using TSL2591 & TSL2591 are both commented in my_user_config.h and in user_config_override.h, so I'm surprised to see the 16th driver enabled
01:26:28.743 MQT: stat/meteo32/RESULT = {"I2CDriver":"7,8,9,10,11,12,13,14,15,16,17,18,20,24,29,31,32,36,41,42,44,46,48,58"}
How can I suppress any driver at compile time?
Thanks again
R.

sfromis, 2021-03-31:
As described, you can [#undef](https://tasmota.github.io/docs/Compile-your-build/#customize-your-build) default features (of which the base build has a lot). You can also leave features enabled, but [not enable](https://github.com/arendst/Tasmota/blob/development/tasmota/my_user_config.h#L531-L535) I2C drivers.

rosariopomillo, 2021-03-31:
sorry if I insist, TSL2561 & TSL2591 are both  #undef in my files.
anyway at the moment this is not an issue, but remains unclear to me. 
bye
R.

closed 2021-03-31 by rosariopomillo

sfromis, 2021-03-31:
What you showed in a screen shot was *not* you having made #undef, just having #define commented out. You need active #undef to remove default features. Without you making any #define, lots of features are already enabled.

rosariopomillo, 2021-03-31:
Now it's clear! Thanks!
Sorry for that: as you can imagine I'm not an expert...
bye