arendst/Tasmota #11517: If more than 4 relay and steoption80 on. Alexa detect hue but doesn't work for the 5st friendlyname ... ? and other ? (connection lost?)

johnder1972, 2021-03-30:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

By using a sonoff  4CH and adding one pwm output (for ex on GPIO13)., web interface contains 4 power buttons and one slider for dimming.
By activating the Hue emulation in the "config other" menu, 5 frinendly names are created. 
By the way all seems working except when I begin to control them with Alexa. Alexa says that the device is not responding. 
Belking Wemo emulation is working good
I'm using it for controling  2 shutters.
I tested the http://192.168.1.58/api/user/lights/number/state/on:false command and the console report this 
```lua
10:34:59.209 /lights path=/lights/217825317/state/on:false
10:34:59.219 HTP: Hue Result ({"state":{"on":true,"bri":0,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","nam...
10:35:05.836 UDP: Packet (119)
10:35:05.839 UDP: HUE
10:35:06.855 UDP: Packet (119)
10:35:06.858 UDP: HUE
10:35:07.779 UDP: Packet (119)
10:35:07.782 UDP: HUE
10:35:08.003 WIF: Checking connection...
10:35:08.802 UDP: Packet (119)
10:35:08.805 UDP: HUE
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x]  Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Geekcreit "Sonoff pro" 4 Channel Relay RF 220V (https://fr.aliexpress.com/item/1005001650583201.html?spm=a2g0o.productlist.0.0.1e7c5fa73Dv7Sj&algo_pvid=30a6e146-ee8e-4fe5-b731-bf84bb1833b1&algo_expid=30a6e146-ee8e-4fe5-b731-bf84bb1833b1-0&btsid=0b0a556416170848707078568e831f&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_)
- [x] Tasmota binary firmware version number used: 9.1.3.2 to have issue about detection solved ( the unique id of the hue api must have a end like :11-XX , with two digits and not only one as in the 9.3.1 version)
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: ESP-Flasher-Windows-x64
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
07:33:02.817 CMD: Backlog Template; Module; GPIO 255
07:33:02.891 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
07:33:03.101 RSL: RESULT = {"Module":{"0":"Generic"}}
07:33:03.353 RSL: RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"226":"Relay3"},"GPIO5":{"225":"Relay2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"448":"PWM_i1"},"GPIO14":{"0":"None"},"GPIO15":{"227":"Relay4"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
07:34:21.773 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota1","Tasmota2","Tasmota3","Tasmota4","Tasmota5"],"Topic":"tasmota_CFBC02","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
07:34:21.803 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:03:20","StartupUTC":"2021-03-30T06:31:01","Sleep":50,"CfgHolder":4617,"BootCount":41,"BCResetTime":"2021-03-29T16:47:00","SaveCount":281,"SaveAddress":"FB000"}}
07:34:21.837 RSL: STATUS2 = {"StatusFWR":{"Version":"9.3.1.2(tasmota)","BuildDateTime":"2021-03-28T19:14:21","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"384/699"}}
07:34:21.855 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Dorothy",""],"TelePeriod":300,"Resolution":"55C180C0","SetOption":["00008001","2805C8000100060000005A0A000000000000","48000080","08006000","00000000"]}}
07:34:21.885 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":595,"Free":408,"Heap":27,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
07:34:21.917 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota_CFBC02-7170","IPAddress":"192.168.1.58","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"C8:2B:96:CF:BC:02","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
07:34:21.939 RSL: STATUS7 = {"StatusTIM":{"UTC":"2021-03-30T06:34:21","Local":"2021-03-30T07:34:21","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"06:31","Sunset":"19:17"}}
07:34:21.958 RSL: STATUS10 = {"StatusSNS":{"Time":"2021-03-30T07:34:21","Shutter1":{"Position":100,"Direction":0,"Target":100},"Shutter2":{"Position":100,"Direction":0,"Target":100}}}
07:34:21.974 RSL: STATUS11 = {"StatusSTS":{"Time":"2021-03-30T07:34:21","Uptime":"0T00:03:20","UptimeSec":200,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","Dimmer":84,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Dorothy","BSSId":"A6:91:B1:86:DA:80","Channel":1,"RSSI":80,"Signal":-60,"LinkCount":1,"Downtime":"0T00:00:05"}}}
07:34:21.014 RSL: STATUS13 = {"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":100,"Close":100,"50perc":50,"Delay":0,"Opt":"0000","Calib":[300,500,700,900,1000],"Mode":"0"},"SHT1":{"Relay1":3,"Relay2":4,"Open":100,"Close":100,"50perc":50,"Delay":0,"Opt":"0000","Calib":[300,500,700,900,1000],"Mode":"0"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:

I tested the http://ipaddress/api/user/lights/number/state/on:false command and the console report this 
```lua
  Console output here:

10:34:59.209 /lights path=/lights/217825317/state/on:false
10:34:59.219 HTP: Hue Result ({"state":{"on":true,"bri":0,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","nam...
10:35:05.836 UDP: Packet (119)
10:35:05.839 UDP: HUE
10:35:06.855 UDP: Packet (119)
10:35:06.858 UDP: HUE
10:35:07.779 UDP: Packet (119)
10:35:07.782 UDP: HUE
10:35:08.003 WIF: Checking connection...
10:35:08.802 UDP: Packet (119)
10:35:08.805 UDP: HUE
```

### TO REPRODUCE
_Steps to reproduce the behavior:_

By using a sonoff pro or compatible :
put 9.3.1.2  dev firmware to fix de alexa disocover bug 
add pwm to gpio 13
backlog setoption80 1;shutterrelay1 1; shutterrelay2 3
do the reconize with alexa 
and then tasmota5 still "off line", so uable to control with alexa.

(if you set setoption80 0 , all is ok with alexa)

(Use http://ipaddress/api/user/lights/number/state/on:false command to optain the console one result in the console)

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Alexa should be capable to control the "tasmota5"

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

thx

**(Please, remember to close the issue when the problem has been addressed)**

rrussev, 2021-04-02:
I can confirm there is something wrong with Alexa when there are more than four relays. I also have two shutters and two other relays and I was unable to make all of them work with Alexa. Unfortunately I didn’t have enough time to test more.

johnder1972, 2021-04-03:
Hello, I forget to post one screen. Don't hesitate to ask me more info. I can develop also in "arduino". I'm not a king but I could maybe help. I've a few investigated but I think that this bug it's not so simple to troubleshoot. By waiting, please find following the "config other" screen  : 
[image]

CU
John

arendst, 2021-04-03:
Pls give latest development version a try and report back.

johnder1972, 2021-04-03:
Hello, 

Yes. With my first test, it works now.
(I searched about a ">4 "  syntax but I was in the xdr_20_hue file , not in "shutters" file. May be next time I  will be better :-) )

thx a lot Theo (and all others)
I will reopen it if I find something other.

Nice day to all people working on tasmota. And really thx again.

closed 2021-04-03 by ascillato2